fwupd/fwupd #1723: fwupd uses 2% CPU constantly on HP Spectre x360 running Fedora 31

davewthompson, 2020-01-24:
**Issue**
On starting laptop, fwupd uses 2% of the CPU constantly according to htop; this causes the fan to kick in.
Stopping fwupd using systemctl stop fwupd shuts off the fan.

**Requested Details**
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
[dave@spectre-360 ~]$ fwupdmgr --version
client version:	1.3.6
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.3.6
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
[dave@spectre-360 ~]$ fwupdmgr get-devices
HP Spectre Convertible 13-ae0xx
│
├─Thunderbolt Controller:
│     Device ID:           e9f936367f6f98a8550bbfd1a688dbcff9cd3e4b
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              HP (TBT:0x00F0)
│     GUIDs:               15105a7c-9b49-5e4b-a8ed-3a2084d61805
│                          a2e18c4b-a520-5292-ab5e-9dc968e11c0e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─SAMSUNG MZVLB512HAJQ-000H1:
│     Device ID:           f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
│     Summary:             NVM Express Solid State Drive
│     Current version:     EXA72H1Q
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          f11f7049-b2e8-59dd-ae1d-29034b3e8a19
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           b6d3640817c3334f2f36890539014b6156262f18
│     Current version:     15.49.0
│     Minimum Version:     15.49.0
│     Vendor:              HP (DMI:AMI)
│     GUID:                5582f387-7110-4075-ad36-a19523e1c89f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─USB2.0 Hub:
│     Device ID:           8ad56c3d40864e26aecb5621725c2ff0ad4ec52a
│     Summary:             USB 2.x Hub
│     Current version:     2.23
│     Vendor:              VIA Labs, Inc. (USB:0x2109)
│     Install Duration:    15 seconds
│     GUIDs:               c6711eb4-d70f-5155-be78-3d72e3d17ceb
│                          3f0581b6-15ed-5f87-91eb-296db71e0838
│                          c93be422-8c70-50ad-a260-5e896bb4d55a
│                          e7581d99-314a-5ba9-94af-005dca091db6
│                          0adb66da-75a7-58ef-9abd-11baf5acd519
│     Device Flags:        • Updatable
│                          • Cryptographic hash verification is available
│                          • Device stages updates
│                          • Device can recover flash failures
│   
└─USB3.0 Hub:
      Device ID:           708a5b99d0afc737cdfcab5e16716ecbc6dee884
      Summary:             USB 3.x Hub
      Current version:     2.23
      Vendor:              VIA Labs, Inc. (USB:0x2109)
      Install Duration:    15 seconds
      GUIDs:               650573b6-dafe-5fcb-918e-f604c8c96654
                           26f33695-3a3e-5c08-badb-f6141390ebd9
                           c93be422-8c70-50ad-a260-5e896bb4d55a
                           16c861c2-c66a-509e-b185-bf04540f08d5
                           d87dbd95-2f89-5d8d-bd8f-899b3781d57d
      Device Flags:        • Updatable
                           • Cryptographic hash verification is available
                           • Device stages updates
                           • Device can recover flash failures
```

```shell
[dave@spectre-360 ~]$ efibootmgr -v
BootCurrent: 0005
Timeout: 10 seconds
BootOrder: 0005,0001,0000,0006,9999
Boot0000* Windows Boot Manager	HD(1,GPT,65af894c-71da-41fe-a59d-eb62fd1bb139,0x800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* neon	HD(1,GPT,f0039ef5-43c2-4313-9372-ab341832ea1e,0x800,0x100000)/File(\EFI\neon\shimx64.efi)
Boot0005* Fedora	HD(1,GPT,bac0d504-80e1-4114-bc34-548ed1c5e0f7,0x800,0x12c000)/File(\EFI\fedora\shimx64.efi)
Boot0006* Solid State Disk	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-8C-71-01-A1-87)/HD(1,GPT,bac0d504-80e1-4114-bc34-548ed1c5e0f7,0x800,0x12c000)..BO
Boot9999* USB Drive (UEFI)	PciRoot(0x0)/Pci(0x1d,0x0)/USB(16,0)..BO
```

```shell
[dave@spectre-360 ~]$ efivar -l | grep fw
bash: efivar: command not found...
Install package 'efivar' to provide command 'efivar'? [N/y] n
```

```shell
[dave@spectre-360 ~]$ tree /boot
/boot
├── config-5.3.7-301.fc31.x86_64
├── config-5.4.12-200.fc31.x86_64
├── config-5.4.8-200.fc31.x86_64
├── efi [error opening dir]
├── elf-memtest86+-5.01
├── extlinux
│   ├── cat.c32
│   ├── chain.c32
│   ├── cmd.c32
│   ├── cmenu.c32
│   ├── config.c32
│   ├── cptime.c32
│   ├── cpu.c32
│   ├── cpuid.c32
│   ├── cpuidtest.c32
│   ├── debug.c32
│   ├── dhcp.c32
│   ├── dir.c32
│   ├── disk.c32
│   ├── dmi.c32
│   ├── dmitest.c32
│   ├── elf.c32
│   ├── ethersel.c32
│   ├── gfxboot.c32
│   ├── gpxecmd.c32
│   ├── hdt.c32
│   ├── hexdump.c32
│   ├── host.c32
│   ├── ifcpu64.c32
│   ├── ifcpu.c32
│   ├── ifmemdsk.c32
│   ├── ifplop.c32
│   ├── kbdmap.c32
│   ├── kontron_wdt.c32
│   ├── ldlinux.c32
│   ├── lfs.c32
│   ├── libcom32.c32
│   ├── libgpl.c32
│   ├── liblua.c32
│   ├── libmenu.c32
│   ├── libutil.c32
│   ├── linux.c32
│   ├── ls.c32
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── grub2 [error opening dir]
├── initramfs-0-rescue-5d22e142e5784b8f8e146fdc1c4e4eec.img
├── initramfs-5.3.7-301.fc31.x86_64.img
├── initramfs-5.4.12-200.fc31.x86_64.img
├── initramfs-5.4.8-200.fc31.x86_64.img
├── loader
│   └── entries [error opening dir]
├── lost+found [error opening dir]
├── memtest86+-5.01
├── System.map-5.3.7-301.fc31.x86_64
├── System.map-5.4.12-200.fc31.x86_64
├── System.map-5.4.8-200.fc31.x86_64
├── vmlinuz-0-rescue-5d22e142e5784b8f8e146fdc1c4e4eec
├── vmlinuz-5.3.7-301.fc31.x86_64
├── vmlinuz-5.4.12-200.fc31.x86_64
└── vmlinuz-5.4.8-200.fc31.x86_64

6 directories, 77 files
```

Please answer the following questions:

- Operating system and version: Fedora 31 Workstation
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Part of distribution
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? Yes

superm1, 2020-01-24:
Probably a duplicate of #1712

joakim-tjernlund, 2020-01-27:
> Probably a duplicate of #1712

Don't think so, he is using gusb 0.3.1

jacobemcken, 2020-01-27:
I am also using Fedora 31 and found this issue because I was seeing often seeing a "very high" load from `fwupd`. "Very high" translates into consuming around 2-3 % over 30 secs using `top` to monitor the processes.

Let me know if you want a similar report as the one davewthompson did.

superm1, 2020-01-29:
Would it be possible to attach a debugger during that time and grab a backtrace on what it's doing?

davewthompson, 2020-01-29:
More than happy to. If you can provide me the commands to run, I'll do it for you tonight (I'm in NY, so EST timezone)

superm1, 2020-01-29:
Get the debug symbols installed if you don't already have them installed.  Then you'll use gdb's attach feature to attach to the already running daemon.

Some sample stuff about how to do it is here: https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/developer_guide/debugging-running-application#inspecting-application-state-gdb_attaching-gdb

Then from that run `bt` in the gdb prompt to get a backtrace of what's currently going on.

mykaul, 2020-01-30:
On my F31, the thread GUsbEventThread is the one taking all the CPU:
```
   4886 root      20   0  578036  35704  20228 S   2.0   0.2  47:06.25 GUsbEventThread                                                                                                                                                        
   4882 root      20   0  578036  35704  20228 S   0.0   0.2   0:03.34 fwupd                                                                                                                                                                  
   4884 root      20   0  578036  35704  20228 S   0.0   0.2   0:01.92 gmain                                                                                                                                                                  
   4885 root      20   0  578036  35704  20228 S   0.0   0.2   0:00.03 fwupd                                                                                                                                                                  
   4887 root      20   0  578036  35704  20228 S   0.0   0.2   0:00.12 gdbus   
```

And all it is doing is polling:
```
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
poll([{fd=12, events=POLLIN}, {fd=14, events=POLLIN}], 2, 1) = 0 (Timeout)
```
And here's what lsof says:
```
fwupd        4882    4886 GUsbEvent    root    0r      CHR                1,3       0t0       2063 /dev/null
fwupd        4882    4886 GUsbEvent    root    1u     unix 0x0000000001a253b3       0t0      73740 type=STREAM
fwupd        4882    4886 GUsbEvent    root    2u     unix 0x0000000001a253b3       0t0      73740 type=STREAM
fwupd        4882    4886 GUsbEvent    root    3r      REG              253,1   8406312    1574534 /var/lib/sss/mc/passwd
fwupd        4882    4886 GUsbEvent    root    4u     unix 0x00000000228e10bf       0t0      72590 type=STREAM
fwupd        4882    4886 GUsbEvent    root    5u  a_inode               0,14         0      12357 [eventfd]
fwupd        4882    4886 GUsbEvent    root    6r  a_inode               0,14         0      12357 inotify
fwupd        4882    4886 GUsbEvent    root    7u  a_inode               0,14         0      12357 [eventfd]
fwupd        4882    4886 GUsbEvent    root    8u      REG              253,1     32768    1445371 /var/lib/fwupd/pending.db
fwupd        4882    4886 GUsbEvent    root    9u  netlink                          0t0      72591 KOBJECT_UEVENT
fwupd        4882    4886 GUsbEvent    root   10r     FIFO               0,13       0t0      72592 pipe
fwupd        4882    4886 GUsbEvent    root   11w     FIFO               0,13       0t0      72592 pipe
fwupd        4882    4886 GUsbEvent    root   12r     FIFO               0,13       0t0      72593 pipe
fwupd        4882    4886 GUsbEvent    root   13w     FIFO               0,13       0t0      72593 pipe
fwupd        4882    4886 GUsbEvent    root   14u  a_inode               0,14         0      12357 [timerfd]
fwupd        4882    4886 GUsbEvent    root   15u  netlink                          0t0      73749 KOBJECT_UEVENT
fwupd        4882    4886 GUsbEvent    root   16u  netlink                          0t0      73747 KOBJECT_UEVENT
fwupd        4882    4886 GUsbEvent    root   17u  netlink                          0t0      73748 KOBJECT_UEVENT
fwupd        4882    4886 GUsbEvent    root   18u     unix 0x000000004aca4614       0t0      73750 type=STREAM
fwupd        4882    4886 GUsbEvent    root   19u  a_inode               0,14         0      12357 [eventfd]
```

mykaul, 2020-01-30:
Could it be https://github.com/hughsie/libgusb/issues/24 ?
In my case, I am seeing the thread GUsbEventThread consuming that CPU usage.

joakim-tjernlund, 2020-01-30:
> Could it be [hughsie/libgusb#24](https://github.com/hughsie/libgusb/issues/24) ?
> In my case, I am seeing the thread GUsbEventThread consuming that CPU usage.

It sure looks the same, maybe his gusb(0.3.1) is patched with 
https://github.com/hughsie/libgusb/commit/985b9e4ec39ac521e7398a8e3017c96f2c4d75ec ?

superm1, 2020-01-30:
Actually, I think the `fwupdmgr --version` output is misleading in this case.  It's a "Compile time" version, not a runtime version.  So that means that fwupd in F31 was compiled against 0.3.1, but  runtime library may have been upgraded to something later.

According to git, fedora 31 has had libgusb 0.3.2 for about a month: https://src.fedoraproject.org/rpms/libgusb/commits/f31

So I highly suspect it's actually 0.3.2 on the system.  You can check with package manager commands to confirm.

joakim-tjernlund, 2020-01-30:
feels like libgusb 0.3.3 is need real soon now ...

hughsie, 2020-01-30:
> feels like libgusb 0.3.3 is need real soon now ...

Planned for tomorrow.

superm1, 2020-01-30:
duplicate of #1712

closed 2020-01-30 by superm1

hughsie, 2020-01-30:
> Planned for tomorrow.

I've just done 0.3.3 upstream and am building for Fedora now.