fwupd/fwupd #1724: Disable autoupdate

m-czernek, 2020-01-26:
Hi,

Upon issuing `fwupdmgr get-devices` , I received  the following message:

```
Devices that have been updated successfully:

 • X1 Carbon Thunderbolt Controller (33.00 → 43.00)
```
Did fwupdmgr automatically update firmware without asking me for permission? If so, where can I turn off this behavior, and set fwupd to always ask me for permission?

```shell
$ fwupdmgr --version
client version:	1.2.12
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.2.12

```

hughsie, 2020-01-26:
We never autoupdate. Did you do fwupdmgr update or click an update in gnome software perhaps?

m-czernek, 2020-01-27:
Strange. I don't typically touch gnome software, but it is possible I accepted the update somewhere, not paying attention.

Could I have accepted the update using dnf?

hughsie, 2020-01-27:
> Could I have accepted the update using dnf

No, although if you're using another desktop to GNOME that might have a client that's doing autoupdates...

m-czernek, 2020-01-27:
Mm. I am using GNOME, but I cannot say with 100% certainty that I didn't touch any updates in the gnome center. I'm leaning towards an error on my part.

Thank you for the discussion! I'll be more careful and test it next time to see if I can replicate the behavior. Closing for now.

closed 2020-01-27 by m-czernek