fwupd/fwupd #1725: Unable to run fwupdmgr on Ubuntu 19.10 due to bad client version

afdta, 2020-01-28:
In the past I've been successful following the basic usage guidelines outlined here: https://github.com/fwupd/fwupd#basic-usage-flow-command-line

But now I get an error message: 

    sudo fwupdmgr get-devices
    Unsupported daemon version 1.3.6, client version is 1.2.10. 

I'm running Ubuntu 19.10 and have installed the latest fwupd from the Snap store.

Here's the question I posted on StackOverflow: https://stackoverflow.com/questions/59952052/what-do-i-need-to-install-to-use-fwupd-to-update-my-computers-firmware

hughsie, 2020-01-29:
If you're using the snap version and the distro version, you need to use `fwupd.fwupdmgr get-devices` -- i know it's insane, but it's seemingly a limitation of snap.

closed 2020-01-29 by hughsie

superm1, 2020-01-29:
You didn't remove the Deb package first.

You can use fwupd.fwupdmgr or remove the Deb package and reinstall snap.

https://github.com/fwupd/fwupd/wiki/fwupd-snap

superm1, 2020-01-29:
Specifically it's because of the order of PATH by default. `/usr/bin` comes before `/snap/bin`.