fwupd/fwupd #1726: Why are the firmware from the "https://github.com/Logitech/fw_updates/tree/update2019-08-27" branch, not in the LVFS repository?

BlakeCLewis, 2020-01-29:
Why are the firmware from the "https://github.com/Logitech/fw_updates/tree/update2019-08-27" branch, not in the LVFS repository?

I spent the evening figuring out why my new U0007 Unifying Receiver would not update.

U0007 with bootloader BOT01.02 would update, but U0007 with bootloader BOT01.04 would not.
-- No upgrades for Unifying Receiver: No releases found: Bootloader is not compatible

Logitech/fw_updates has a branch: 'update2019-08-27'  that has an update for bootloader 1.04 (fw version RQR12.11.B0032) at "https://github.com/Logitech/fw_updates/tree/update2019-08-27/RQR12"
I tried version of fwupdmgr below and also the snap(also 1.3.6), but neither found 12.11.B0032  which is 5 months old.

--- this worked ---
dnf install gcab
git clone git@github.com:Logitech/fw_updates.git
cd fw_updates
git checkout update2019-08-27
cd RQR12/RQR12.11/lvfs/
make all

-- put it into bootloader mode, but failed firmware upgrade
fwupdmgr install Logitech-Unifying-RQR12.11_B0032.cab
```
Decompressing…           [***************************************]
Authenticating…          [***************************************]
failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7
```
--- ran a second time with success
fwupdmgr install Logitech-Unifying-RQR12.11_B0032.cab
```
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Installing on Unifying Receiver…                                 ]
Installing on Unifying Receiver…*********************************]
Writing…                 [***************************************]
Successfully installed firmware
```
fwupdmgr get-devices
```
Unifying Receiver:
      Device ID:           8758d19fe8acce5a4c658b348e6f93ff1291f34a
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.11_B0032
      Bootloader Version:  BOT01.04_B0016
      Vendor:              Logitech, Inc. (USB:0x046D)
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server
```

---- configuration  and pre updated Unifying Receiver ---
```
client version: 1.3.6
compile-time dependency versions
        gusb:   0.3.1
        efivar: 37
daemon version: 1.3.6
```
```
Unifying Receiver:
      Device ID:           8758d19fe8acce5a4c658b348e6f93ff1291f34a
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.09_B0030
      Bootloader Version:  BOT01.04_B0016
      Vendor:              Logitech, Inc. (USB:0x046D)
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                           279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:        • Updatable
                           • Supported on remote server
```

- Operating system and version: Fedora 31
- How did you install: 'fwupd dnf install fwupd', also tried snap
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes: SK hynix BC501
- Is secure boot enabled (only for the UEFI plugin)? yes

49616e, 2020-01-29:
I have the same problem. There's a U-0008 with BOT03.01_B0008 and a U-0007 with BOT01.04_B0016 that can't be updated by fwupd right now because of incompatible bootloaders. Having the appropriate fw blobs available in LVFS would certainly be nice.

superm1, 2020-01-30:
These are not fwupd client or daemon issues, they're missing firmwares on LVFS.
You should report these issues to https://github.com/Logitech/fw_updates/issues.

closed 2020-01-30 by superm1