fwupd/fwupd #1729: OverrideESPMountPoint has an impact on detected devices

x70b1, 2020-01-29:
I use EFISTUB without a bootloader. Now I have tried to get `fwupd` running.

It seems to work if I change `OverrideESPMountPoint` in `/etc/fwupd/uefi.conf` to `OverrideESPMountPoint=/boot`. But the output of the identified devices changes. 

Actually I did not find a doku for EFISTUB and I am not sure if this is supported. 
But I see no reason why this should not be possible.

```shell
$ fwupdmgr --version
client version:	1.3.6-64-g2a026147
compile-time dependency versions
	gusb:	0.3.2
	efivar:	37
daemon version:	1.3.6-64-g2a026147
```
```shell
$ tree /boot
/boot
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
└── vmlinuz-linux

0 directories, 4 files
```
- Operating system and version: Arch Linux
- How did you install fwupd: AUR/fwupd-git
- Have you tried rebooting? Yes

Default config with all devices:
```shell
$ fwupdmgr get-devices
20QD00M4GE
│
├─Thunderbolt Controller:
│     Device ID:           a07783a8b146d42d35d52ddf8d3e3e2bc00d4fc2
│     Zusammenfassung:     Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               c7920601-0cda-507f-851e-92229eb1d370 ← TBT-01091801-native
│                          70931c8b-ee82-52c9-bc3d-f85e64379bf9 ← TBT-01091801-native-0000:07:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│
├─System Firmware:
│     Device ID:           f09b748a98b31bb5c7c64d83f24ddbc80dff7c5f
│     Current version:     0.1.23
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     Update State:        failed
│     Update Error:        Failed to create '/boot/EFI/arch/fw': Das Dateisystem ist nur lesbar
│     Last modified:       2020-01-29 09:52
│     GUIDs:               4bbc40fa-f81e-4206-bc70-a1f7b744d964
│                          230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│                          171d8529-50f6-550b-9db6-7f21af7414ef ← UEFI\RES_{4BBC40FA-F81E-4206-BC70-A1F7B744D964}
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│
├─Touchpad:
│     Device ID:           91961e6ddc950eb21e71db6f33ec09a133933a95
│     Current version:     1.3.3013412
│     Bootloader Version:  8.0
│     Vendor:              Synaptics (HIDRAW:0x06CB)
│     GUIDs:               10010313-27c0-5c7f-bd99-955028fd3a01 ← HIDRAW\VEN_06CB&DEV_CD8C&REV_00
│                          71463ccd-7553-5d7d-9fac-7ed8073a7853 ← HIDRAW\VEN_06CB&DEV_CD8C
│                          981824ee-4283-595b-a2e4-46eae92dba40 ← SYNAPTICS_RMI\TM3468-010
│                          0957b08b-b580-5c3a-8073-98b9bb15bc7f ← SYNAPTICS_RMI\TM3468
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Supported on remote server
│
├─UEFI Device Firmware:
│     Device ID:           0ec6a2d51c2c55a649247d119ffb8e229168d6c9
│     Current version:     192.47.1524
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                c3e4be53-e714-4ea1-bb9c-7fe13a98b556
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
├─UEFI Device Firmware:
│     Device ID:           c085a3cbf4dc1d60d110b7616816e2ffd279b64e
│     Current version:     0.1.10
│     Minimum Version:     0.1.10
│     Vendor:              DMI:LENOVO
│     GUID:                f72e048b-65bd-4e71-9071-1ac7045223e5
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
├─UEFI Device Firmware:
│     Device ID:           b596f8dec94a959248534b35641fe2341e27ed06
│     Current version:     1.0.8343
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                9045ae09-34ab-46d6-826d-b02f859dd20b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
├─UEFI Device Firmware:
│     Device ID:           97d441c3228fe8615f883b166a8809389803f9e2
│     Current version:     0.0.26403
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                68d3b08f-6ed2-43ca-86f0-f1ad6e41b390
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
└─WDC PC SN730 SDBQNTY-512G-1001:
      Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
      Zusammenfassung:     NVM Express Solid State Drive
      Current version:     11110101
      Vendor:              Sandisk Corp (NVME:0x15B7)
      Serial Number:       19360N454610
      GUIDs:               a39943dd-3afb-54f8-b110-c5a21f071200 ← NVME\VEN_15B7&DEV_5006&REV_00
                           fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2 ← NVME\VEN_15B7&DEV_5006
                           06b4e2aa-91af-508b-b06e-65e3b3189e97 ← WDC PC SN730 SDBQNTY-512G-1001
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

Changed config with less devices detected:
```shell
$ fwupdmgr get-devices
20QD00M4GE
│
├─Thunderbolt Controller:
│     Device ID:           a07783a8b146d42d35d52ddf8d3e3e2bc00d4fc2
│     Zusammenfassung:     Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               c7920601-0cda-507f-851e-92229eb1d370 ← TBT-01091801-native
│                          70931c8b-ee82-52c9-bc3d-f85e64379bf9 ← TBT-01091801-native-0000:07:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│
├─Touchpad:
│     Device ID:           91961e6ddc950eb21e71db6f33ec09a133933a95
│     Current version:     1.3.3013412
│     Bootloader Version:  8.0
│     Vendor:              Synaptics (HIDRAW:0x06CB)
│     GUIDs:               10010313-27c0-5c7f-bd99-955028fd3a01 ← HIDRAW\VEN_06CB&DEV_CD8C&REV_00
│                          71463ccd-7553-5d7d-9fac-7ed8073a7853 ← HIDRAW\VEN_06CB&DEV_CD8C
│                          981824ee-4283-595b-a2e4-46eae92dba40 ← SYNAPTICS_RMI\TM3468-010
│                          0957b08b-b580-5c3a-8073-98b9bb15bc7f ← SYNAPTICS_RMI\TM3468
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Supported on remote server
│
└─WDC PC SN730 SDBQNTY-512G-1001:
      Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
      Zusammenfassung:     NVM Express Solid State Drive
      Current version:     11110101
      Vendor:              Sandisk Corp (NVME:0x15B7)
      Serial Number:       19360N454610
      GUIDs:               a39943dd-3afb-54f8-b110-c5a21f071200 ← NVME\VEN_15B7&DEV_5006&REV_00
                           fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2 ← NVME\VEN_15B7&DEV_5006
                           06b4e2aa-91af-508b-b06e-65e3b3189e97 ← WDC PC SN730 SDBQNTY-512G-1001
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

superm1, 2020-01-29:
Is your EFI system partition mounted read only like the error says?

x70b1, 2020-01-29:
You are right, I get this error if I run `# fwupdmgr update` with the default config. 
But it is read+write mounted:
```
# mount | grep /boot
/dev/nvme0n1p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,utf8,errors=remount-ro)
```

superm1, 2020-01-29:
I would say check and see if the systemd unit is missing /boot in RW paths?

x70b1, 2020-01-29:
I override the systemd config:
```
systemctl show fwupd.service | grep ReadWritePaths
ReadWritePaths=-/boot/efi -/efi/EFI -/boot/EFI -/boot
```

The problem is the same: With `OverrideESPMountPoint=/boot` I have 3 devices, can write to /boot and all of them are up to date.
With the default config I have 8 devices, but can't write to /boot as its still read-only.

superm1, 2020-01-30:
You reloaded the daemon after doing that right?

By default `fwupd.service` has `ProtectSystem=full` set which will make `/boot` read only.  Adding it to `ReadWritePaths` I would have expected to override this as that's what the systemd documentation says.

> 
> ProtectSystem=
>
> Takes a boolean argument or the special values "full" or "strict". If true, mounts the /usr and /boot directories read-only for processes invoked by this unit. If set to "full", the /etc directory is mounted read-only, too. If set to "strict" the entire file system hierarchy is mounted read-only, except for the API file system subtrees /dev, /proc and /sys (protect these directories using PrivateDevices=, ProtectKernelTunables=, ProtectControlGroups=). This setting ensures that any modification of the vendor-supplied operating system (and optionally its configuration, and local mounts) is prohibited for the service. It is recommended to enable this setting for all long-running services, unless they are involved with system updates or need to modify the operating system in other ways. If this option is used, ReadWritePaths= may be used to exclude specific directories from being made read-only. This setting is implied if DynamicUser= is set. This setting cannot ensure protection in all cases. In general it has the same limitations as ReadOnlyPaths=, see below. Defaults to off.
> 
> 

https://www.freedesktop.org/software/systemd/man/systemd.exec.html

So I would say try to remove the ProtectSystem attribute from the unit.  If you find that helps, this sounds like a systemd bug and I would recommend [reporting it to them](https://github.com/systemd/systemd/issues).

x70b1, 2020-02-02:
For me the first question is why the output of `get-devices` is different.
This has nothing to do with a denied write permission to /boot.

But yes, your idea with that systemd-bug could be true. 

I am not 100% sure what my workaround was but I guess I used the untouched `uefi.conf` (otherwise not all my updates are found), created the path `/boot/EFI/arch/fw` with mkdir and everything worked.

superm1, 2020-02-03:
Regarding your question to why the devices are different you can try adding `--show-all-devices` to your command.  Some devices might be getting filtered because they're not actually updatable?

Alternatively if that doesn't tell you, you can check the journal for `fwupd.service` for hints.  If that still doesn't tell you, change the daemon to run in verbose mode (modify `/etc/fwupd/daemon.conf`) restart it, and check the journal for hints.

superm1, 2020-02-03:
> created the path /boot/EFI/arch/fw with mkdir and everything worked.

Do you mean you manually created it?  Normally fwupd will do this for you, but I suspect this may have been due to the order of events with `ProtectSystem` attribute not letting you make the directory `/boot/EFI` from `fwupd.service`.

We might need to specially flag this case for the error message to tell you to manually create it if so.

x70b1, 2020-02-04:
The output with `fwupdmgr get-devices --show-all-devices` and `OverrideESPMountPoint=/boot` in `uefi.conf`:
```
20QD00M4GE
│
├─Thunderbolt Controller:
│     Device ID:           a07783a8b146d42d35d52ddf8d3e3e2bc00d4fc2
│     Zusammenfassung:     Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               c7920601-0cda-507f-851e-92229eb1d370 ← TBT-01091801-native
│                          70931c8b-ee82-52c9-bc3d-f85e64379bf9 ← TBT-01091801-native-0000:07:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:     73.8.17568.5511
│ │   Vendor:              ST Microelectronics (TPM:STM)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
│ │                        84df3581-f896-54d2-bd1a-372602f04c32 ← TPM\VEN_STM&DEV_0001
│ │                        bfaed10a-bbc1-525b-a329-35da2f63e918 ← TPM\VEN_STM&MOD_
│ │                        70b7b833-7e1a-550a-a291-b94a12d0f319 ← TPM\VEN_STM&DEV_0001&VER_2.0
│ │                        06f005e9-cb62-5d1a-82d9-13c534c53c48 ← TPM\VEN_STM&MOD_&VER_2.0
│ │   Device Flags:        • Internal device
│ │
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       Vendor:            ST Microelectronics (TPM:STM)
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff ← system-tpm-eventlog
│       Device Flags:      • Internal device
│
├─Touchpad:
│     Device ID:           91961e6ddc950eb21e71db6f33ec09a133933a95
│     Current version:     1.3.3013412
│     Bootloader Version:  8.0
│     Vendor:              Synaptics (HIDRAW:0x06CB)
│     GUIDs:               10010313-27c0-5c7f-bd99-955028fd3a01 ← HIDRAW\VEN_06CB&DEV_CD8C&REV_00
│                          71463ccd-7553-5d7d-9fac-7ed8073a7853 ← HIDRAW\VEN_06CB&DEV_CD8C
│                          981824ee-4283-595b-a2e4-46eae92dba40 ← SYNAPTICS_RMI\TM3468-010
│                          0957b08b-b580-5c3a-8073-98b9bb15bc7f ← SYNAPTICS_RMI\TM3468
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Supported on remote server
│
├─UHD Graphics 620 (Whiskey Lake):
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     02
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               6986c21a-0f3f-5538-bcef-fe4e8cda7ab7 ← PCI\VEN_8086&DEV_3EA0&REV_02
│                          52aa8ac7-c837-561f-af32-b7979cd2f6ed ← PCI\VEN_8086&DEV_3EA0
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│
└─WDC PC SN730 SDBQNTY-512G-1001:
      Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
      Zusammenfassung:     NVM Express Solid State Drive
      Current version:     11110101
      Vendor:              Sandisk Corp (NVME:0x15B7)
      Serial Number:       19360N454610
      GUIDs:               a39943dd-3afb-54f8-b110-c5a21f071200 ← NVME\VEN_15B7&DEV_5006&REV_00
                           fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2 ← NVME\VEN_15B7&DEV_5006
                           06b4e2aa-91af-508b-b06e-65e3b3189e97 ← WDC PC SN730 SDBQNTY-512G-1001
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update

```

You are right. There are more devices (like TPM). But things like `System Firmware` are still not there. That's strange.

Ich checked the verbose output of the daemon. What I found is this log line:
```
fwupd[48144]: 09:52:12:0111 FuEngine             disabling plugin because: failed to coldplug using uefi: invalid OverrideESPMountPoint=/boot specified in config: /boot/EFI does not exist
fwupd[48144]: 09:52:12:0112 FuEngine             using plugins: altos, ata, colorhug, csr, dell_dock, dfu, ebitdo, emmc, fastboot, jabra, logitech_hidpp, nitrokey, nvme, optionrom, rts54hid, rts54hub, solokey, steelseries, superio, synaptics_cxaudio, synaptics_mst, synaptics_prometheus, synaptics_rmi, thelio_io, thunderbolt, tpm, tpm_eventlog, upower, vli_usbhub, wacom_raw, wacom_usb, thunderbolt_power
```

If I use the untouched `uefi.conf` the plugin `uefi` is still enabled and is printed at the end of the 2nd line.

superm1, 2020-02-04:
Actually looking at the code, this is the expected behavior.  We helpfully left a comment here for our futureselves to explain it too.  It's entirely from the systemd sandboxing behavior.

https://github.com/fwupd/fwupd/blob/master/plugins/uefi/fu-uefi-common.c#L334

I've written up a wiki page summarizing this behavior that we can refer to in the future next time this comes up.  
https://github.com/fwupd/fwupd/wiki/UEFI-Plugin,-ESP-locations

I'd recommend if you can modify the Arch wiki page around fwupd and UEFI to reference that too.

closed 2020-02-04 by superm1