fwupd/fwupd #1730: fwupdate ignored on reboot

Martin-Fahrenberger, 2020-01-29:
Hi fwupd Team,

On my Lenovo X1 Yoga (Gen. 4) firmware updates (both automatic and with manually downloaded and supplied .cab files) are "successful" within 'fwupdmgr update' and the 'arch' directory is created within my /boot partition (alongside my arch_grub).
However the updates are ignored upon startup.
When I try to manually select 'arch' rather than 'arch_grub' on startup the screen flickers quickly and then returns me to the selection screen. (When I select arch_grub it starts up normally)

It is also possible that I m simply misusing the utility or that the issue lies withing my grub, however I could not find any solutions online.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version

client version:	1.3.6-64-g2a026147
compile-time dependency versions
	gusb:	0.3.2
	efivar:	37
daemon version:	1.3.6-64-g2a026147
```

```shell
$ fwupdmgr get-devices

20QGS01M00
│
├─Thunderbolt Controller:
│     Device ID:           563d5fd644dc4f08cbad97804000894dec202b18
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               c7920601-0cda-507f-851e-92229eb1d370
│                          70931c8b-ee82-52c9-bc3d-f85e64379bf9
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Prometheus:
│ │   Device ID:           51a5f0521bddd3e9d93388f831b2b5ffdf225d9a
│ │   Summary:             Fingerprint reader
│ │   Current version:     10.01.2959683
│ │   Vendor:              Synaptics (USB:0x06CB)
│ │   Install Duration:    2 seconds
│ │   GUIDs:               09e8bf16-3e69-50f5-bb66-c7a040248352
│ │                        8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │   Device Flags:        • Updatable
│ │                        • Cryptographic hash verification is available
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:         a717531e28f0288a4e7af638b8e6afc0a1c8743e
│       Current version:   0016
│       Minimum Version:   0016
│       Vendor:            Synaptics (USB:0x06CB)
│       GUID:              7c5a1e62-38fa-5859-9337-09dbac6377e4
│       Device Flags:      • Updatable
│     
├─SAMSUNG MZVLB1T0HBLR-000L7:
│     Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:             NVM Express Solid State Drive
│     Current version:     3L2QEXF7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          b77af583-f45c-537c-8584-fc5d1c80fa20
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
[... 456 lines cut ...]
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initramfs-linux-fallback.img
├── initramfs-linux.img
└── vmlinuz-linux

9 directories, 321 files

```

Please answer the following questions:

- Operating system and version: 
Arch 5.4.15
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
I installed fwupd-git using the yay helper for the Arch User Repositories
- Have you tried rebooting?
Yes
- Are you using an NVMe disk?
Yes
- Is secure boot enabled (only for the UEFI plugin)?
No

superm1, 2020-01-29:
If I was going to guess; you probably have boot order locked in BIOS setup

Martin-Fahrenberger, 2020-01-30:
Your guess was right ^^ This is a bit embarrassing... ;) Thanks for your help!

closed 2020-01-30 by Martin-Fahrenberger

hughsie, 2020-01-30:
@superm1 is there anything we can do to detect this? e.g. parse the `Setup` UEFI key?

superm1, 2020-01-30:
Gotta talk to lenovo to see how they export it. I can tell you how Dell would export this type of thing if we did it, but it's definitely specific to us.

hughsie, 2020-01-30:
> I can tell you how Dell would export this type of thing if we did it

That would be interesting indeed!

> but it's definitely specific to us.

Urgh, I kinda hoped it would be AMI/Phoenix specific.

superm1, 2020-01-30:
> That would be interesting indeed

We don't offer boot order locking, I'm just saying if we did I could tell you how :)

> Urgh, I kinda hoped it would be AMI/Phoenix specific.

No, Dell doesn't use the general purpose AMI core.  Some systems have some components from AMI, but not the full core.

superm1, 2020-01-30:
@hughsie If lenovo won't offer you a way to query that BIOS setup option in Linux I guess something that can be done is to analyze the symptoms of the behavior and offer a hint at cause of the failure across the reboot if:
* same firmware version that it started with across the reboot
* `last_attempt_status` shows zero
*  it's a lenovo system

hughsie, 2020-01-30:
I've asked Lenovo for the appropriate course of action. I don't really want to use heuristics if there's something concrete we can measure.

superm1, 2020-02-21:
@mrhpearson, I saw you interacting on platform-drivers-x86 mailing list around some lenovo stuff.  Can you try to get an interface made available to userland on lenovo machines that tells whether boot order is locked or not?  RO interface is fine, just something that when fwupd is running on a lenovo machine can discover if that bit is set from setup and not offer updates.

mrhpearson, 2020-02-21:
Hi Mario,
Thanks for pointing me at the thread. A agree - it sounds like something we should provide; I'll go and have a look into what we can do. 
Mark.