fwupd/fwupd #1733: 1.3.7: flashrom plugin buid fails

kloczek, 2020-01-31:
I'm using flashrom 1.1 + two meson build patches from from git and flashrom plugin build fails with:
```
[303/463] gcc -Iplugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha -Iplugins/flashrom -I../plugins/flashrom -I. -I../ -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -I/usr/include/libxmlb-1 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/libsoup-2.4 -I/usr/include/gudev-1.0 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -std=c99 -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-cast-function-type -Wno-address-of-packed-member -Wno-unknown-pragmas -Wno-deprecated-declarations -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -fstack-protector-strong -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_BSD_SOURCE -D_XOPEN_SOURCE=700 '-DMM_REQUIRED_VERSION="1.10.0"' '-DQMI_REQUIRED_VERSION="1.23.1"' -O2 -g -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none -Os -fPIC -pthread '-DG_LOG_DOMAIN="FuPluginFlashrom"' '-DLOCALSTATEDIR="/var"' -MD -MQ 'plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o' -MF 'plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o.d' -o 'plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o' -c ../plugins/flashrom/fu-plugin-flashrom.c
FAILED: plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o
gcc -Iplugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha -Iplugins/flashrom -I../plugins/flashrom -I. -I../ -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -I/usr/include/libxmlb-1 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/libsoup-2.4 -I/usr/include/gudev-1.0 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -std=c99 -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-cast-function-type -Wno-address-of-packed-member -Wno-unknown-pragmas -Wno-deprecated-declarations -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -fstack-protector-strong -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_BSD_SOURCE -D_XOPEN_SOURCE=700 '-DMM_REQUIRED_VERSION="1.10.0"' '-DQMI_REQUIRED_VERSION="1.23.1"' -O2 -g -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none -Os -fPIC -pthread '-DG_LOG_DOMAIN="FuPluginFlashrom"' '-DLOCALSTATEDIR="/var"' -MD -MQ 'plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o' -MF 'plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o.d' -o 'plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o' -c ../plugins/flashrom/fu-plugin-flashrom.c
In file included from ../plugins/flashrom/fu-plugin-flashrom.c:28:
/usr/include/libflashrom.h:57:72: error: unknown type name 'bool'
   57 | void flashrom_flag_set(struct flashrom_flashctx *, enum flashrom_flag, bool value);
      |                                                                        ^~~~
/usr/include/libflashrom.h:58:1: error: unknown type name 'bool'
   58 | bool flashrom_flag_get(const struct flashrom_flashctx *, enum flashrom_flag);
      | ^~~~
/usr/include/libflashrom.h:69:37: error: unknown type name 'uint8_t'
   69 |   struct flashrom_flashctx *, const uint8_t *buf, size_t len);
      |                                     ^~~~~~~
ninja: build stopped: subcommand failed.
```

superm1, 2020-01-31:
What patches are you including?  You probably need:

https://github.com/flashrom/flashrom/commit/40f0757750f246bd78981fb7c02aadf1d47b18e8

kloczek, 2020-01-31:
At the moment with flashrom 1.1 I'm using 
https://review.coreboot.org/cgit/flashrom.git/patch/?id=cb973683
https://review.coreboot.org/cgit/flashrom.git/patch/?id=93db6e16

Will try to add patch which you've mention.

closed 2020-02-01 by superm1

kloczek, 2020-02-01:
flashron with https://github.com/flashrom/flashrom/commit/40f0757750f246bd78981fb7c02aadf1d47b18e8 does not build
```
[76/150] gcc  -o libflashrom.so.1.0.0 'flashrom@sha/usbdev.c.o' 'flashrom@sha/pcidev.c.o' 'flashrom@sha/atavia.c.o' 'flashrom@sha/buspirate_spi.c.o' 'flashrom@sha/ch341a_spi.c.o' 'flashrom@sha/dediprog.c.o' 'flashrom@sha/developerbox_spi.c.o' 'flashrom@sha/digilent_spi.c.o' 'flashrom@sha/drkaiser.c.o' 'flashrom@sha/dummyflasher.c.o' 'flashrom@sha/ft2232_spi.c.o' 'flashrom@sha/gfxnvidia.c.o' 'flashrom@sha/board_enable.c.o' 'flashrom@sha/cbtable.c.o' 'flashrom@sha/chipset_enable.c.o' 'flashrom@sha/internal.c.o' 'flashrom@sha/processor_enable.c.o' 'flashrom@sha/amd_imc.c.o' 'flashrom@sha/dmi.c.o' 'flashrom@sha/ichspi.c.o' 'flashrom@sha/it85spi.c.o' 'flashrom@sha/it87spi.c.o' 'flashrom@sha/mcp6x_spi.c.o' 'flashrom@sha/sb600spi.c.o' 'flashrom@sha/wbsio_spi.c.o' 'flashrom@sha/it8212.c.o' 'flashrom@sha/linux_mtd.c.o' 'flashrom@sha/linux_spi.c.o' 'flashrom@sha/nic3com.c.o' 'flashrom@sha/nicintel.c.o' 'flashrom@sha/nicintel_eeprom.c.o' 'flashrom@sha/nicintel_spi.c.o' 'flashrom@sha/nicrealtek.c.o' 'flashrom@sha/ogp_spi.c.o' 'flashrom@sha/pony_spi.c.o' 'flashrom@sha/rayer_spi.c.o' 'flashrom@sha/satamv.c.o' 'flashrom@sha/satasii.c.o' 'flashrom@sha/serprog.c.o' 'flashrom@sha/usbblaster_spi.c.o' 'flashrom@sha/bitbang_spi.c.o' 'flashrom@sha/hwaccess.c.o' 'flashrom@sha/physmap.c.o' 'flashrom@sha/custom_baud.c.o' 'flashrom@sha/serial.c.o' 'flashrom@sha/82802ab.c.o' 'flashrom@sha/at45db.c.o' 'flashrom@sha/edi.c.o' 'flashrom@sha/en29lv640b.c.o' 'flashrom@sha/flashchips.c.o' 'flashrom@sha/flashrom.c.o' 'flashrom@sha/fmap.c.o' 'flashrom@sha/helpers.c.o' 'flashrom@sha/ich_descriptors.c.o' 'flashrom@sha/jedec.c.o' 'flashrom@sha/layout.c.o' 'flashrom@sha/libflashrom.c.o' 'flashrom@sha/opaque.c.o' 'flashrom@sha/print.c.o' 'flashrom@sha/programmer.c.o' 'flashrom@sha/sfdp.c.o' 'flashrom@sha/spi25.c.o' 'flashrom@sha/spi25_statusreg.c.o' 'flashrom@sha/spi.c.o' 'flashrom@sha/sst28sf040.c.o' 'flashrom@sha/sst49lfxxxc.c.o' 'flashrom@sha/sst_fwhub.c.o' 'flashrom@sha/stm50.c.o' 'flashrom@sha/udelay.c.o' 'flashrom@sha/w29ee011.c.o' 'flashrom@sha/w39.c.o' -Wl,--as-needed -Wl,--no-undefined -shared -fPIC -Wl,--start-group -Wl,-soname,libflashrom.so.1 -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none -Os -Wl,-z,relro -Wl,--as-needed -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -flto=auto -flto-partition=none -fuse-linker-plugin -Wl,--version-script,/home/tkloczko/rpmbuild/BUILD/flashrom-1.1/libflashrom.map /usr/lib64/libusb-1.0.so /usr/lib64/libpci.so /usr/lib64/libftdi1.so -Wl,--end-group
FAILED: libflashrom.so.1.0.0
gcc  -o libflashrom.so.1.0.0 'flashrom@sha/usbdev.c.o' 'flashrom@sha/pcidev.c.o' 'flashrom@sha/atavia.c.o' 'flashrom@sha/buspirate_spi.c.o' 'flashrom@sha/ch341a_spi.c.o' 'flashrom@sha/dediprog.c.o' 'flashrom@sha/developerbox_spi.c.o' 'flashrom@sha/digilent_spi.c.o' 'flashrom@sha/drkaiser.c.o' 'flashrom@sha/dummyflasher.c.o' 'flashrom@sha/ft2232_spi.c.o' 'flashrom@sha/gfxnvidia.c.o' 'flashrom@sha/board_enable.c.o' 'flashrom@sha/cbtable.c.o' 'flashrom@sha/chipset_enable.c.o' 'flashrom@sha/internal.c.o' 'flashrom@sha/processor_enable.c.o' 'flashrom@sha/amd_imc.c.o' 'flashrom@sha/dmi.c.o' 'flashrom@sha/ichspi.c.o' 'flashrom@sha/it85spi.c.o' 'flashrom@sha/it87spi.c.o' 'flashrom@sha/mcp6x_spi.c.o' 'flashrom@sha/sb600spi.c.o' 'flashrom@sha/wbsio_spi.c.o' 'flashrom@sha/it8212.c.o' 'flashrom@sha/linux_mtd.c.o' 'flashrom@sha/linux_spi.c.o' 'flashrom@sha/nic3com.c.o' 'flashrom@sha/nicintel.c.o' 'flashrom@sha/nicintel_eeprom.c.o' 'flashrom@sha/nicintel_spi.c.o' 'flashrom@sha/nicrealtek.c.o' 'flashrom@sha/ogp_spi.c.o' 'flashrom@sha/pony_spi.c.o' 'flashrom@sha/rayer_spi.c.o' 'flashrom@sha/satamv.c.o' 'flashrom@sha/satasii.c.o' 'flashrom@sha/serprog.c.o' 'flashrom@sha/usbblaster_spi.c.o' 'flashrom@sha/bitbang_spi.c.o' 'flashrom@sha/hwaccess.c.o' 'flashrom@sha/physmap.c.o' 'flashrom@sha/custom_baud.c.o' 'flashrom@sha/serial.c.o' 'flashrom@sha/82802ab.c.o' 'flashrom@sha/at45db.c.o' 'flashrom@sha/edi.c.o' 'flashrom@sha/en29lv640b.c.o' 'flashrom@sha/flashchips.c.o' 'flashrom@sha/flashrom.c.o' 'flashrom@sha/fmap.c.o' 'flashrom@sha/helpers.c.o' 'flashrom@sha/ich_descriptors.c.o' 'flashrom@sha/jedec.c.o' 'flashrom@sha/layout.c.o' 'flashrom@sha/libflashrom.c.o' 'flashrom@sha/opaque.c.o' 'flashrom@sha/print.c.o' 'flashrom@sha/programmer.c.o' 'flashrom@sha/sfdp.c.o' 'flashrom@sha/spi25.c.o' 'flashrom@sha/spi25_statusreg.c.o' 'flashrom@sha/spi.c.o' 'flashrom@sha/sst28sf040.c.o' 'flashrom@sha/sst49lfxxxc.c.o' 'flashrom@sha/sst_fwhub.c.o' 'flashrom@sha/stm50.c.o' 'flashrom@sha/udelay.c.o' 'flashrom@sha/w29ee011.c.o' 'flashrom@sha/w39.c.o' -Wl,--as-needed -Wl,--no-undefined -shared -fPIC -Wl,--start-group -Wl,-soname,libflashrom.so.1 -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none -Os -Wl,-z,relro -Wl,--as-needed -Wl,-z,n [... cut ...] czko/rpmbuild/BUILD/flashrom-1.1/libflashrom.map /usr/lib64/libusb-1.0.so /usr/lib64/libpci.so /usr/lib64/libftdi1.so -Wl,--end-group
/usr/bin/ld: flashrom@sha/developerbox_spi.c.o (symbol from plugin): in function `usb_ctx':
(.text+0x0): multiple definition of `usb_ctx'; flashrom@sha/dediprog.c.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
error: Bad exit status from /var/tmp/rpm-tmp.6vILry (%build)
```

superm1, 2020-02-01:
Can you please try with flashrom master?

kloczek, 2020-02-01:
will try in ~1h .. my family is calling me to eaty something :)

kloczek, 2020-02-01:
Looks like fwupd builds correctly against master git flashrom (which requires small patch)

superm1, 2020-02-01:
Fwupd patch or flashrom patch? Can up please submit it?

kloczek, 2020-02-01:
Against flashrom.
Meson fails because missing entry in meson_options.txt
```
--- a/meson_options.txt~        2020-02-01 15:12:29.000000000 +0000
+++ b/meson_options.txt 2020-02-01 15:24:48.903121676 +0000
@@ -13,8 +13,8 @@
 option('config_dummy', type : 'boolean', value : true, description : 'dummy tracing')
 option('config_ft2232_spi', type : 'boolean', value : true, description : 'FT2232 SPI dongles')
 option('config_gfxnvidia', type : 'boolean', value : true, description : 'NVIDIA graphics cards')
-option('config_internal', type : 'boolean', value : true, description : 'internal/onboard')
 option('config_internal_dmi', type : 'boolean', value : true, description : 'Use internal DMI parser')
+option('config_internal', type : 'boolean', value : true, description : 'internal/onboard')
 option('config_it8212', type : 'boolean', value : true, description : 'ITE IT8212F PATA')
 option('config_linux_mtd', type : 'boolean', value : true, description : 'Linux MTD interfaces')
 option('config_linux_spi', type : 'boolean', value : true, description : 'Linux spidev interfaces')
@@ -32,4 +32,5 @@
 option('config_satamv', type : 'boolean', value : true, description : 'Marvell SATA controllers')
 option('config_satasii', type : 'boolean', value : true, description : 'SiI SATA controllers')
 option('config_serprog', type : 'boolean', value : true, description : 'serprog')
+option('config_stlinkv3_spi', type : 'boolean', value : true, description : '')
 option('config_usbblaster_spi', type : 'boolean', value : true, description : 'Altera USB-Blaster dongles')
```
above as well sorts options.

superm1, 2020-02-02:
Gotcha. Would you mind submitting a MR to their Gerrit instance?