fwupd/fwupd #1736: Build failure with Fedora 32

hughsie, 2020-02-01:
I have very little idea what's wrong :(

    [385/457] ld -o plugins/uefi/efi/fwup.so -T /usr/lib64/gnuefi/elf_x86_64_efi.lds -shared -Bsymbolic -nostdlib -znocombreloc -L /usr/lib64 /usr/lib64/gnuefi/crt0-efi-x86_64.o plugins/uefi/efi/fwupdate.o plugins/uefi/efi/fwup-debug.o plugins/uefi/efi/fwup-efi.o plugins/uefi/efi/fwup-common.o -lefi -lgnuefi /usr/lib/gcc/x86_64-redhat-linux/10/libgcc.a
FAILED: plugins/uefi/efi/fwup.so 
    ld -o plugins/uefi/efi/fwup.so -T /usr/lib64/gnuefi/elf_x86_64_efi.lds -shared -Bsymbolic -nostdlib -znocombreloc -L /usr/lib64 /usr/lib64/gnuefi/crt0-efi-x86_64.o plugins/uefi/efi/fwupdate.o plugins/uefi/efi/fwup-debug.o plugins/uefi/efi/fwup-efi.o plugins/uefi/efi/fwup-common.o -lefi -lgnuefi /usr/lib/gcc/x86_64-redhat-linux/10/libgcc.a
    ld: cannot find -lefi
    ld: cannot find -lgnuefi

https://kojipkgs.fedoraproject.org//work/tasks/8170/41298170/build.log

superm1, 2020-02-01:
Maybe the directory for `efi-libdir` is wrong for F32?

superm1, 2020-02-03:
Well there is a *similar* weird failure happening in Debian unstable and Ubuntu focal right now for a few architectures.

Might not be the exact same problem, but figure I should share in case it is.

superm1, 2020-02-03:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=950550

superm1, 2020-02-04:
The debian one ended up being a libgcc symlink/linker script bug.  So I guess not the same as what was happening for you in F32.  Did it get sorted out yet?

closed 2020-02-05 by superm1

tomasbw, 2020-02-09:
commit 7d11d44242b40a6201260baaba1ba375edcbadff breaks the build on fedora 30 
Reverting, fixes the issue: 

[513/630] Generating fwup.so with a custom command.
FAILED: plugins/uefi/efi/fwup.so
ld -o plugins/uefi/efi/fwup.so -T /usr/lib64/gnuefi/elf_x86_64_efi.lds -shared -Bsymbolic -nostdlib -znocombreloc -L /usr/lib64/gnuefi /usr/lib64/gnuefi/crt0-efi-x86_64.o plugins/uefi/efi/fwupdate.o plugins/uefi/efi/fwup-debug.o plugins/uefi/efi/fwup-efi.o plugins/uefi/efi/fwup-common.o -lefi -lgnuefi /usr/lib/gcc/x86_64-redhat-linux/9/libgcc.a
ld: cannot find -lefi
ld: cannot find -lgnuefi