fwupd/fwupd #1738: Selinux failure on Fedora 31

bepvte, 2020-02-01:
Bug caused by https://github.com/fwupd/fwupd/commit/d81ea2e3fc0de9cf73a63fcd9f7360e54e03ed31.

`avc:  denied  { mounton } for  pid=16412 comm="(fwupd)" path="/run/systemd/unit-root/run/motd.d" dev="tmpfs" ino=32552 scontext=system_u:system_r:init_t:s0 tcontext=system_u:object_r:pam_var_run_t:s0 tclass=dir permissive=0`

Interestingly, setroubleshootd tells me that the default label of "/run/systemd/unit-root/run/motd.d" is "init_var_run_t", and that I should reset it via "restorecon". However, the folder does not exist until the unit is started, so I cannot. I have also tried running "fixfiles". I'm guessing that systemd needs to know the selinux class that the folder should be set to? Or maybe there needs to be a prelaunch script.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.7
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.3.7
```

Please answer the following questions:

- Operating system and version: Fedora 31 (Workstation Edition)
- How did you install fwupd: fedora package manager, updates-testing repo
- Have you tried rebooting?: Yes
- Are you using an NVMe disk?: Yes
- Is secure boot enabled (only for the UEFI plugin)?: No

hughsie, 2020-02-02:
@superm1 is this expected? If so I can file an issue against selinux-policy upstream.

superm1, 2020-02-02:
I don't actually know. I'm not familiar with selinux.

superm1, 2020-02-02:
Maybe the right thing to do is check with systemd guys if this is an expected behavior for systemd given our fwupd.service.

bepvte, 2020-02-02:
It looks like the problem is that /run/motd.d is marked as only accessible to PAM. Systemd on fedora is not allowed to use `mounton` on it because of this. It needs to bind mount it into the sandboxed namespace that fwupd is in. It seems like this might not be your bug, but I will let you decide and close this if you want. I will file a bug on fedoras selinux policy.

```
Successfully mounted /run/systemd/inaccessible/dir to /run/systemd/unit-root/home
Applying namespace mount on /run/systemd/unit-root/proc
Applying namespace mount on /run/systemd/unit-root/root
Successfully mounted /run/systemd/inaccessible/dir to /run/systemd/unit-root/root
Applying namespace mount on /run/systemd/unit-root/run/motd.d
Followed source symlinks /run/motd.d → /run/motd.d.
Failed to mount /run/motd.d to /run/systemd/unit-root/run/motd.d: Permission denied
fwupd.service: Failed to set up mount namespacing: /run/systemd/unit-root/run/motd.d: Permission denied
fwupd.service: Failed at step NAMESPACE spawning /usr/libexec/fwupd/fwupd: Permission denied
```

closed 2020-02-02 by bepvte

superm1, 2020-02-03:
> I will file a bug on fedoras selinux policy.

Can you please drop a link here?  It would be good to follow along in case they think we should do something different.

bepvte, 2020-02-03:
https://bugzilla.redhat.com/show_bug.cgi?id=1797324