fwupd/fwupd #1739: `\EFI\ubuntu\fw\fwupdate-bc5p2v.cap': Not Found

fornellas, 2020-02-02:
I'm getting this when trying to update things:

[image]

I had to set BIOS settings to default, `rm -rf /boot/efi/*`, `grub-install`, `fwupdmgr update` and manually F12 to select firmware upgrade to finally get to this error.

It seems that it is pointing to the wrong path or that `fwupdate-bc5p2v.cap` is missing to be added:

```
# cat /sys/firmware/efi/efivars/fwupdate-fc5552dc-7882-409b-9b98-699361c74fc4-0-0abba7dc-e516-4167-bbf5-4d9d1c739416
�jJ\EFI\ubuntu\fw\fwupdate-bc5p2v.cap�
```

```shell
$ fwupdmgr --version
client version:	1.3.6
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.3.6
```

```shell
$ fwupdmgr get-devices
20LES05P0J
│
├─Thunderbolt Controller:
│     Device ID:           212023058330fa0e3ea5a95ad23f2312d4efc4a7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              LENOVO (TBT:0x0109)
│     GUIDs:               10bc8610-a3cd-5e74-a187-8d01f3ff345d ← TBT-01091705
│                          fb595a4f-7e77-5030-a518-e4494392378d ← TBT-01091705-0000:08:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─SAMSUNG MZVLB512HAJQ-000L7:
│     Device ID:           310f45f1f223064b5c16bf6dff31146755a64480
│     Summary:             NVM Express Solid State Drive
│     Current version:     5L2QEXA7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:       S3TNNX0M540033
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                          47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                          79d6cfae-a5a2-5936-9248-5aebd23480f7 ← SAMSUNG MZVLB512HAJQ-000L7
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           d3db6c5633d83a940028fc88d8102d1f1563e3a4
│     Current version:     0.1.35
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     Update State:        success
│     GUID:                fc5552dc-7882-409b-9b98-699361c74fc4
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           08b4314f260e5da5cb2a747a05521b60c0892523
[... 422 lines cut ...]
│       ├── videotest.mod
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-5.0.0-37-generic
├── lost+found
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.0.0-37-generic
└── vmlinuz-5.0.0-37-generic

13 directories, 357 files
```

- Operating system and version: Linux Mint 19.3 Tricia 
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `snap install fwupd`
- Have you tried rebooting? yes, and manually F12 to run the fw update
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? Secure boot is disabled on the BIOS.

superm1, 2020-02-04:
Hmm, so it is picking up the old `fwupdate-*` entry you have on your system.  We don't have a cleanup path for that right now from the fwupdate->fwupd transition.  Manually remove them, and I'll send up a PR for removing them automatically in the future.

closed 2020-02-04 by superm1

fornellas, 2020-02-05:
@superm1  manually remove what exactly? I already tried rm -rf `/boot/efi/*` and reset BIOS settings, did I miss something?

superm1, 2020-02-05:
/sys/firmware/efi/efivarfs/fwupdate-*

You will need to change the immutable bit and then remove those files.

fornellas, 2020-02-05:
It worked, thanks! Surprisingly efivars are not cleared by setting BIOS to default...

commit 8527405c (2020-02-04): Cleanup ancient fwupdate-* EFI variables too (Fixes: #1739)

It seems that fwupd.efi is picking up both fwupdate-* and fwupd-*
entries that are sitting around in efivarfs.