fwupd/fwupd #1751: Dell Precision 3530 BIOS Update Hangs in fwupdate after reboot

esmsnt, 2020-02-07:
Summary:
Attempt to install latest BIOS update on this system.
```
sudo fwupdmgr update
No upgrades for PCIe SSD: No releases found: no HWIDs matched b31aa89d-cfb8-5c67-8502-ce76a9e7b580
Upgrade available for System Firmware from 1.10.1 to 1.11.1
Precision 3530 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Downloading 1.11.1 for System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[ -                                     ]
Scheduling…              [***************************************]
Successfully installed firmware
An update requires a reboot to complete. Restart now? [y|N]: N
```
Reboot the system and it does reboot the fwupdate loader, debug below from setting: ```/usr/libexec/fwupd/fwupdate -d``` prior to reboot.
Output from ```/usr/libexec/fwupd/fwupdate -L```

```
../plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-ux-capsule../plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-479d7d09-78d8-4fb2-af14-ca6ecc80692d-0../plugins/uefi/efi/fwupdate.c:714:efi_main(): Adding new capsule../plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)/\EFI\fedora\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap...../plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000006479A7159307343884012C00)/HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C) and PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000006479A7159307343884012C00)/HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C) and Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000006479A7159307343884012C00)/HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C) and Pci(0x0,0x0)/Path(131,23,0x010000006479A7159307343884012C00)/HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C) and Path(131,23,0x010000006479A7159307343884012C00)/HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C) and HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\fedora\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap../plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device../plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume../plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file../plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): Read file; 67898 bytes../plugins/uefi/efi/fwupdate.c:548:fwup_add_update_capsule(): updates guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697../plugins/uefi/efi/fwupdate.c:549:fwup_add_update_capsule(): File guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697../plugins/uefi/efi/fwupdate.c:566:fwup_add_update_capsule(): Checking GOP for ux capsule../plugins/uefi/efi/fwupdate.c:714:efi_main(): Adding new capsule../plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)/\EFI\fedora\fw\fwupd-479d7d09-78d8-4fb2-af14-ca6ecc80692d.cap...../plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000006479A7159307343884012C00)/HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C) and PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000006479A7159307343884012C00)/HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C) and Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000006479A7159307343884012C00)/HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C) and Pci(0x0,0x0)/Path(131,23,0x010000006479A7159307343884012C00)/HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C) and Path(131,23,0x010000006479A7159307343884012C00)/HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C) and HD(Part1,SigE3EE1A1E-167F-4199-B224-D505718F747C)../plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\fedora\fw\fwupd-479d7d09-78d8-4fb2-af14-ca6ecc80692d.cap../plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device../plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume../plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file../plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): Read file; 12811172 bytes../plugins/uefi/efi/fwupdate.c:548:fwup_add_update_capsule(): updates guid: 479D7D09-78D8-4FB2-AF14-CA6ECC80692D../plugins/uefi/efi/fwupdate.c:549:fwup_add_update_capsule(): File guid: 479D7D09-78D8-4FB2-AF14-CA6ECC80692D../plugins/uefi/efi/fwupdate.c:729:efi_main(): n_updates: 2
``` 
At this point the fans go to 100% (laptop is plugged in) and it just stays there, waited approximately 30 minutes, no change.  I would expect it to reboot to the Dell logo and actually do the update.

fwupdate also shows that it was attempted:
```
sudo /usr/libexec/fwupd/fwupdate -i
Information for the update status entry 0:
  Information Version: 7
  Firmware GUID: {479d7d09-78d8-4fb2-af14-ca6ecc80692d}
  Capsule Flags: 0x00131072x
  Hardware Instance: 0
  Update Status: attempted
  Capsule File Path: /EFI/fedora/fw/fwupd-479d7d09-78d8-4fb2-af14-ca6ecc80692d.cap
```

Although it is not showing up in my fwupdmgr get-history I am sure that I used this same process to update to BIOS v1.10.1 months ago and it worked fine.

```shell
$ fwupdmgr --version
client version:	1.3.7
compile-time dependency versions
	gusb:	0.3.3
	efivar:	37
daemon version:	1.3.7
```

```shell
$ fwupdmgr get-devices
Precision 3530
│
├─PCIe SSD:
│     Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:             NVM Express Solid State Drive
│     Current version:     ECFM12.1
│     Vendor:              Phison Electronics Corporation (NVME:0x1987)
│     GUIDs:               8cb1e2fe-eb01-5508-9fb3-98add4bb7c34
│                          a44eb54c-5441-56f2-8cc0-5e48964c6457
│                          8d128eab-f266-513f-81e7-910de65fd73a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs shutdown after installation
│                          • Device is usable for the duration of the update
│   
└─System Firmware:
      Device ID:           6eaf130f9d1252a4cc3157471aa80fac80318068
      Current version:     1.10.1
      Minimum Version:     1.10.1
      Vendor:              Dell Inc. (DMI:Dell Inc.)
[... 18 lines cut ...]
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-479d7d09-78d8-4fb2-af14-ca6ecc80692d-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Fedora 31
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf
```
Package fwupd-1.3.7-2.fc31.x86_64 is already installed.
```
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? Secure Boot Disabled, TPM Disabled, Virtualization enabled.

superm1, 2020-02-07:
In this case, would you be able to try to use the EXE file from support.dell.com to see if it has the same behavior?  I expect it should, but a confirmation would be good.

1. Get a USB key formatted FAT32
2. Download EXE from support.dell.com to USB key
3. Turn off PC.
4. At POST, press F12, and select the option to flash new BIOS.
5. Select the BIOS from USB key.

esmsnt, 2020-02-07:
Just did that, selected the exe in the dell efi flashing utility and it is at the Dell logo showing flashing progress right now.  Will confirm when it is complete.  The fwupd process never makes it this far, it just stays at n_update: 2 (when debugging is turned on).

superm1, 2020-02-07:
Interesting.  Do you have any idea what version of fwupd you might have used in the past?  I wonder if we have some regression in how the capsule is getting built, or maybe the UX capsule is throwing stuff off?

esmsnt, 2020-02-07:
I am working on retracing my steps, from the BIOS log I can see that I updated this device three previous times. 5/8/2019, 6/27/2019, and 9/20/2019.

This laptop is one of my daily usage devices and it gets updated pretty much anytime software center notifies.  All of those updates would have been in Fedora 30.  I upgrade to Fedora 31 right after it came out, so that would have been late October, maybe Nov 1, 2019.

Based on this, I am thinking it would have been fwupd 1.2.9 or 1.2.10.
https://koschei.fedoraproject.org/package/fwupd?last_seen_ts=1578269065&collection=f30

commit d5bf8e45 (2020-02-10): Revert "Correctly delete UEFI variables"

This reverts commit 44f55e2ee6df0b317df3029675afce86219cd9cd.

This behavior caused fwupdx64.efi to loop for a very long time until
either aborting, running out of memory or some other problems.
Fixes: #1756
Fixes: #1751
closed 2020-02-10 by superm1