fwupd/fwupd #1752: Updating Altos ChaosKey fails with message `not a supported ELF format: no error`

chutz, 2020-02-08:
```shell
# fwupdmgr update
Upgrade available for Altos ChaosKey from 1.6.6 to 1.6.7
Altos ChaosKey and all connected devices may not be usable while updating. Continue with update? [Y|n]:
Downloading 1.6.7 for Altos ChaosKey...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
not a supported ELF format: no error
```

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.3.7
compile-time dependency versions
        gusb:   0.3.2
        efivar: 37
daemon version: 1.3.7
```

```shell
$ fwupdmgr get-devices
System Product Name
│
├─Altos ChaosKey:
│     Device ID:           b95c8c172d3bfebd1c631bc4dadc9d2e790d7f89
│     Summary:             A USB hardware random number generator
│     Current version:     1.6.6
│     Vendor:              altusmetrum.org (USB:0x1D50)
│     Update State:        failed
│     Update Error:        not a supported ELF format: no error
│     Last modified:       2020-02-08 07:24
│     GUIDs:               c08b0e2c-535e-5d88-8b5c-31f3086057d0
│                          7cfedbcb-3741-502d-a02c-2bc686c3499b
│                          b62500d7-c981-595b-a798-eb6cf4d4942b
│     Device Flags:        • Updatable
│                          • Supported on remote server
│   
├─INTEL SSDPED1D280GA:
│     Device ID:           310f45f1f223064b5c16bf6dff31146755a64480
│     Summary:             NVM Express Solid State Drive
│     Current version:     E2010435
│     Vendor:              Intel Corporation (NVME:0x8086)
│     GUIDs:               82914f54-002c-5a60-bff9-6edfdb9363e8
│                          bd38a208-a535-5db5-b5f6-b20195a93991
│                          5ce12e21-5fdb-5d3a-a3d5-f56147e3b027
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation

# * snip* (remove HDD/SSD drives, which is irrelevant to a USB device)

│   
├─System Firmware:
│     Device ID:           cacb188703e68b1e91250076455571daceb5c0b8
│     Current version:     0.0.5125
│     Minimum Version:     0.0.5125
│     Vendor:              System manufacturer (DMI:American Megatrends Inc.)
│     GUID:                7645d05e-edbc-539c-8661-b8a82fb53f04
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:           066b8ecb265fbae43538d4a9728287b85989ffd8
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR24.11_B0036
      Bootloader Version:  BOT03.02_B0009
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server

```
Please answer the following questions:

- Operating system and version: Gentoo Linux (rolling release, no versions)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `emerge fwupd`
- Have you tried rebooting? Yes, many times, and power cycles. Seeing this on 2 different machines with 2 different ChaosKeys
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2020-02-10:
So that error comes from here: https://github.com/fwupd/fwupd/blob/master/plugins/altos/fu-altos-firmware.c#L60 which has been that way for a very long time (3y).  I would suspect a bad firmware binary.  Could you pull apart the CAB file and examine the binary to see if it actually looks like an ELF file?

chutz, 2020-02-11:
Seems to be valid ELF
```shell
# file chaoskey-v1.0-1.6.7.elf
chaoskey-v1.0-1.6.7.elf: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, with debug_info, not stripped
# readelf -e chaoskey-v1.0-1.6.7.elf
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           ARM
  Version:                           0x1
  Entry point address:               0x80011b1
  Start of program headers:          52 (bytes into file)
  Start of section headers:          241484 (bytes into file)
  Flags:                             0x5000200, Version5 EABI, soft-float ABI
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         3
  Size of section headers:           40 (bytes)
  Number of section headers:         19
  Section header string table index: 16

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .interrupt        PROGBITS        20000000 010000 0000c0 00  WA  0   0  4
  [ 2] .text             PROGBITS        08001100 011100 0027ac 00  AX  0   0  4
  [ 3] .boot             NOBITS          200000c0 0100c0 00000c 00  WA  0   0  4
  [ 4] .data             PROGBITS        200000cc 0200cc 000080 00  WA  0   0  4
  [ 5] .bss              NOBITS          2000014c 02014c 00086c 00  WA  0   0  4
  [ 6] .debug_info       PROGBITS        00000000 02014c 00b903 00      0   0  1
  [ 7] .debug_abbrev     PROGBITS        00000000 02ba4f 00383a 00      0   0  1
  [ 8] .debug_aranges    PROGBITS        00000000 02f290 000500 00      0   0  8
  [ 9] .debug_line       PROGBITS        00000000 02f790 0029f2 00      0   0  1
  [10] .debug_str        PROGBITS        00000000 032182 001c7f 01  MS  0   0  1
  [11] .comment          PROGBITS        00000000 033e01 000038 01  MS  0   0  1
  [12] .ARM.attributes   ARM_ATTRIBUTES  00000000 033e39 000031 00      0   0  1
  [13] .debug_loc        PROGBITS        00000000 033e6a 002521 00      0   0  1
  [14] .debug_ranges     PROGBITS        00000000 03638b 000448 00      0   0  1
  [15] .debug_frame      PROGBITS        00000000 0367d4 000d1c 00      0   0  4
  [16] .shstrtab         STRTAB          00000000 03ae8e 0000bc 00      0   0  1
  [17] .symtab           SYMTAB          00000000 0374f0 002760 10     18 407  4
  [18] .strtab           STRTAB          00000000 039c50 00123e 00      0   0  1
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings), I (info),
  L (link order), O (extra OS processing required), G (group), T (TLS),
  C (compressed), x (unknown), o (OS specific), E (exclude),
  y (purecode), p (processor specific)

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x000000 0x1fff0000 0x07ff1000 0x100c0 0x100cc RW  0x10000
  LOAD           0x011100 0x08001100 0x08001100 0x027ac 0x027ac R E 0x10000
  LOAD           0x0200cc 0x200000cc 0x080038ac 0x00080 0x008ec RW  0x10000

 Section to Segment mapping:
  Segment Sections...
   00     .interrupt .boot
   01     .text
   02     .data .bss
```

superm1, 2020-02-11:
Well I pulled apart the CAB file on LVFS (fromhttps://fwupd.org/lvfs/devices/com.altusmetrum.ChaosKey.firmware) and looked at it.

```
<?xml version="1.0" encoding="UTF-8"?>
<!-- Copyright 2017 Richard Hughes <richard@hughsie.com> -->
<component type="firmware">
  <id>com.altusmetrum.ChaosKey.firmware</id>
  <name>ChaosKey</name>
  <summary>Firmware for the Altus Metrum ChaosKey</summary>
  <description>
    <p>
      Updating the firmware on your ChaosKey device improves performance and adds
      new features.
    </p>
  </description>
  <provides>
    <!-- USB\VID_1D50&PID_60C6 -->
    <firmware type="flashed">b62500d7-c981-595b-a798-eb6cf4d4942b</firmware>
  </provides>
  <url type="homepage">http://www.altusmetrum.com/ChaosKey/</url>
  <metadata_license>CC0-1.0</metadata_license>
  <project_license>Proprietary</project_license>
  <developer_name>Altus Metrum</developer_name>
  <releases>
    <release urgency="medium" version="1.6.7" date="2017-01-01">
      <checksum filename="chaoskey-v1.0-1.6.7.ihx" target="content"/>
      <description>
        <p>
          Change the ADC clock speed to eliminate sampling problems which
          cleans up the chaoskey raw data.
        </p>
      </description>
    </release>
  </releases>
  <screenshots>
    <screenshot type="default">
      <image type="source">https://people.freedesktop.org/~hughsient/temp/chaoskey.svg</image>
      <caption>Remove the plastic cover, then connect pins 1 and 5 whilst inserting into a USB socket.</caption>
    </screenshot>
  </screenshots>
</component>
```

I notice that it's referring to the `ihx` file not the `elf` file.  The ihx file is listed as:
```
$ file chaoskey-v1.0-1.6.7.ihx
chaoskey-v1.0-1.6.7.ihx: ASCII text
```

@hughsie what's supposed to actually flash to the device?  The `ihx` or `elf`?

hughsie, 2020-02-11:
> what's supposed to actually flash to the device? The ihx or elf?

The .elf I believe...

superm1, 2020-02-11:
> The .elf I believe...

That seems to match here: https://altusmetrum.org/ChaosKey/
`$ ao-usbload chaoskey-v1.0-*.elf`

The firmware seems super out of date though too, that upload being from 2017?  Seems that there has been a bunch of releases since then: https://altusmetrum.org/AltOS/
So maybe can those guys get it updated?

@chutz for now, please try using `fwupdtool install-blob`.  Like this:
`/usr/libexec/fwupd/fwupdtool install-blob file.elf`

That should get you updated at least until the CAB is fixed.

chutz, 2020-02-11:
It looks like 1.6.7 is actually the latest version, from the website.

Running it manually on a downloaded .elf file did not work either:
```
# /usr/libexec/fwupd/fwupdtool install-blob chaoskey-v1.0-1.6.7.elf
Loading…                 [-                                      ]ERROR:esys:src/tss2-esys/esys_tcti_default.c:210:get_tcti_default() No standard TCTI could be loaded
ERROR:esys:src/tss2-esys/esys_context.c:68:Esys_Initialize() Initialize default tcti. ErrorCode (0x00070002)
18:04:13:0408 FuEngine             failed to add USB device 0bda:1100: USB error on device 0bda:1100 : Resource busy [-6]
18:04:13:0507 FuEngine             device 74861bcb9108e6a21cde76f31dad912828a79c03 [HGST HUS724020ALE640] does not define a vendor-id!
18:04:13:0512 FuEngine             device e1d6939452f58ba2c46050b3033cd16902a8873d [HGST HUS724020ALE640] does not define a vendor-id!
18:04:13:0518 FuEngine             device 1cb4548db4ea4d3a939a960a602d905d8f035e7a [Samsung SSD 850 EVO 1TB] does not define a vendor-id!
18:04:13:0532 FuEngine             device c38554fef33caf6a4e0bc72bcab72b405faed299 [INTEL SSDSC2CW240A3] does not define a vendor-id!
18:04:13:0537 FuEngine             device 7b1d1cf1423f0c73a8750f21876b9126fc5bf8e3 [HGST HDN724040ALE640] does not define a vendor-id!
Loading…                 [***************************************]
Choose a device:
0.      Cancel
1.      <snip>
2.      b95c8c172d3bfebd1c631bc4dadc9d2e790d7f89 (Altos ChaosKey)
12.     cacb188703e68b1e91250076455571daceb5c0b8 (System Firmware)
<snip>
2
verification only supported in bootloader
```

hughsie, 2020-02-12:
> verification only supported in bootloader

You need to do the paper-clip trick before inserting to get into BL mode.

superm1, 2020-02-14:
@hughsie can you ping the folks who made the CAB to fix it and include the right file?