fwupd/fwupd #1756: Unable to update BIOS on Latitude 7490

triceo, 2020-02-10:
Update gets stuck at "Adding new capsule", nothing happens for a couple minutes, then I restart computer and return to the OS.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.7
compile-time dependency versions
	gusb:	0.3.3
	efivar:	37
daemon version:	1.3.7
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Latitude 7490
│
├─Thunderbolt Controller:
│     Device ID:           ed76848154d43bd1c351513a63e41bdde9b1066a
│     Souhrn:              Unmatched performance for high-speed I/O
│     Current version:     23.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               ae4a91ee-0479-5cd6-9a03-2e6bd5190086 ← TBT-00d4081c
│                          4308d02d-77d2-53f0-b304-04c2b07394c8 ← TBT-00d4081c-0000:05:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─INTEL SSDSCKKF512G8 SATA 512GB:
│     Device ID:           64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Souhrn:              ATA Drive
│     Current version:     LHFD04N
│     Serial Number:       BTLA80840G84512K
│     GUIDs:               4a5c2dd4-fdf6-5212-9e14-a1d369a4d277 ← IDE\INTEL_SSDSCKKF512G8_SATA_512GB__________LHFD04N
│                          9211da3c-68b7-54c2-ad4d-e496abe00377 ← IDE\0INTEL_SSDSCKKF512G8_SATA_512GB__________
│                          fd0a50d4-35ed-56b1-9d35-c017216df868 ← INTEL SSDSCKKF512G8 SATA 512GB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─System Firmware:
      Device ID:           38390d47b6b0e59dab94e5feafcd6adefefdb76e
      Current version:     1.11.0
      Minimum Version:     1.11.0
      Vendor:              Dell Inc. (DMI:Dell Inc.)
      Update State:        success
      GUID:                a86a3f07-b7d3-4545-9238-0274fc1ba682
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Cryptographic hash verification is available
                           • Device is usable for the duration of the update
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0002,0003
[... 129 lines cut ...]
│       └── d1de729fe7fb468cac8bf24395dff02b-5.4.17-200.fc31.x86_64.conf
├── lost+found
├── memtest86+-5.01
├── System.map-5.4.14-200.fc31.x86_64
├── System.map-5.4.15-200.fc31.x86_64
├── System.map-5.4.17-200.fc31.x86_64
├── vmlinuz-0-rescue-d1de729fe7fb468cac8bf24395dff02b
├── vmlinuz-5.4.14-200.fc31.x86_64
├── vmlinuz-5.4.15-200.fc31.x86_64
└── vmlinuz-5.4.17-200.fc31.x86_64
```

Please answer the following questions:

- Operating system and version: Fedora 31
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `dnf`, comes preinstalled.
- Have you tried rebooting? Yes.
- Are you using an NVMe disk? No.
- Is secure boot enabled (only for the UEFI plugin)? Yes, but disabling didn't change anything.

This is the debugging output of fwupd:

```
../plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-ux-capsule../plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-a86a3f07-b7d3-4545-9238-0274fc1ba682-0../plugins/uefi/efi/fwupdate.c:714:efi_main(): Adding new capsule../plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)/\EFI\fedora\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap...../plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x17,0x0)/Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and PciRoot(0)/Pci(0x17,0x0)/Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and Pci(0x17,0x0)/Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\fedora\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap../plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device../plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume../plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file../plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): Read file; 82898 bytes../plugins/uefi/efi/fwupdate.c:548:fwup_add_update_capsule(): updates guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697../plugins/uefi/efi/fwupdate.c:549:fwup_add_update_capsule(): File guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697../plugins/uefi/efi/fwupdate.c:566:fwup_add_update_capsule(): Checking GOP for ux capsule../plugins/uefi/efi/fwupdate.c:714:efi_main(): Adding new capsule../plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)/\EFI\fedora\fw\fwupd-a86a3f07-b7d3-4545-9238-0274fc1ba682.cap...../plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x17,0x0)/Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and PciRoot(0)/Pci(0x17,0x0)/Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and Pci(0x17,0x0)/Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\fedora\fw\fwupd-a86a3f07-b7d3-4545-9238-0274fc1ba682.cap../plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device../plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume../plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file../plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): Read file; 10884287 bytes../plugins/uefi/efi/fwupdate.c:548:fwup_add_update_capsule(): updates guid: A86A3F07-B7D3-4545-9238-0274FC1BA682../plugins/uefi/efi/fwupdate.c:549:fwup_add_update_capsule(): File guid: A86A3F07-B7D3-4545-9238-0274FC1BA682../plugins/uefi/efi/fwupdate.c:729:efi_main(): n_updates: 2
```

superm1, 2020-02-10:
This sounds like the same thing as https://github.com/fwupd/fwupd/issues/1751.  Any chance you can downgrade to an older fwupd to see if it's helping?

superm1, 2020-02-10:
Reformatting your debugging log:
```

../plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-ux-capsule
../plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-a86a3f07-b7d3-4545-9238-0274fc1ba682-0
../plugins/uefi/efi/fwupdate.c:714:efi_main(): Adding new capsule
../plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)/\EFI\fedora\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap...
../plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x17,0x0)/Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and PciRoot(0)/Pci(0x17,0x0)/Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and Pci(0x17,0x0)/Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)
../plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\fedora\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
../plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): Read file; 82898 bytes
../plugins/uefi/efi/fwupdate.c:548:fwup_add_update_capsule(): updates guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697../plugins/uefi/efi/fwupdate.c:549:fwup_add_update_capsule(): File guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697
../plugins/uefi/efi/fwupdate.c:566:fwup_add_update_capsule(): Checking GOP for ux capsule../plugins/uefi/efi/fwupdate.c:714:efi_main(): Adding new capsule
../plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)/\EFI\fedora\fw\fwupd-a86a3f07-b7d3-4545-9238-0274fc1ba682.cap...../plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)../plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x17,0x0)/Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and PciRoot(0)/Pci(0x17,0x0)/Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and Pci(0x17,0x0)/Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and Sata(0x4,0x8000,0x0)/HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70) and HD(Part1,SigB4C8F95A-D876-4B9C-8343-27EC4C0D6F70)
../plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\fedora\fw\fwupd-a86a3f07-b7d3-4545-9238-0274fc1ba682.cap
../plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): Read file; 10884287 bytes
../plugins/uefi/efi/fwupdate.c:548:fwup_add_update_capsule(): updates guid: A86A3F07-B7D3-4545-9238-0274FC1BA682
../plugins/uefi/efi/fwupdate.c:549:fwup_add_update_capsule(): File guid: A86A3F07-B7D3-4545-9238-0274FC1BA682
../plugins/uefi/efi/fwupdate.c:729:efi_main(): n_updates: 2

```

We're not reaching this point of debugging output:
https://github.com/fwupd/fwupd/blob/master/plugins/uefi/efi/fwupdate.c#L598

So my (educated) guess would say that it's https://github.com/fwupd/fwupd/commit/44f55e2ee6df0b317df3029675afce86219cd9cd#diff-8e42b6d31c3f3bff11fde53a87097d86 causing it.

Something is failing/hanging when deleting those boot entries.  Can you possibly revert that commit?  If you're not comfortable building from scratch maybe @hughsie can give you an RPM build with it reverted to try.
@esmsnt since you're already upgraded in #1751 probably not too helpful unless you want to try the reinstall (I expect the bug to be exercised the same)

triceo, 2020-02-10:
I can confirm that the following version upgraded my BIOS just fine:

    [triceo@degrasse ~]$ fwupdmgr  --version
    client version:	1.2.11
    compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
    daemon version:	1.2.11

There was no particular reason for why I downgraded to this version, other than the RPM being the easiest to find.

superm1, 2020-02-10:
OK thanks, that confirms there is definitely a fwupd induced bug at play here.

superm1, 2020-02-10:
If I'm right and it's that bug at play what I would guess is happening is and infinite loop where `GetNextVariableName` keeps returning variables and the logic is failing either here:
`info_size < sizeof(EFI_LOAD_OPTION` or `		if (_StrHasPrefix(load_op->description, L"Linux Firmware Updater") ||` each time.

triceo, 2020-02-10:
Just FYI: unlike #1751, my laptop's fan did not spin up when the issue happened, in fact it was completely silent. From that evidence, I would guess the problem being something else than an infinite loop - unless inside that infinite loop, some IO operation is performed that makes the CPU wait.

superm1, 2020-02-10:
Well I wouldn't consider them different yet.  When running in UEFI it's a single threaded context that the fans are controlled entirely by the EC and the CPU and PCH will both be in their highest C states and package C states respectively.  It could easily be that the beefier CPU in #1751 generated more heat more quickly than the one in your system.

superm1, 2020-02-10:
CC @Ivanhu5866 

On an XPS 7390 I have on hand, I just tried master and reproduce the same behavior.  I added extra debugging output to see what's going on, and every variable it reads it takes a good 15-25 seconds to *process* before moving on the next one.

I let it go through to the `end`, and eventually it landed on `Boot0004` which corresponded to "Windows Boot Manager" for me and then the system rebooted, no more logs.

closed 2020-02-10 by superm1

Ivanhu5866, 2020-02-11:
I believe this is because 
_StrHasPrefix(&load_op->description, L"Linux Firmware Updater"

the "&" hasn't been applied on the original PR

commit d5bf8e45 (2020-02-10): Revert "Correctly delete UEFI variables"

This reverts commit 44f55e2ee6df0b317df3029675afce86219cd9cd.

This behavior caused fwupdx64.efi to loop for a very long time until
either aborting, running out of memory or some other problems.
Fixes: #1756
Fixes: #1751