fwupd/fwupd #1762: infinite loop on comparing memory

Ivanhu5866, 2020-02-11:
The patch "Correctly delete UEFI variables"(44f55e2ee6df0b317df3029675afce86219cd9cd) fix the real reason Boot#### never was found. But when the Boot#### found, it seems _StrHasPrefix which compares the memory for finding out the description causes the infinite loop.

Please answer the following questions:

- Operating system and version:
Ubuntu focal, fwupd verison 1.3.7-2
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
apt-get 
- Have you tried rebooting?  yes
- Are you using an NVMe disk? No
- Is secure boot enabled (only for the UEFI plugin)? No

I have built a test package fwupd 1.3.7-2.1 on my test ppa, https://launchpad.net/~ivan.hu/+archive/ubuntu/test/+packages,
which is base on the fwupd 1.3.7-2 (included the patch "Correctly delete UEFI variables" 44f55e2ee6df0b317df3029675afce86219cd9cd) and only modified 
-		if (_StrHasPrefix(load_op->description, L"Linux Firmware Updater") ||
-		    _StrHasPrefix(load_op->description, L"Linux-Firmware-Updater")) {
+		if (_StrHasPrefix(&load_op->description, L"Linux Firmware Updater") ||
+		    _StrHasPrefix(&load_op->description, L"Linux-Firmware-Updater")) {

it works fine on my machine, please help to check it.
[0001-uefi-fix-the-infinite-loop-when-comparing-memory.zip](https://github.com/fwupd/fwupd/files/4186554/0001-uefi-fix-the-infinite-loop-when-comparing-memory.zip)

superm1, 2020-02-11:
The code ended up getting ripped out instead; you might have missed https://github.com/fwupd/fwupd/pull/1760 being filed.

Ivanhu5866, 2020-02-12:
Getting ripped out the patch "Correctly delete UEFI variables" causes the BootEntry created by fwupd never  deleted, that concerned by some OEM/ODMs and users.
I am thinking if you can apply the patch "Correctly delete UEFI variables" and fix the infinite loop issue.

Applied the patch "Correctly delete UEFI variables" and code on above comment work fine on deleting BootEntry and updating firmware on my machine, but need your help to check.

Ivanhu5866, 2020-02-12:
Ok, it seems that you have decided not to remove the BootEntry which created by the fwupd. I will let our OEM/ODMs know the decision and get their feedback.

closed 2020-02-12 by Ivanhu5866