fwupd/fwupd #1770: fwupd 1.3.8: Uploading firmware update report fails with "invalid option selected"

aeckstein, 2020-02-13:
Upon running fwupdmgr after a successful update, upload of the update report is offered, but choosing '2' (upload and ask again) gives an error message. CapsLock is off btw.

```
fwupdmgr get-updates
No upgrades for System Firmware: current version is 0.1.45: 0.1.41=older, 0.1.40=older, 0.1.38=older, 0.1.37=older, 0.1.36=older, 0.1.45=same, 0.1.44=older, 0.1.43=older
No upgrades for UEFI Device Firmware: current version is 184.70.3626: 184.70.3626=same, 184.65.3590=older, 184.60.3561=older, 184.55.3510=older
No upgrades for UEFI Device Firmware: current version is 0.1.20: 0.1.20=same, 0.1.19=older, 0.1.18=older, 0.1.17=older, 0.1.15=older
________________________________________________

Devices that have been updated successfully:

 • System Firmware (0.1.44 → 0.1.45)
 • UEFI Device Firmware (0.1.19 → 0.1.20)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection):
0.      Do not upload reports at this time, but prompt again for future updates
1.      Do not upload reports, and never ask to upload reports for future updates
2.      Upload reports just this one time, but prompt again for future updates
3.      Upload reports this time and automatically upload reports after completing future updates
2
invalid option selected
```

Version info:

```shell
$ fwupdmgr --version
client version: 1.3.8
compile-time dependency versions
        gusb:   0.3.2
        efivar: 37
daemon version: 1.3.8
```

Please answer the following questions:

- Operating system and version: Gentoo Linux, up-to-date
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): portage, by renaming in-repo ebuild version 1.3.7 to 1.3.8
- Have you tried rebooting? Yes. Didn't make a difference, unsurprisingly.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? No.

hughsie, 2020-02-13:
This is my fault, apologies. @superm1 do you think the bug is that `FU_UTIL_HISTORY_PROMPT` isn't handled in the case statement, or that the `fu_util_maybe_enable_automatic()` method should just return TRUE if it gets to the end?

superm1, 2020-02-13:
Dang, this is going to get very annoying for people very quickly who haven't set the value.

I guess the code was confusing to follow that this common case should return true.  I think that if it gets to the end it should return TRUE.

aeckstein, 2020-02-13:
That's not it, unfortunately. I have applied the patch

```
diff -Naur fwupd-1.3.8.orig/src/fu-util.c fwupd-1.3.8/src/fu-util.c
--- fwupd-1.3.8.orig/src/fu-util.c      2020-02-13 18:47:02.000112447 +0100
+++ fwupd-1.3.8/src/fu-util.c   2020-02-13 18:47:22.680112899 +0100
@@ -295,7 +295,7 @@
                             FWUPD_ERROR,
                             FWUPD_ERROR_NOTHING_TO_DO,
                             "invalid option selected");
-       return FALSE;
+       return TRUE;
 }
 
 static gboolean
```
and got

```
Upload report now? (Requires internet connection):
0.      Do not upload reports at this time, but prompt again for future updates
1.      Do not upload reports, and never ask to upload reports for future updates
2.      Upload reports just this one time, but prompt again for future updates
3.      Upload reports this time and automatically upload reports after completing future updates
2

(fwupdmgr:13152): Fwupd-CRITICAL **: 18:53:34.771: fwupd_client_get_remotes: assertion 'error == NULL || *error == NULL' failed
invalid option selected
```

hughsie, 2020-02-13:
> think that if it gets to the end it should return TRUE.

Agree. I'm in daddy mode but can do a PR tmw.

superm1, 2020-02-13:
@aeckstein try https://github.com/fwupd/fwupd/pull/1773

aeckstein, 2020-02-13:
That worked fine, thanks superm1!

commit d803f11f (2020-02-13): fu-util: fix a logic error in report uploading introduced in e076d48afb74d1447936d353f12781755f13b047

Fixes: #1770
PR #1773 (2020-02-13): fu-util: fix a logic error in report uploading introduced in e076d48a…
…fb74d1447936d353f12781755f13b047

Fixes: #1770

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 5c7e1727 (2020-02-13): fu-util: fix a logic error in report uploading introduced in e076d48afb74d1447936d353f12781755f13b047

Fixes: #1770
commit 5c5674c3 (2020-02-28): fu-util: fix a logic error in report uploading introduced in e076d48afb74d1447936d353f12781755f13b047

Fixes: #1770
closed 2020-02-13 by superm1