fwupd/fwupd #1772: Updating Lenovo X1 Carbon 6th Gen does not work

dead10ck, 2020-02-13:
An update appeared this morning for my Lenovo X1 Carbon 6th gen:

```
$ fwupdmgr get-updates
No upgrades for Thunderbolt Controller: current version is 43.00: 43.00=same
No upgrades for UEFI Device Firmware: current version is 184.70.3626: 184.70.3626=same, 184.65.3590=older, 184.60.3561=older, 184.55.3510=older
20KGS3Y900
│
├─System Firmware:
│ │   Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│ │   Current version:     0.1.44
│ │   Minimum Version:     0.0.1
│ │   Vendor:              LENOVO (DMI:LENOVO)
│ │   Update State:        success
│ │   GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │                        • Cryptographic hash verification is available
│ │                        • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad X1 Carbon 6th System Update:
│       New version:       0.1.45
│       Remote ID:         lvfs
│       Summary:           Lenovo ThinkPad X1 Carbon 6th System Firmware
│       License:           Proprietary
│       Size:              9.5 MB
│       Vendor:            Lenovo Ltd.
│       Flags:             is-upgrade
│       Description:       Lenovo ThinkPad X1 Carbon 6th System Firmware Version 1.45
│       
│       The computer will be restarted automatically after updating BIOS completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.
│       
│       This stable release fixes the following issues:
│       
│        • Fixed an issue where Thunderbolt controller device might disappear from
│       
│       the Device Manager at 2nd boot.
│       
│       (Note) ECP version 1.20 or later is required for complete solution.
│     
└─UEFI Device Firmware:
  │   Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  │   Current version:     0.1.19
  │   Minimum Version:     0.1.19
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th Embedded Controller Update:
        New version:       0.1.20
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th Embedded Controller Firmware
        License:           Proprietary
        Size:              767.1 kB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad X1 Carbon 6th Embedded Controller Firmware 1.20
        
        The computer will be restarted automatically after updating EC completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.
        
        This stable release fixes the following issues:
[... 169 lines cut ...]
│   │       │   ├── fwupd-3babca5f-b2bf-4f4b-a72e-2bdc84eb4019.cap
│   │       │   └── fwupd-a4b51dca-8f97-4310-8821-3330f83c9135.cap
│   │       ├── fwupdx64.efi
│   │       ├── gcdia32.efi
│   │       ├── gcdx64.efi
│   │       ├── grub.cfg
│   │       ├── grubenv
│   │       ├── grubia32.efi
│   │       ├── grubx64.efi
│   │       ├── mmia32.efi
│   │       ├── mmx64.efi
│   │       ├── shim.efi
│   │       ├── shimia32.efi
│   │       ├── shimia32-fedora.efi
│   │       ├── shimx64.efi
│   │       └── shimx64-fedora.efi
```

Please answer the following questions:

- Operating system and version: Fedora 31
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

hughsie, 2020-02-13:
Please use 1.3.8 -- 1.3.7 had this regression :(

hughsie, 2020-02-13:
https://koji.fedoraproject.org/koji/taskinfo?taskID=41480810

hughsie, 2020-02-13:
or https://bodhi.fedoraproject.org/updates/FEDORA-2020-797aab4e5a

closed 2020-02-13 by hughsie

dead10ck, 2020-02-13:
Ah, thank you! I'll have to wait for that update to propagate to the mirrors, then I'll try it again.

alexmurray, 2020-02-17:
I am running fwupd 1.3.8 on Ubuntu 20.04 (devel) with a Lenovo X1 Carbon 6th Generation and am seeing something similar - but in this case after doing a `fwupdmgr update` and rebooting, the BIOS never appears to detect the update capsule and so the installation is never done. @hughsie do you want me to open a new bug about this? Also any logs or similar which I could provide?