fwupd/fwupd #1774: Lenovo T480s: "UEFI Device Firmware" naming

joakim-tjernlund, 2020-02-13:
It is confusing that several devices are just named "UEFI Device Firmware" in 
fwupdmgr get-devices:

```
├─UEFI Device Firmware:
│     Device ID:           cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
│     Current version:     184.70.3626
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                676af093-2a5c-4238-9c29-db8063a33532
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           323bda90831520867f8c02544efb9711dfca9fb6
│     Current version:     0.1.20
│     Minimum Version:     0.1.20
│     Vendor:              DMI:LENOVO
│     GUID:                4d254d6e-cd67-477b-97d5-bc3048af45c4
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           b4afb730aec9550f9c9aedd4b082716ef1611d18
│     Current version:     0.73.4
│     Minimum Version:     0.73.4
│     Vendor:              DMI:LENOVO
│     GUID:                6d1cf038-74ab-49b5-bb5c-6a1e99f3aea4
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
```
Same for USB2 HUB:
```
├─USB2.0 Hub:
│     Device ID:           54c916aa848583256a9b6959446d0890818cbe18
│     Summary:             USB 2.x Hub
│     Current version:     10.74
│     Vendor:              VIA Labs, Inc. (USB:0x17EF)
│     Install Duration:    15 seconds
│     GUIDs:               112b3acf-d42a-53cc-91c8-458c3b58f359 ← USB\VID_17EF&PID_3071&REV_0A74
│                          e2c96871-8be3-530e-a796-5652cee64eeb ← USB\VID_17EF&PID_3071
│                          3545c1f3-4d81-59f7-b9d9-d25d5811e4be ← USB\VID_17EF&PID_3071&HUB_10
│                          9af0af2d-99ff-551b-9134-e94809cc8016 ← USB\VID_17EF&PID_3071&DEV_VL820Q7
│     Device Flags:        • Updatable
│                          • Cryptographic hash verification is available
│                          • Device stages updates
│                          • Device can recover flash failures
│   
└─USB2.0 Hub:
      Device ID:           672c087de09848d9e7ee32aa1dea2fbeb8b81e6b
      Summary:             USB 2.x Hub
      Current version:     10.73
      Vendor:              VIA Labs, Inc. (USB:0x17EF)
      Install Duration:    15 seconds
      GUIDs:               fa8ac445-2c73-5bd7-99ab-6cfd94d1e40c ← USB\VID_17EF&PID_3071&REV_0A73
                           e2c96871-8be3-530e-a796-5652cee64eeb ← USB\VID_17EF&PID_3071
                           6f3f32bd-ac98-523e-9128-ef67941fb614 ← USB\VID_17EF&PID_3071&HUB_20
                           193e6679-e01f-5afc-a4b9-bcf31e14f09f ← USB\VID_17EF&PID_3071&DEV_VL820Q8
      Device Flags:        • Updatable
                           • Cryptographic hash verification is available
                           • Device stages updates
                           • Device can recover flash failures
```

superm1, 2020-02-13:
The infrastructure for this (The EFI sytem resource table or ESRT) doesn't specify anything about what device is what.  Other than modifying the specs, convincing the vendors to support the updates and then modifying the linux infrastructure the only way this could be improved would be by adding quirks in fwupd for individual GUIDs from the ESRT as it's discovered what they correspond to.

hughsie, 2020-02-13:
@superm1 anything you can get from Dell? I can ask Lenovo too.

superm1, 2020-02-13:
@hughsie for Dell the only thing that is _currently_ exported is system firmware, not any particular device firmware.  This will change in `$FUTURE`, and can revisit it then for Dell as needed.

joakim-tjernlund, 2020-02-13:
I see, one can see a better descr. in the FW itself though:
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24HT.firmware
``` 
Lenovo ThinkPad T480 Embedded Controller Firmware 1.20

superm1, 2020-02-13:
> I see, one can see a better descr. in the FW itself though:
> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24HT.firmware
> 
> ```
> Lenovo ThinkPad T480 Embedded Controller Firmware 1.20
> ```

Right, so for that device adding a quirk that the GUID from the ESRT corresponds to Embedded controller needs to be done manually in `uefi.quirk`.
PR welcome :)

hughsie, 2020-02-13:
I guess "Embedded Controller" is what we want as a name, right?

superm1, 2020-02-13:
Yeah I think so

hughsie, 2020-02-14:
Okay fixed in https://github.com/fwupd/fwupd/pull/1779 I think.

closed 2020-02-14 by hughsie

joakim-tjernlund, 2020-02-14:
Thanks a lot !