fwupd/fwupd #1775: 1.3.8: test suite is failing

kloczek, 2020-02-13:
1.3.7 was OK.
```
+ /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 48 --print-errorlogs
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.3.8/x86_64-redhat-linux-gnu'
ninja: no work to do.
 1/20 fwupd-exported-api                      OK       0.01 s
 2/20 fwupd-self-test                         FAIL    25.47 s (killed by signal 6 SIGABRT)
 3/20 fwupdplugin-exported-api                OK       0.02 s
 4/20 fwupdplugin-self-test                   OK       2.28 s
 5/20 fu-self-test                            OK      12.86 s
 6/20 dfu-self-test                           OK       0.06 s
 7/20 nitrokey-self-test                      OK       0.03 s
 8/20 synaptics-prometheus-self-test          OK       0.06 s
 9/20 wacom-usb-self-test                     OK       0.03 s
10/20 vli-self-test                           OK       0.06 s
11/20 ata-self-test                           OK       0.07 s
12/20 logitech-hidpp-self-test                OK       0.04 s
13/20 optionrom-self-test                     OK       0.04 s
14/20 tpm-eventlog-self-test                  OK       0.04 s
15/20 nvme-self-test                          OK       0.06 s
16/20 thunderbolt-self-test                   OK       3.17 s
17/20 redfish-self-test                       OK       0.04 s
18/20 dell-self-test                          OK       0.04 s
19/20 synaptics-mst-self-test                 OK       3.16 s
20/20 uefi-self-test                          OK       0.04 s

Ok:                   19
Expected Fail:         0
Fail:                  1
Unexpected Pass:       0
Skipped:               0
Timeout:               0

The output from the failed tests:

 2/20 fwupd-self-test                         FAIL    25.47 s (killed by signal 6 SIGABRT)

--- command ---
19:06:40 /home/tkloczko/rpmbuild/BUILD/fwupd-1.3.8/x86_64-redhat-linux-gnu/libfwupd/fwupd-self-test
--- stdout ---
# random seed: R02S236cd7e5945a306633f6353e1408b072
1..11
# Start of fwupd tests
ok 1 /fwupd/enums
ok 2 /fwupd/common{machine-hash}
ok 3 /fwupd/common{guid}
ok 4 /fwupd/release

ColorHug2
  DeviceId:             USB:foo
  Guid:                 2082b5e0-7a64-478a-b1b2-e3404fab6dad
  Guid:                 00000000-0000-0000-0000-000000000000
  Flags:                updatable|require-ac
  Checksum:             SHA1(beefdead)
  Icon:                 input-gaming,input-mouse
  Created:              1970-01-01
  Modified:             1970-01-02

  [Release]
  AppstreamId:          org.dave.ColorHug.firmware
  Description:          <p>Hi there!</p>
  Version:              1.2.3
  Filename:             firmware.bin
  Checksum:             SHA1(deadbeef)
  Size:                 1.0?kB
  Uri:                  http://foo.com
  Flags:                trusted-payload
ok 5 /fwupd/device
ok 6 /fwupd/remote{download}
ok 7 /fwupd/remote{base-uri}
ok 8 /fwupd/remote{no-path}
ok 9 /fwupd/remote{local}
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
(/home/tkloczko/rpmbuild/BUILD/fwupd-1.3.8/x86_64-redhat-linux-gnu/libfwupd/fwupd-self-test:1191808): GLib-DEBUG: 19:06:40.044: setenv()/putenv() are not thread-safe and should not be used after threads are created
Bail out! Fwupd:ERROR:../libfwupd/fwupd-self-test.c:489:fwupd_client_remotes_func: assertion failed (error == NULL): Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached (g-io-error-quark, 24)
--- stderr ---
**
Fwupd:ERROR:../libfwupd/fwupd-self-test.c:489:fwupd_client_remotes_func: assertion failed (error == NULL): Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached (g-io-error-quark, 24)
-------

Full log written to /home/tkloczko/rpmbuild/BUILD/fwupd-1.3.8/x86_64-redhat-linux-gnu/meson-logs/testlog.txt
error: Bad exit status from /var/tmp/rpm-tmp.8F1p05 (%check)
```

hughsie, 2020-02-13:
What distro? Is it reproducible?

superm1, 2020-02-13:
I would recommend running the build in a container or skip test suite if you can't. I've seen something like this in Ubuntu as well, specifically when there is an installed daemon in the system.

kloczek, 2020-02-13:
Why do you reccomend skip test suite?
Something is wrong wit those test units?

PS. I'm using my own distribution based on Fedora rawhide with LTO and --as-needed used on linking.

superm1, 2020-02-14:
Dbus activation of the systemd service on your system already is failing.  I don't know why, I never investigated it when it happened to me on Ubuntu. I just am telling you how to avoid it.

If you build it in a container you won't have the existing fwupd.service or dbus running. If you skip the unit tests you won't encounter it either.

hughsie, 2020-02-14:
> with LTO and --as-needed

I don't think the plugin loader works with these options. I seem to remember some failure with this a while ago, although that was before the libfwupdplugin stuff. It might be worth trying to `ninja install` and then running `sudo ./src/fwupd --verbose` and see if that works.

kloczek, 2020-02-15:
> then running sudo ./src/fwupd --verbose and see if that works.

Does it means that test suite needs root privs??

hughsie, 2020-02-15:
No, but the test suite does need a functional daemon. I was wondering if your compiled daemon actually ran...

kloczek, 2020-03-03:
It was not running because I had installed 1.37 and fwupd was not runnyng because usb library was broke. Looks like test suite is relaing on already running fwupd sertvice. In other word looks like at least part of the test suite is not testing just produced binaries but partoially already running fwupd service.

BTW: f`wupd --verbose` was completly freezing (without timeoput) on trying to communicate over dbus.

I think that that test suite should start separated dbus-session and communicate only with processeses inside that session.

PS. feel free to close that issue as I sorted out my issue. However IMO looks like it is still something to do in context of the testing produced binaries so I'm leaving you decision what to do with that :)

Nevertheless ..Thx :)

superm1, 2020-03-04:
This is what I was mentioning above in https://github.com/fwupd/fwupd/issues/1775#issuecomment-586033829.  There is definitely a situation that the test suite fails if there is an installed fwupd on the system that is broken.

I've hit it before myself too, but I don't see this as a priority to fix.  If you hit it, just don't run the test suite, or build it in a container.  However if someone hits it and wants to fix it, please open a pull request.

tomasbw, 2020-09-02:
This is still failing when running in fedora docker. 
 
1/24 fwupd-exported-api             OK             0.01s
 2/24 fwupd-self-test                OK             0.02s
 3/24 fwupdplugin-exported-api       OK             0.01s
 **4/24 fwupdplugin-self-test          FAIL           2.47s (killed by signal 6 SIGABRT)**
 5/24 fu-self-test                   OK             6.13s

hughsie, 2020-09-02:
> when running in fedora docker

Can you provide instructions to reproduce please. Thanks.

tomasbw, 2020-09-02:
```bash 
OS=fedora ./contrib/ci/generate_docker.py
podman run --privileged   -t   -v `pwd`/dist:/build/dist  fwupd-fedora
```

tomasbw, 2020-09-02:
Forgot to mention the tests fails in rpm build stage and succeeds  in a regular build inside the docker.

hughsie, 2020-09-02:
It's failing here:

    ERROR:../dist-unpack/fwupd-1.5.0/libfwupdplugin/fu-self-test.c:1661:fu_efivar_func: assertion failed (error == NULL): failed to set /build/build/meson-private/dist-unpack/fwupd-1.5.0/data/tests/efi/efivars/Test-8be4df61-93ca-11d2-aa0d-00e098032b8c as mutable: failed to get flags: Input/output error (g-io-error-quark, 0)

I'm guessing we can't set the "mutable" bit when running under docker. Digging a little deeper I see:

        rc = ioctl (fd, FS_IOC_GETFLAGS, &flags);
        if (rc < 0) {
                /* check for tmpfs */
                if (errno == ENOTTY || errno == ENOSYS) {
                        is_immutable = FALSE;
                } else {
                        g_set_error (error,
                                     G_IO_ERROR,
                                     G_IO_ERROR_FAILED,
                                     "failed to get flags: %s",
                                     strerror (errno));
                        return FALSE;
                }
        } else {
                is_immutable = (flags & FS_IMMUTABLE_FL) > 0;
        }

So I'm guessing `errno` is set to something different when run in a container.

tomasbw, 2020-09-02:
I guess the difference is that first compilation is done on mounted volume but rpmbuild is done inside docker filesystem. 
Maybe we can setup rpmbuild folder under dist ?

hughsie, 2020-09-02:
I'm not proficient with docker to be able to find the value of errno when run the rpmbuild in the container. If it's something that makes sense we can just add it to the other two that we ignore.

superm1, 2020-09-02:
I think it would be good to conclude why this isn't failing in CI though. Shouldn't it fail for same reason there? Or is it because we have --privileged on our CI run?

superm1, 2020-09-02:
Oh I see that in above comment `--privileged` was used too though.

superm1, 2020-09-02:
I just re-reran a CI job for 1_3_X: https://travis-ci.org/github/fwupd/fwupd/jobs/716714637 and it passes there, so how to reproduce this?

tomasbw, 2020-09-03:
It's failing on canary check.  Not sure yet how to enable core dump in docker. 

```
(/root/rpmbuild/BUILD/fwupd-1.5.0/x86_64-redhat-linux-gnu/libfwupdplugin/fwupdplugin-self-test:9818): GLib-DEBUG: 05:47:06.475: setenv()/putenv() are not thread-safe and should not be used after threads are created
ok 24 /fwupd/common{kernel-lockdown}
*** stack smashing detected ***: terminated
Aborted (core dumped)
```

kloczek, 2020-10-03:
I've just retestesd 1.4.5 and looks like everything is OK now.

Closing.

closed 2020-10-03 by kloczek