fwupd/fwupd #1783: Some tools shouldn't be in libexec

superm1, 2020-02-15:
I was reading through FHS after a MR was submitted to fwupd packaging in Debian and came across `libexec` details.

> /usr/libexec includes internal binaries that are not intended to be executed directly by users or shell scripts. Applications may use a single subdirectory under /usr/libexec.
> 
> Applications which use /usr/libexec in this way must not also use /usr/lib to store internal binaries, though they may use /usr/lib for the other purposes documented here.
> 
> 

https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch04s07.html

So that makes me think we should actually not be storing the following items there (as users or shell scripts may choose to access them directly).
* `fwupdagent`
* `fwupdate`
* `fwupdoffline`
* `fwupdtool`
* `fwupdtpmevlog`

The EFI binary it makes sense to me to still store there however, and the daemon.

Thoughts?

FFY00, 2020-02-15:
> as users or shell scripts may choose to access them directly

If this is supported then move then to `/usr/lib`.

> The EFI binary it makes sense to me to still store there however, and the daemon.

Yes , I agree. It should only be used internaly.

hughsie, 2020-02-15:
I'm not sure any of the tools should be used by "end users" except fwupdmgr. I really don't think debugging tools or executables designed to be run from systemd services belong in usrbin.

superm1, 2020-02-16:
Not `usr/bin`, but `usr/lib/$(arch)` or `usr/lib/$(arch)/fwupd`

hughsie, 2020-02-17:
> Not usr/bin, but usr/lib/$(arch)

Ahh, gotcha, sorry. I'm sorta struggling with the difference between usr/lib/$(arch) and libexec if I'm honest....

FFY00, 2020-02-18:
I think the technical differences between `libdir` and `libexecdir` have already been established for now. But I'll do a quick summary:

`bin` is for files that are supposed to be run by users, architecture dependent or otherwise.
`libexec` is for architecture dependent files that can be run by an external program.
`lib` is for architecture dependent files that should not be run by an external program.

If anyone still has questions feel free to ask :)

What we need to discuss is which components should be supported to be run by external programs. I am not comfortable enough giving an opinion on the matter as I am not that involved in the project and probably lack a lot of background.

So @hughsie and @superm1 you two should discuss.

hughsie, 2020-02-19:
> bin is for files that are supposed to be run by users, architecture dependent or otherwise.

So that's `fwupdmgr` for sure, but `fwupdtool` and `fwupdtpmevt` are really only designed for debugging -- but it designed to be run my users. I'm not sure I want to signify "user support" for these binaries by moving them into /usr/bin -- as that means man pages and all the jazz.

> libexec is for architecture dependent files that can be run by an external program.

So I guess this is the right place for `fwupdagent` and I guess systemd is an external program in this sense too, so `fwupd` itself should live here.

> lib is for architecture dependent files that should not be run by an external program.

I think the only thing that qualifies here is the efi binary, right?

superm1, 2020-02-19:
> libexec is for architecture dependent files that can be run by an external program.
lib is for architecture dependent files that should not be run by an external program.

You mixed up these two I thought; look at the link again in my first comment.

>  as that means man pages and all the jazz.

since we use `help2man` for `fwupdmgr` already, I think that's actually pretty easy.

FFY00, 2020-02-19:
> You mixed up these two I thought; look at the link again in my first comment.

Ah sorry, I mixed some things up. I am really sorry about that. `libdir` and `libexecdir` both contain architecture dependent files that should not be executed by an external program or users. `libexecdir` is optional and if it exists it should contain internal binaries while `libdir` contains object files or libraries.

I apologize, yet again, for mixing up some of the concepts. I re-read the FHS with more calm and verified the information in this reply is correct.

superm1, 2020-02-19:
OK so the right thing to do then is binaries that users should be running, they should go to `/usr/bin`.  That's what my new PR does. https://github.com/fwupd/fwupd/pull/1788

superm1, 2020-02-20:
#1788 was merged, clsoing this one.

commit 7d80338f (2020-02-18): Move various tools out of `libexecdir` into `libdir` (Fixes: #1783)

FHS 3.0 says that binaries not intended for execution directly by users
or scripts should be in `libexecdir`.
commit eae5eaca (2020-02-18): Move various tools out of `libexecdir` into `libdir` (Fixes: #1783)

FHS 3.0 says that binaries not intended for execution directly by users
or scripts should be in `libexecdir`.
commit 07805d08 (2020-02-18): Move various tools out of `libexecdir` into `libdir` (Fixes: #1783)

FHS 3.0 says that binaries not intended for execution directly by users
or scripts should be in `libexecdir`.
commit 5a4a3b10 (2020-02-18): Move various tools out of `libexecdir` into `libdir` (Fixes: #1783)

FHS 3.0 says that binaries not intended for execution directly by users
or scripts should be in `libexecdir`.
closed 2020-02-20 by superm1