fwupd/fwupd #1789: fwupdmgr Fails to Connect to Daemon

moriel5, 2020-02-19:
DIstro: Solus
Kernel 5.5.3
Client version: 1.3.8-16-g457a7c37
`gusb` version: 0.3.2 at compile-time
`efivar` version: 37 at compile-time (since removed, I am booted via legacy mode/CSM).

Whenever I try running fwupdmgr (self-compiled), it merely times-out when attempting to connect to the daemon.
Additionally, when running `/usr/libexec/fwupd/fwupd -v`, it segfaults immediately after the line: `13:52:22:0428 FuPlugin             added udev subsystem watch of drm_dp_aux_dev`.

I had just compiled `fwupd` a short while ago.

hughsie, 2020-02-19:
> it segfaults immediately

Can you get a backtrace please?

moriel5, 2020-02-19:
I just got off my laptop, so I'll do that either in the evening, or tomorrow afternoon, depending on how much time I'll have.

moriel5, 2020-02-19:
Update:
Is this good?
[strace.txt](https://github.com/fwupd/fwupd/files/4226618/strace.txt)

superm1, 2020-02-19:
@moriel5 a backtrace using `gdb` would be more useful (make sure you have debug symbols included though).

moriel5, 2020-02-19:
@superm1 Thanks, do you mean debug symbols for `fwupd`? Since I had compiled `fwupd`, and as a result I do not have them, and since some of the dependencies are not yet in Solus (I had to compile them as well), I won't be able to install them.

As for `dmesg`, I'll check.

superm1, 2020-02-19:
@moriel5 it will come down to your `CFLAGS` whether they are part of your hand compiled build.  You basically want `-g` https://gcc.gnu.org/onlinedocs/gcc/Debugging-Options.html

It doesn't matter the rest of dependencies debug symbols (usually).  We just need to see what is going on in fwupd for now, and if it's a below library we'll know from the gdb back trace.

moriel5, 2020-02-19:
> @moriel5 it will come down to your `CFLAGS` whether they are part of your hand compiled build. You basically want `-g` https://gcc.gnu.org/onlinedocs/gcc/Debugging-Options.html
>
> It doesn't matter [...]

Ah, thanks. I'll try then.

superm1, 2020-02-20:
> I'm stuck, since I don't see such a way to do this with meson.

Meson will respect your CLFAGS, just change your build environment's flags and it will pick them up.

>I had also thought, maybe #1077 may have to do with this, since I am also using Solus, and I did not touch the default flags, however I did not have any errors, and while -Wl does appear in the log (in multiple different combinations), -Bsymbolic-functions does not appear at all.

That's a good point to check.
Can you check what `LDFLAGS` the build environment is exporting?

moriel5, 2020-02-23:
@superm1 How do I go about those two? (I'm still learning).

superm1, 2020-02-23:
It's distro specific the right places to check and set all that, you're better off asking someone familiar with that distro.

moriel5, 2020-02-26:
Thanks, when I'll have the time, I'll do so.
Hopefully that will be this week.

closed 2020-04-03 by stale[bot]