fwupd/fwupd #1792: unifying: "No upgrades" message looks strange

phomes, 2020-02-19:
On a freshly updated fedora 32 I wanted to check if there were any firmware updates. I tested with two different receivers.  A C-U0007 and a C-U0008. One had no devices paired and the other two. The message looks strange on both but is slightly different.

```
# fwupdmgr --version
client version:	1.3.8
compile-time dependency versions
	gusb:	0.3.3
	efivar:	37
daemon version:	1.3.8
```

```
#  fwupdmgr get-devices
HP EliteBook 8540w
│
├─HP Integrated Module:
│     Device ID:           708a5b99d0afc737cdfcab5e16716ecbc6dee884
│     Current version:     3.6
│     Vendor:              Broadcom Corp (USB:0x03F0)
│     GUIDs:               08412d80-639e-522b-abd2-7d46b9445334 ← USB\VID_03F0&PID_231D&REV_0306
│                          79814083-7581-5801-b6f3-8792cc340f0c ← USB\VID_03F0&PID_231D
│     Device Flags:        • Updatable
│   
├─Samsung SSD 850 PRO 256GB:
│     Device ID:           602b0a6cc821d155208724f0e22f8d111542b74c
│     Summary:             ATA Drive
│     Current version:     EXM01B6Q
│     Serial Number:       S1SUNSAF810657M
│     GUIDs:               d45e7b4d-7d9d-5472-95b3-432d1e023c6f ← IDE\Samsung_SSD_850_PRO_256GB_______________EXM01B6Q
│                          c3a0ec2d-5408-5d6b-a512-a66eb772e04d ← IDE\0Samsung_SSD_850_PRO_256GB_______________
│                          985db2d9-379d-57e7-8b23-6743cd575744 ← Samsung SSD 850 PRO 256GB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           123fd4143619569d8ddb6ea47d1d3911eb5ef07a
│     Current version:     68CVD Ver. F.24
│     Vendor:              Hewlett-Packard
│     Update Error:        UEFI Capsule updates not available or enabled
│     GUID:                230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│     Device Flags:        • Internal device
│                          • Requires AC power
│                          • Needs a reboot after installation
│   
├─Unifying Receiver:
│     Device ID:           1e10c51ee65b4f02c36c3e741fbe7357a375aea2
│     Summary:             A miniaturised USB wireless receiver
│     Current version:     RQR24.11_B0036
│     Bootloader Version:  BOT03.02_B0009
│     Vendor:              USB:0x046D
│     Install Duration:    30 seconds
│     GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
│                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
│                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
│     Device Flags:        • Updatable
│                          • Supported on remote server
│   
└─Unifying Receiver:
      Device ID:           7b672a65fad989a03496eb4b451fa96168fab8a6
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.11_B0032
      Bootloader Version:  BOT01.04_B0016
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                           279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:        • Updatable
                           • Supported on remote server

```
    
With one receiver attached:
```
# fwupdmgr get-updates
No upgrades for Unifying Receiver: No releases found: Bootloader is not compatible, Bootloader is not compatible, Device Unifying Receiver doesn't support com.logitech.unifyingsigned, Bootloader is not compatible, Bootloader is not compatible
```

And the other:
```
# fwupdmgr get-updates
No upgrades for Unifying Receiver: No releases found: Device Unifying Receiver doesn't support com.logitech.unifyingsigned, Bootloader is not compatible, Bootloader is not compatible, Bootloader is not compatible, Bootloader is not compatible
```

With both:
```
# fwupdmgr get-updates
No upgrades for Unifying Receiver: No releases found: Device Unifying Receiver doesn't support com.logitech.unifyingsigned, Bootloader is not compatible, Bootloader is not compatible
No upgrades for Unifying Receiver: No releases found: Bootloader is not compatible, Bootloader is not compatible
```

I have testing firmware enabled:
```
# fwupdmgr refresh --force
Fetching metadata https://cdn.fwupd.org/downloads/firmware-testing.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc

Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Successfully downloaded new metadata: 2 local devices supported

```

superm1, 2020-02-19:
Well the confusing reason here is that the message is being shown for each release.  I do have a PR open that would just tear out the reason:
https://github.com/fwupd/fwupd/pull/1782

You can try that, and I expect you should get something like this:
```
• Unifying Receiver is updated to the latest version available.
• Unifying Receiver is updated to the latest version available.
```

FFY00, 2020-02-19:
This has to do with how the plugin is implemented. Those messages are actually a big step up from what they used to be (#1065).

#1782 seems way better though.

commit 2471fc36 (2020-02-20): Discard the reason upgrades aren't available (Fixes: #1678)

Also fixes: #1792
commit 4d1aa800 (2020-02-20): Discard the reason upgrades aren't available (Fixes: #1678)

Also fixes: #1792
commit b681ce40 (2020-02-20): Discard the reason upgrades aren't available (Fixes: #1678)

Also fixes: #1792
commit 27164b70 (2020-02-20): Discard the reason upgrades aren't available (Fixes: #1678)

Also fixes: #1792
closed 2020-02-20 by superm1