fwupd/fwupd #1793: Cant compile on ubuntu bionic

Ntemis, 2020-02-19:
|WARNING: Project targetting '>=0.47.0' but tried to use feature introduced in '0.50.0': install arg in configure_file
|Configuring libxmlb.test using configuration
|Configuring xb-version.h using configuration

subprojects/libxmlb/src/meson.build:105:2: ERROR: Program(s) ['g-ir-scanner'] not found or not executable

This is what meson-log contains
https://paste.ubuntu.com/p/7jPxThTSK6/

Ntemis, 2020-02-19:
hum false alarm :)
sudo apt install gobject-introspection

superm1, 2020-02-20:
To get a list of the packages you need to build, use this (new) helper job:
```
./contrib/ci/generate_dependencies.py -o ubuntu
```

superm1, 2020-02-20:
I've updated https://github.com/fwupd/fwupd/wiki/Compilation to better explain how to install missing dependencies.

FYI, for compilation on Ubuntu 18.04 you will find that some minimum package versions are not in the repositories.  In some instances you can disable those plugins, but in others you may need to hand compile.  

The snap builds on Ubuntu 18.04.  So you can also reference the snap build for the packages that it pulls that are not in the Ubuntu 18.04 repositories.
https://github.com/fwupd/fwupd/blob/master/snap/snapcraft.yaml

closed 2020-02-20 by superm1