fwupd/fwupd #1794: fwupd 1.3.8 fails to schedule / install UEFI Firmware upgrades for Lenovo X1 Carbon 6th

alexmurray, 2020-02-20:
After scheduling a firmware upgrade for 2 devices, they never get installed - and journalctl for fwupd shows an error about missing or invalid capsule header - see below for full details (I mentioned this in passing in https://github.com/fwupd/fwupd/issues/1772#issuecomment-586909620 but figured I should file a separate bug about it).

```shell
$ fwupdmgr get-updates
No upgrades for Thunderbolt Controller: current version is 43.00: 43.00=same
No upgrades for UEFI Device Firmware: current version is 184.70.3626: 184.70.3626=same, 184.65.3590=older, 184.60.3561=older, 184.55.3510=older
20KHCTO1WW
│
├─System Firmware:
│ │   Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│ │   Current version:     0.1.44
│ │   Minimum Version:     0.0.1
│ │   Vendor:              LENOVO (DMI:LENOVO)
│ │   Update State:        success
│ │   GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │                        • Cryptographic hash verification is available
│ │                        • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad X1 Carbon 6th System Update:
│       New version:       0.1.45
│       Remote ID:         lvfs
│       Summary:           Lenovo ThinkPad X1 Carbon 6th System Firmware
│       License:           Proprietary
│       Size:              9.5 MB
│       Vendor:            Lenovo Ltd.
│       Flags:             is-upgrade
│       Description:       Lenovo ThinkPad X1 Carbon 6th System Firmware Version 1.45
│       
│       The computer will be restarted automatically after updating BIOS completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.
│       
│       This stable release fixes the following issues:
│       
│        • Fixed an issue where Thunderbolt controller device might disappear from
│       
│       the Device Manager at 2nd boot.
│       
│       (Note) ECP version 1.20 or later is required for complete solution.
│     
└─UEFI Device Firmware:
  │   Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  │   Current version:     0.1.19
  │   Minimum Version:     0.1.19
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th Embedded Controller Update:
        New version:       0.1.20
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th Embedded Controller Firmware
        License:           Proprietary
        Size:              767.1 kB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad X1 Carbon 6th Embedded Controller Firmware 1.20
        
        The computer will be restarted automatically after updating EC completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.
        
        This stable release fixes the following issues:
[... 559 lines cut ...]
├── vmlinuz-5.4.0-14-generic
└── vmlinuz.old -> vmlinuz-5.4.0-12-generic

9 directories, 309 files
```

Please answer the following questions:

- Operating system and version:
```shell
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu Focal Fossa (development branch)
Release:	20.04
Codename:	focal
```
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
  - Installed via apt
- Have you tried rebooting?
  - Yes
- Are you using an NVMe disk?
  - Yes
- Is secure boot enabled (only for the UEFI plugin)?
  -Yes

superm1, 2020-02-20:
To rule out a signing problem, can you confirm if this happens with secure boot turned off too?

alexmurray, 2020-02-20:
I tried disabling Secure Boot in the BIOS and it still did not seem to find the updates - ie. it continued to boot into grub etc - then I re-ran `fwupdmgr update` to try and reschedule the updates, rebooted again and it still did not find the updates - so this does not seem relevant. Is there any other info / debugging I can try and get to help diagnose what is going on?

superm1, 2020-02-21:
Lenovo, so maybe Boot order lock?

alexmurray, 2020-02-21:
No Boot Order is not locked - also even if I manually choose `Linux-Firmware-Updater` from the `Choose a temporary boot device` option at startup it simply proceeds to boot as normal so I suspect there is something amiss with either the cap files themselves or perhaps the original update files downloaded from lvfs - is it safe for me to remove the cap files and then try re-running `fwupdmgr update` - is this likely to achieve anything?

superm1, 2020-02-21:
You can try to turn on debugging using `/usr/libexec/fwupdate -d`, try again and then see if the logging is viewable in `/usr/libexec/fwupdate -l`

alexmurray, 2020-02-21:
```shell
$ sudo /usr/lib/fwupd/fwupdate -d
Enabled fwupdate debugging
```

```shell
$ fwupdmgr update
No upgrades for Thunderbolt Controller: current version is 43.00: 43.00=same
Upgrade available for System Firmware from 0.1.44 to 0.1.45
20KHCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading 0.1.45 for System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[ -                                     ]
Scheduling…              [***************************************]
Successfully installed firmware
No upgrades for UEFI Device Firmware: current version is 184.70.3626: 184.70.3626=same, 184.65.3590=older, 184.60.3561=older, 184.55.3510=older
Upgrade available for UEFI Device Firmware from 0.1.19 to 0.1.20
20KHCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading 0.1.20 for UEFI Device Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating UEFI Device Firmware…                                   ]
Scheduling…              [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]: N
```

```shell
$ sudo /usr/lib/fwupd/fwupdate -l
system-firmware type, {a4b51dca-8f97-4310-8821-3330f83c9135} version 65580 can be updated to any version above 0
device-firmware type, {9c9d9769-32fa-4841-b550-ea998e754e99} version 3091598890 can be updated to any version above 3091598889
device-firmware type, {3babca5f-b2bf-4f4b-a72e-2bdc84eb4019} version 65555 can be updated to any version above 65554
device-firmware type, {cea87551-1701-43fb-afbc-6e8ce9728345} version 4784132 can be updated to any version above 4784131
```

Rebooting....

superm1, 2020-02-21:
After the reboot can you show the debugging log output (if any)

alexmurray, 2020-02-21:
After reboot it is the same from what I can see:

```shell
$ sudo /usr/lib/fwupd/fwupdate -l
system-firmware type, {a4b51dca-8f97-4310-8821-3330f83c9135} version 65580 can be updated to any version above 0
device-firmware type, {9c9d9769-32fa-4841-b550-ea998e754e99} version 3091598890 can be updated to any version above 3091598889
device-firmware type, {3babca5f-b2bf-4f4b-a72e-2bdc84eb4019} version 65555 can be updated to any version above 65554
device-firmware type, {cea87551-1701-43fb-afbc-6e8ce9728345} version 4784132 can be updated to any version above 4784131
```

I notice in https://github.com/fwupd/fwupd/issues/1101 others had roughly the same issue and the fix there seemed to be resetting the BIOS to defaults - so I am going to try this next and see how that goes unless you think there is something else I can try first?

superm1, 2020-02-21:
That's the wrong argument for showing the log, sorry. Can you look at --help for it, I don't recall it offhand

alexmurray, 2020-02-21:
```shell
$ sudo /usr/lib/fwupd/fwupdate -L
fwupdate.c:468:find_updates(): Found update fwupdate-3babca5f-b2bf-4f4b-a72e-2bdc84eb4019-0
fwupdate.c:468:find_updates(): Found update fwupdate-a4b51dca-8f97-4310-8821-3330f83c9135-0
fwupdate.c:1290:efi_main(): Adding new capsule
fwupdate.c:647:search_file(): File \EFI\ubuntu\fw\fwupdate-d3Ji7z.cap searched
fwupdate.c:708:open_file(): Found device
fwupdate.c:715:open_file(): Found volume
fwupdate.c:723:open_file(): Found file
fwupdate.c:1050:add_capsule(): Read file; 767104 bytes
fwupdate.c:1051:add_capsule(): updates guid: 3BABCA5F-B2BF-4F4B-A72E-2BDC84EB4019
fwupdate.c:1052:add_capsule(): File guid: G0
fwupdate.c:1086:add_capsule(): Image does not have embedded header
fwupdate.c:1088:add_capsule(): Allocating 767132 for capsule header.
fwupdate.c:1290:efi_main(): Adding new capsule
fwupdate.c:647:search_file(): File \EFI\ubuntu\fw\fwupdate-wlqPp4.cap searched
fwupdate.c:708:open_file(): Found device
fwupdate.c:715:open_file(): Found volume
fwupdate.c:723:open_file(): Found file
fwupdate.c:1050:add_capsule(): Read file; 9455008 bytes
fwupdate.c:1051:add_capsule(): updates guid: A4B51DCA-8F97-4310-8821-3330F83C9135
fwupdate.c:1052:add_capsule(): File guid: A4B51DCA-8F97-4310-8821-3330F83C9135
fwupdate.c:1072:add_capsule(): Image has capsule image embedded
fwupdate.c:1304:efi_main(): n_updates: 2
fwupdate.c:1154:apply_capsules(): QueryCapsuleCapabilities: Success max: 519196672 reset:1
fwupdate.c:1155:apply_capsules(): Capsules: 2
```

alexmurray, 2020-02-21:
(I re-ran fwupdmgr update after rebooting there I think)

superm1, 2020-02-21:
So it sure looks it's running.  @hughsie I guess this is a lenovo firmware issue then right/

CitadelCore, 2020-02-22:
Same issue with P72.

alexmurray, 2020-02-24:
I tried resetting the BIOS Settings (Load Optimised Defaults as they call it in the BIOS) but still no luck so not sure what else I can try (other than perhaps a different fwupd version?)

alexmurray, 2020-02-26:
I also tried enabling Thunderbolt BIOS Assist mode but that didn't make a difference either. I notice this issue is tagged `needinfo` but I am not sure what other information is required / requested - is there anything else I can do to try and debug this?

jasonnfls, 2020-02-28:
I am having the exact issue. I am using Ubuntu 20.04 dev on x1 carbon 6th gen.

alexmurray, 2020-03-05:
@hughsie  is there anything I can do to try and help debug this?

regulator-g, 2020-03-05:
Hi guys I have a X1 carbon 5th and have the same issue..

schedule firmware to update, reboot, some error flashes up really quickly (cant see what it says) then system boots normally without updating firmware

tomaspinho, 2020-03-05:
Faced the exact same issue today with a Carbon 7th. Was able to do the firmware upgrade by running the bootable CD version of the update here: https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-7th-gen-type-20qd-20qe/downloads/driver-list/component?name=BIOS%2FUEFI

Burn it to a USB drive with `dd` and then boot from it.

regulator-g, 2020-03-05:
has anyone contacted Lenovo about this? I have done many firmware updates without issue until this recent one

panlinux, 2020-03-12:
X1 5th gen user here, same issue. Launchpad bug: https://bugs.launchpad.net/bugs/1866055

superm1, 2020-03-12:
As there were some comments in Launchpad that this might be a regression, if someone affected could perform the following steps it may help in root causing if it really is a firmware bug or fwupd bug.  I currently believe it to be a firmware bug, but someone authoritative at this manufacturer  would need to confirm that.

0. Turn off secure boot.
1. Remove the packaged version using your package manager.
2. Clone fwupd from github.
3. Checkout the stable `1_3_X` branch
4. Follow steps https://github.com/fwupd/fwupd/wiki/Compilation to compile and install.
5. Try to run the update, verify it still fails.
6. Run a git bisection between the last time it was known to work and `1_3_X` HEAD
7. Share the bisection results.

Note: If your bisection concludes https://github.com/fwupd/fwupd/commit/44f55e2ee6df0b317df3029675afce86219cd9cd#diff-2f5c39c775e4cac7ebbeabed4cefca6a is the first bad commit, this is a false positive.  We know that is a problematic commit, and you may need to manually revert it during your bisection.

julian-klode, 2020-03-12:
This is https://github.com/rhboot/shim/issues/194

superm1, 2020-03-13:
@hughsie could you check if Fedora picked up the newer shim in the dev release there too to see if it's also affected?  It seems it should be a pretty generic problem.

hughsie, 2020-03-16:
I see this: https://koji.fedoraproject.org/koji/packageinfo?packageID=14502

i.e. Fedora is using an old shim "15" that was built on 2018-10-02

superm1, 2020-03-16:
Oh I guess that Ubuntu has been using snapshots then for a while since there hasn't been a tagged release. @vathpela can you check this problem?

hughsie, 2020-03-16:
I believe @martinezjavier is looking at this right now.

superm1, 2020-03-16:
OK as confirmed on that linked bug, this was fixed in https://github.com/rhboot/shim/commit/1870bae796022f8bbf60465352eac329ff1d6ffd so Ubuntu 20.04 needs to update to a newer snapshot.  Nothing else really to be done from fwupd side.

closed 2020-03-16 by superm1

martinezjavier, 2020-03-16:
> I believe @martinezjavier is looking at this right now.

Yes, @superm1 correctly pointed out the commit that introduced the regression. This has already fixed by @vathpela some ago and I mentioned in the [shim issue](https://github.com/rhboot/shim/issues/194#issuecomment-599720022) the commit that contains the fix.

So they should either use a newer snapshot or cherry-pick that change.

regulator-g, 2020-03-17:
Hi guys, apologies if this is off-track I cant make sense of the discussion but I'm on Ubuntu 19.10 (Pop OS) with this issue it's not just 20.04 that is affected..

As above I have a x1 carbon 5th and have been successfully updating my firmware for over a year until the last release (1.45)

superm1, 2020-03-17:
Check your shim version, see if it's been upgraded. If it has its the same issue. If not, open a new one so we can figure out what is happening.

regulator-g, 2020-03-18:
> Check your shim version, see if it's been upgraded. If it has its the same issue. If not, open a new one so we can figure out what is happening.

sorry mate how do I check my shim version?

superm1, 2020-03-18:
You have the old shim, different issue. Check for boot order lock in you BIOS setup. If that's off, open a new issue for debugging.

regulator-g, 2020-03-18:
ok thanks for the info, will give that a go.