fwupd/fwupd #1797: Secure boot is enabled, but shim isn't installed to the EFI system partition

Scarlet-Beard, 2020-02-20:
How's it going? I recently installed Elementary OS 5.1.2 Hera on my Lenovo ThinkPad T480 laptop and I get the below output when I try to update my firmware. Any help would be greatly appreciated! This was a clean install.

```shell
$ fwupdmgr update
Downloading 0.1.30 for 20L5000YUS System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating 20L5000YUS System Firmware…                             ]
Scheduling…              [***************************************]
Secure boot is enabled, but shim isn't installed to the EFI system partition
```

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.2.10
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
T480 Thunderbolt Controller
  DeviceId:             5ab931635e543ab18f1f3ab679e7215fcd658c27
  Guid:                 fe328c2d-5300-597b-a21d-991116db8ab6
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2020-02-20

20L5000YUS System Firmware
  DeviceId:             48477ba14bd9bd0c97323a07caa54b7766f917c2
  Guid:                 dc7f0308-1ef8-4774-9ba5-89a58c4d731c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.19
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2020-02-20
  Modified:             2020-02-20
  UpdateState:          failed-transient
  UpdateError:          Secure boot is enabled, but shim isn't installed to the EFI system partition

UEFI Device Firmware
  DeviceId:             ea0fd60df22477668e549841f6a22a8c31cae587
  Guid:                 9e21f98b-fe98-455c-b388-da5450ab6979
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.60.3561
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-02-20

UEFI Device Firmware
  DeviceId:             c19580db344e418cf2b23297bcf46d75c06e432a
[... 404 lines cut ...]
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-5.3.0-28-generic
├── initrd.img-5.3.0-40-generic
├── System.map-5.3.0-28-generic
├── System.map-5.3.0-40-generic
├── vmlinuz-5.3.0-28-generic
└── vmlinuz-5.3.0-40-generic

5 directories, 328 files
```

Please answer the following questions:

- Operating system and version: Elementary OS 5.1.2 Hera
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? Yes

superm1, 2020-02-20:
What does your `/etc/os-release` show?  The way that check works is that it uses contents in that file to dictate where to find `shimx64.efi` and where to install `fwupdx64.efi`.

Can you please also share that output of `tree` command in your first post when run as root instead?

Scarlet-Beard, 2020-02-20:
OS-Release info below.

```shell
NAME="elementary OS"
VERSION="5.1.2 Hera"
ID=elementary
ID_LIKE=ubuntu
PRETTY_NAME="elementary OS 5.1.2 Hera"
LOGO=distributor-logo
VERSION_ID="5.1.2"
HOME_URL="https://elementary.io/"
SUPPORT_URL="https://elementary.io/support"
BUG_REPORT_URL="https://github.com/elementary/os/issues/new"
PRIVACY_POLICY_URL="https://elementary.io/privacy-policy"
VERSION_CODENAME=hera
UBUNTU_CODENAME=bionic
```
```shell
$ sudo tree /boot
/boot
├── config-5.3.0-28-generic
├── config-5.3.0-40-generic
├── efi
│   └── EFI
│       ├── BOOT
│       │   ├── BOOTX64.EFI
│       │   └── fbx64.efi
│       ├── elementary
│       │   └── fw
│       │       ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│       │       └── fwupd-dc7f0308-1ef8-4774-9ba5-89a58c4d731c.cap
│       └── ubuntu
│           ├── BOOTX64.CSV
│           ├── fw
│           ├── fwupx64.efi
│           ├── grub.cfg
│           ├── grubx64.efi
│           ├── mmx64.efi
│           └── shimx64.efi
├── grub
│   ├── fonts
│   │   └── unicode.pf2
│   ├── gfxblacklist.txt
│   ├── grub.cfg
│   ├── grubenv
│   ├── locale
│   │   ├── ast.mo
│   │   ├── bg.mo
│   │   ├── ca.mo
│   │   ├── ca@valencia.mo
│   │   ├── cs.mo
│   │   ├── da.mo
│   │   ├── de_CH.mo
│   │   ├── de@hebrew.mo
│   │   ├── de.mo
│   │   ├── en@arabic.mo
│   │   ├── en_AU.mo
│   │   ├── en_CA.mo
│   │   ├── en@cyrillic.mo
│   │   ├── en_GB.mo
│   │   ├── en@greek.mo
│   │   ├── en@hebrew.mo
│   │   ├── en@piglatin.mo
│   │   ├── en@quot.mo
│   │   ├── eo.mo
│   │   ├── es.mo
│   │   ├── fi.mo
│   │   ├── fr.mo
│   │   ├── gl.mo
│   │   ├── hr.mo
[... 277 lines cut ...]
│       ├── video_cirrus.mod
│       ├── video_colors.mod
│       ├── video_fb.mod
│       ├── videoinfo.mod
│       ├── video.lst
│       ├── video.mod
│       ├── videotest_checksum.mod
│       ├── videotest.mod
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-5.3.0-28-generic
├── initrd.img-5.3.0-40-generic
├── System.map-5.3.0-28-generic
├── System.map-5.3.0-40-generic
├── vmlinuz-5.3.0-28-generic
└── vmlinuz-5.3.0-40-generic

11 directories, 338 files
```

superm1, 2020-02-20:
>ID=elementary
> ID_LIKE=ubuntu

This is the problem.  fwupd is looking at the `ID` key to figure out where to put (and look for) bootloaders.  The `ID_LIKE` key is where stuff is actually getting put though.

Unfortunately switching to `ID_LIKE` generically will not work because on **real** Ubuntu it's set to `debian` (and **real** Ubuntu doesn't put stuff in `/efi/debian`)

For a short term fix, you can just copy shimx64.efi into the elementary subdirectory on your ESP.  We'll have to think about what to do otherwise though.

What GRUB and Shim end up doing is using the `GRUB_DISTRIBUTOR` key from `/etc/default/grub` instead of `/etc/os-release`.

I wonder if we should source that first too if it exists.  @hughsie, thoughts?

superm1, 2020-02-20:
Funny enough, looking at `/etc/default/grub` on my system it's this:

```
GRUB_DISTRIBUTOR=`lsb_release -i -s 2> /dev/null || echo Debian`
```

So elementary must have changed something there I guess?  What's your `/etc/default/grub` look like?

Scarlet-Beard, 2020-02-20:
Worked perfectly and makes sense. Thank you so much!

superm1, 2020-02-20:
@Scarlet-Beard great.  can you please share your `/etc/default/grub` though so we can think about a longer term fix?

Scarlet-Beard, 2020-02-20:
Sure!

```shell
# If you change this file, run 'update-grub' afterwards to update
# /boot/grub/grub.cfg.
# For full documentation of the options in this file, see:
#   info -f grub -n 'Simple configuration'

GRUB_DEFAULT=0
GRUB_TIMEOUT_STYLE=hidden
GRUB_TIMEOUT=0
GRUB_DISTRIBUTOR=`lsb_release -i -s 2> /dev/null || echo Debian`
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash"
GRUB_CMDLINE_LINUX=""

# Uncomment to enable BadRAM filtering, modify to suit your needs
# This works with Linux (no patch required) and with any kernel that obtains
# the memory map information from GRUB (GNU Mach, kernel of FreeBSD ...)
#GRUB_BADRAM="0x01234567,0xfefefefe,0x89abcdef,0xefefefef"

# Uncomment to disable graphical terminal (grub-pc only)
#GRUB_TERMINAL=console

# The resolution used on graphical terminal
# note that you can use only modes which your graphic card supports via VBE
# you can see them in real GRUB with the command `vbeinfo'
#GRUB_GFXMODE=640x480

# Uncomment if you don't want GRUB to pass "root=UUID=xxx" parameter to Linux
#GRUB_DISABLE_LINUX_UUID=true

# Uncomment to disable generation of recovery mode menu entries
#GRUB_DISABLE_RECOVERY="true"

# Uncomment to get a beep at grub start
#GRUB_INIT_TUNE="480 440 1"
```

superm1, 2020-02-20:
Huh.. OK so that's the same as Ubuntu.  Why is your GRUB and shim getting placed in `/efi/ubuntu` then, not `/efi/elementary`?

superm1, 2020-02-20:
Maybe lets see what your `/etc/lsb-release` looks like?

Scarlet-Beard, 2020-02-20:
```shell
DISTRIB_ID=elementary
DISTRIB_RELEASE=5.1.2
DISTRIB_CODENAME=hera
DISTRIB_DESCRIPTION="elementary OS 5.1.2 Hera"
```

superm1, 2020-02-20:
Well then that doesn't make sense to me, GRUB_DISTRIBUTOR should be returning `elementary` not `ubuntu`.

Digging in further, it looks like they're patching the postinst.
https://bazaar.launchpad.net/~elementary-os/elementaryos/os-patch-grub2-bionic/revision/7

I don't know what we can do about this upstream.  It might need to be fixed in the fwupd package in elementary.

closed 2020-02-20 by superm1

jacklovell, 2020-10-01:
@superm1 I see the same behaviour when trying to update a Lenovo Thinkpad X1 Carbon 5th Gen running Linux Mint 20. This distribution is also based on Ubuntu.

EFI directory:
```
jlovell@jlovell-thinkpad:~$ sudo tree /boot/efi/
/boot/efi/
└── EFI
    ├── BOOT
    │   ├── BOOTX64.EFI
    │   ├── fbx64.efi
    │   └── mmx64.efi
    ├── linuxmint
    │   ├── fw
    │   │   ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
    │   │   └── fwupd-798ffd60-f10e-4ac4-8939-c8beabfe55b4.cap
    │   └── fwupdx64.efi
    └── ubuntu
        ├── BOOTX64.CSV
        ├── grub.cfg
        ├── grubx64.efi
        ├── mmx64.efi
        └── shimx64.efi
```

`/etc/default/grub` is the same as for elementary OS. The other files follow a similar pattern (just with name changes):

```
jlovell@jlovell-thinkpad:~$ cat /etc/os-release 
NAME="Linux Mint"
VERSION="20 (Ulyana)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 20"
VERSION_ID="20"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=ulyana
UBUNTU_CODENAME=focal
```
```
jlovell@jlovell-thinkpad:~$ cat /etc/lsb-release 
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=20
DISTRIB_CODENAME=ulyana
DISTRIB_DESCRIPTION="Linux Mint 20 Ulyana"
```
I do however have an explanation for how `GRUB_DISTRIBUTOR` ends up being `Ubuntu`: it's overridden by a config file:
```
jlovell@jlovell-thinkpad:~$ cat /etc/default/grub.d/50_linuxmint.cfg 
#! /bin/sh
set -e

# Set distributor to "Ubuntu"
#
# In Mint 16:
# shim and grub come with hardcoded ubuntu strings upstream, without this EFI installation is broken
# https://bugs.launchpad.net/ubuntu/+source/grub2/+bug/1242417
#
# In Mint 17:
# Despite the bug report being marked as fixed, the situation is the same. Grub was patched by Ubuntu
# to assign the "ubuntu" distributor ID when it was given "Kubuntu". That results in Kubuntu finally
# getting EFI to work out of the box but still using "ubuntu" as far as grub-install is concerned.
# So we'll stick to "Ubuntu" for this iteration.
#
# In Mint 18:
# Some packages are still hardcoded with the "ubuntu" path.. fwupdate for instance.
# Visually, the grub boot menuentry name is set for derivatives in 10_linux,
# but underneath the EFI path remains "ubuntu" for everything to work properly.

GRUB_DISTRIBUTOR="Ubuntu"
```

@Scarlet-Beard do you have a similar file present in elementary OS by any chance?

superm1, 2020-10-01:
@jacklovell good job root causing.  It's the same problem as Elementary, but overridden in a different way.  When fwupd is compiled in your derivative it needs to be compiled to use the overriden path.  There is a compilation option for it and it has been around since fwupd 1.2.7.
https://github.com/fwupd/fwupd/commit/5d9984e5ad656876858b6c369ec5c0ef14683043

https://github.com/fwupd/fwupd/blob/master/meson_options.txt#L35

superm1, 2020-10-01:
The other option is to try to remove the ESP's linuxmint directory as long as you have fwupd 1.3.4 or higher.  It should see it doesn't exist and try to use `ID_LIKE` key instead:
https://github.com/fwupd/fwupd/commit/e888b58ccd3d7532ee2fbff2c15188982414947c

jacklovell, 2020-10-13:
> The other option is to try to remove the ESP's linuxmint directory as long as you have fwupd 1.3.4 or higher. It should see it doesn't exist and try to use `ID_LIKE` key instead:
> [e888b58](https://github.com/fwupd/fwupd/commit/e888b58ccd3d7532ee2fbff2c15188982414947c)

This worked, thanks. I renamed it to `/boot/efi/EFI/linuxmint-rename` and the firmware upgrade went through successfully.

LinuxOnTheDesktop, 2022-05-16:
The rename trick made no difference for me, sadly: the firmware on my X1CG6 failed, as before, to upgrade (from 0.71.12 to 0.73.20). I'm on Mint Cinnamon 20.3. I tried both on the Mint repository version of fwup (which is 1.7.5-3) and on a recent flatpak.

auspex, 2024-09-05:
> The rename trick made no difference for me, sadly: the firmware on my X1CG6 failed, as before, to upgrade (from 0.71.12 to 0.73.20). I'm on Mint Cinnamon 20.3. I tried both on the Mint repository version of fwup (which is 1.7.5-3) and on a recent flatpak.

Interestingly, on Mint 22 (Wilma), that's what happened the first time I ran the Mint update manager, but when I went back in a little later, it showed "Your system is up to date", and no more pending firmware install, so it appears to have worked.