fwupd/fwupd #1805: Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached

dkwo, 2020-02-24:
fwupdmgr --version
client version: 1.3.8
compile-time dependency versions
        gusb:   0.3.3
        efivar: 37
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached

fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static; vendor preset: disabled)
     Active: failed (Result: exit-code) since Mon 2020-02-24 11:31:31 CET; 2min 48s ago
       Docs: https://fwupd.org/
    Process: 1167 ExecStart=/usr/lib/fwupd/fwupd (code=exited, status=127)
   Main PID: 1167 (code=exited, status=127)

systemd[1]: Starting Firmware update daemon...
fwupd[1167]: /usr/lib/fwupd/fwupd: symbol lookup error: /usr/lib/fwupd/fwupd: undefined symbol: g_usb_version_string, version LIBGUSB_0.1.0
systemd[1]: fwupd.service: Main process exited, code=exited, status=127/n/a
systemd[1]: fwupd.service: Failed with result 'exit-code'.
systemd[1]: Failed to start Firmware update daemon.

- Operating system and version: Arch Linux
- How did you install fwupd: pacman
- Have you tried rebooting? y
- Are you using an NVMe disk? y
- Is secure boot enabled (only for the UEFI plugin)? n

hughsie, 2020-02-24:
If you're using libgusb 0.3.4 you need to rebuild fwupd. See https://github.com/hughsie/libgusb/blob/master/NEWS#L5

FFY00, 2020-02-24:
This is a downstream issue, gusb was updated but fwupd wasn't rebuilt. I am rebuilding it right now.

@hughsie I was going to open an issue in the gusb upstream but it seems like gusb 0.3.3 is not binary compatible with 0.3.4, shouldn't the soname have been updated?

hughsie, 2020-02-24:
> shouldn't the soname have been updated

Nah, it was just a mistake in one symbol version that is used by one other project. I really don't think it's worth the pain of a soname bump.

FFY00, 2020-02-24:
> Nah, it was just a mistake in one symbol version that is used by one other project. I really don't think it's worth the pain of a soname bump.

It's more pain to me to receive bug reports like this than to rebuild 3 packages. It also breaks things for users, which is never a good experience. But it's fine, I am just saying that for you to consider next time :grin:.

dkwo, 2020-02-24:
Solved, thanks.

FFY00, 2020-02-24:
@dkwo forgot to reply here, I pushed a new update to the Archlinux repos.

superm1, 2020-02-24:
Does the library for gusb have a runtime version symbol we can refer? Our compile time symbol check in version was not helpful in discovering this issue.

dkwo, 2020-02-24:
That's what I meant: your update solved my issue :)

FFY00, 2020-02-24:
Ah, great!

superm1, 2020-02-24:
> Does the library for gusb have a runtime version symbol we can refer? Our compile time symbol check in version was not helpful in discovering this issue.

Oh, the fix was to resolve `g_usb_version_string`, so I guess that is not possible to use on an older version.

closed 2020-02-24 by superm1