fwupd/fwupd #1810: Quirks do not load on Chrome OS for > 1.3.3

djcampello, 2020-02-24:
It seems that an issue was introduced with this commit https://github.com/fwupd/fwupd/commit/51a869a01da445cd77dd9baddcf05d3bf0b4debf on v.1.3.3 that quirks are not properly loaded on Chrome OS:

19:06:27:0577 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
19:06:27:0577 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
19:06:27:0577 XbSilo               failed to load silo: Failed to open file ?/var/cache/fwupd/quirks.xmlb?: open() failed: No such file or directory
19:06:27:0577 XbSilo               compiling /usr/share/fwupd/quirks.d/altos.quirk:ctime=1582568391.673676?
19:06:27:0577 FuQuirks             failed to build silo: failed to compile /usr/share/fwupd/quirks.d/altos.quirk:ctime=1582568391.673676: Key file does not start with a group

hughsie, 2020-02-24:
> Key file does not start with a group

What's the contents of `/usr/share/fwupd/quirks.d/altos.quirk` out of interest?

superm1, 2020-02-24:
* Just to confirm, are there any patches being carried in `libxmlb` in chromeos?
For example I recall https://github.com/hughsie/libxmlb/pull/9 was opened and not merged.

* What version of `libxmlb`?

djcampello, 2020-02-24:
# cat /usr/share/fwupd/quirks.d/altos.quirk
# ChaosKey
[DeviceInstanceId=USB\VID_1D50&PID_60C6]
Plugin = altos
Flags = none
Name = Altos ChaosKey
CounterpartGuid = USB\VID_FFFE&PID_000A

[DeviceInstanceId=USB\VID_FFFE&PID_000A]
Plugin = altos
Flags = is-bootloader
Name = Altos [bootloader]

I believe that is the upstream content https://raw.githubusercontent.com/fwupd/fwupd/42f33df3dc9ba9e6b11ea4a199ea60d706e2b273/plugins/altos/altos.quirk

I brought in libxmlb-0.1.14.ebuild and not additional patches there. We got rid of the patch you referenced on https://chromium-review.googlesource.com/c/chromiumos/overlays/chromiumos-overlay/+/1825708

hughsie, 2020-02-24:
Would it be useful for me to have hardware so I can test this before I do releases?

superm1, 2020-02-24:
> Would it be useful for me to have hardware so I can test this before I do releases?

That's a good idea; but I would think a chroot is adequate.  Perhaps @djcampello could help to submit a CI test to libxmlb that tests it in the latest chrome development environment?

djcampello, 2020-02-25:
As @superm1 said this could be run in the chroot

$ sudo chroot /build/$BOARD/
# /usr/libexec/fwupd/fwupdtool get-updates -v

Interestingly enough the bug is not triggered in my chroot. It happens on my test device (sarien board)

hughsie, 2020-02-25:
> It happens on my test device

Is there any easy way to reproduce this on hardware I could buy on ebay? Or an engineering sample you could send me?

superm1, 2020-02-25:
>Interestingly enough the bug is not triggered in my chroot. It happens on my test device (sarien board)

This is surprising.  You're presumably running the exact same binaries on both?  Is this 100% reproducible on the SUT side?  Is plugging a debugger at memory possible? If not maybe instrumenting the verbose output for `libxmlb` to look more closely at what it was trying to parse will be illustrative to what's actually happening.

djcampello, 2020-02-25:
I have been stepping through with gdb and I found the issue in this line https://github.com/hughsie/libxmlb/blob/0.1.14/src/xb-builder-source-ctx.c#L118

https://github.com/hughsie/libxmlb/blob/0.1.14/src/xb-builder-source.c#L463 is called twice as expected: 
The first time with the pre-converted quirk file. Contents:
(gdb) p buf
$3 = "# ChaosKey\n[DeviceInstanceId=USB\\VID_1D50&PID_60C6]\nPlugin = altos\nFlags = none\nName = Altos ChaosKey\nCounterpartGuid = USB\\VID_FFFE&PID_000A\n\n[DeviceInstanceId=USB\\VID_FFFE&PID_000A]\nPlugin = altos\nF"...

The second time with the post-converted file. Contents:
(gdb) p buf
$7 = "<?xml version=\"1.0\" encoding=\"UTF-8\"?>\n<quirk><device id=\"b62500d7-c981-595b-a798-eb6cf4d4942b\"><value key=\"Plugin\">altos</value><value key=\"Flags\">none</value><value key=\"Name\">Altos ChaosKey</value>"...

But in both cases the guess on the problem line returns:
(gdb) p content_type
$10 = (gchar *) 0x55555566cbe0 "text/plain"

hughsie, 2020-02-25:
I guess maybe the content sniffing needs a data file? e.g the mime data? Or the magic database? I also wouldn't mind adding a hack like we did for win32...

superm1, 2020-02-25:
Is the Gentoo/ChromeOS equivalent of `mime-support` not installed maybe?

djcampello, 2020-02-25:
https://github.com/GNOME/glib/blob/glib-2-58/gio/gcontenttype.c#L686 returns <_gio_xdg_type_textplain> "text/plain"

Looking into the mime database in ChromeOS

djcampello, 2020-02-25:
It seems that my test device had /usr/share/mime stripped... Populating it again solves the issue. I need to figure out what is causing the folder to be stripped on an install but it is not a bug with fwupd. Closing this for now. Thanks for all the help

djcampello, 2020-02-25:
I found out that our glib package has USE=-mime because the /usr/share/mime is about 5M in space and we don't want to have that in our rootfs (x2). @hughsie you mentioned a possible hack so we don't need to bring the mime info package in. Is this something that we can get?

hughsie, 2020-02-26:
> you mentioned a possible hack

Have a look at the win32 defines a few lines down. I think deleting the magic db is going to cause other unexpected problems if I'm honest. Could you perhaps just nuke the esoteric stuff and leave the sane bits?

djcampello, 2020-02-26:
Sorry I got a bit lost: Which "esoteric stuff and leave the sane bits" are you referring to?
I did see what was done for Windows and it seems it was just a mapping from some mime information (i.e., ".xml") to the expected mime (i.e., "application/xml"). In this case we have no mime information available at all (i.e., "text/plain") so not sure how to address it. 
As far as I know, nothing else in Chrome OS uses the mime databases from x11-misc/shared-mime-info (they are not installed!) and hence why bringing it in at the storage cost would be contended.

superm1, 2020-02-26:
> Sorry I got a bit lost: Which "esoteric stuff and leave the sane bits" are you referring to?

Slim down the mime types database to just the more interesting things that matter for these use cases.

djcampello, 2020-02-26:
I prepared crrev.com/c/2075997 Lets see what the reviewers say

djcampello, 2020-02-26:
One question I have is why the conversion to .xml is on loop here https://github.com/hughsie/libxmlb/blob/0.1.14/src/xb-builder-source.c#L487? If there was no loop, we would identify the initial file as text/plain but after converting it to xml we would not have to check for mime type again. Am I missing something about this loop?

superm1, 2020-02-26:
It's to allow nesting of data types.  You can see more at this commit: https://github.com/hughsie/libxmlb/commit/04e88acc82a53367b154d5f3718cc8ba5f0ee0ac

djcampello, 2020-02-27:
Could we extend XbBuilderSourceAdapter in https://github.com/hughsie/libxmlb/blob/0.1.14/src/xb-builder-source.c#L39 to include the output_content_type ? Like this we would not need to guess the mime type. The chromium CL is not receiving love at this point.

hughsie, 2020-02-27:
> to include the output_content_type

I this we still need to sniff it to handle the nested content types as before.

djcampello, 2020-02-27:
I sent https://github.com/hughsie/libxmlb/pull/43 to add a new "simple" function in addition to the existing method for adapters. This would be used by fwupd instead (since it only has a single adapter fro .quirk -> .xml). I tested this on a Chromebook and fwupd works

djcampello, 2020-02-27:
Sent https://github.com/fwupd/fwupd/pull/1822 for fwupd side

closed 2020-02-27 by djcampello