fwupd/fwupd #1821: Warn the user if Thunderbolt is going to be powered off

allanday, 2020-02-27:
I recently reported #1814. While the suspend issue has been resolved, there is another aspect to the issue: Thunderbolt is momentarily powered off while it is being updated, but the user has no advance warning that this will happen. At best this can be disconcerting ("did something just go wrong?"), at worst it could interrupt something the user is doing, or even result in data loss (if you're using Thunderbolt to transfer data, or to power an external device).

If hardware is going to power off during an update, the user ought to be alerted to the fact before they run it.

hughsie, 2020-02-27:
@allanday could we do something like we already do for some USB controllers? e.g.

[image]

allanday, 2020-02-27:
> @allanday could we do something like we already do for some USB controllers? e.g.

Yes, in UI terms, we could certainly use a confirmation dialog to communicate what's going to happen.

superm1, 2020-02-27:
We do have a flag that the UX can use for it too: `FWUPD_DEVICE_FLAG_USABLE_DURING_UPDATE`

hughsie, 2020-02-27:
> FWUPD_DEVICE_FLAG_USABLE_DURING_UPDATE

Well, it's usable during update, but it does reboot right at the end, right?

superm1, 2020-02-27:
From a user's perspective, the (automatic) reboot at the end is part of the update process.  It's not possible to remove it from the update flow.

superm1, 2020-02-27:
I should clarify I meant the inverse of that flag.  If a device doesn't set it is when you should show a UX  message indicating such.

superm1, 2020-02-27:
And actually that's exactly what we use for this purpose in the command line tools these days too.
https://github.com/fwupd/fwupd/blob/master/src/fu-util-common.c#L411

superm1, 2020-02-27:
So I would say that this is actually invalid for fwupd since we already issue those warnings for awhile now, but is an issue for any GUI client.
It needs to use those flags to warn the user.

And old me even filed a bug for us to track doing that in `gnome-software` and `gnome-firmware`.
https://gitlab.gnome.org/GNOME/gnome-software/issues/832
https://gitlab.gnome.org/hughsie/gnome-firmware-updater/-/merge_requests/35

closed 2020-02-27 by superm1