fwupd/fwupd #1826: Lenovo T480s System Firmware Update Error

s950449, 2020-02-29:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.8
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.8
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
20L7CTO1WW
│
├─INTEL SSDPEKKW512G8:
│     Device ID:           c82980cd3fe66f7d0a8bb83b353cbc271177efd4
│     Summary:                NVM Express Solid State Drive
│     Current version:     004C
│     Vendor:              Intel Corporation (NVME:0x8086)
│     GUIDs:               79517f86-8df8-5d6e-a18b-33f0b36a78e9
│                          68db11e5-b0cf-5bc9-a94e-17e28496e505
│                          4d10db12-e333-5e97-9ef3-de112dd88175
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           e7cc234e82d08b6a14e246beb6b5f8142091e570
│     Current version:     0.1.38
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     Update Error:        TPM PCR0 differs from reconstruction, please report!
│     GUID:                ebfe8df8-dee7-4692-a721-cbcf5095c5cf
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
│     Current version:     184.70.3626
│     Minimum Version:     184.70.3626
│     Vendor:              DMI:LENOVO
│     GUID:                676af093-2a5c-4238-9c29-db8063a33532
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           323bda90831520867f8c02544efb9711dfca9fb6
      Current version:     0.1.20
      Minimum Version:     0.1.20
      Vendor:              DMI:LENOVO
      GUID:                4d254d6e-cd67-477b-97d5-bc3048af45c4
      Device Flags:        • Internal device
                           • Updatable
[... 615 lines cut ...]
│   │   ├── arch-git.conf
│   │   └── arch-mainline.conf
│   ├── loader.conf
│   └── random-seed
├── System Volume Information
│   ├── IndexerVolumeGuid
│   └── WPSettings.dat
├── vmlinuz-linux
├── vmlinuz-linux-git
└── vmlinuz-linux-mainline

```

Please answer the following questions:

- Operating system and version:
Arch Linux (Linux 5.6.0-rc3-1-mainline)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
pacman
- Have you tried rebooting?
Yes
- Are you using an NVMe disk?
Yes
- Is secure boot enabled (only for the UEFI plugin)?
No

der-eismann, 2020-03-02:
Did you post the actual error anywhere?

superm1, 2020-03-02:
I'd guess it's this he wanted to report:
```
 │     Update Error:        TPM PCR0 differs from reconstruction, please report!
```

Which might be a duplicate of #1803, but you'll need to provide `/usr/libexec/fwupd/fwupdtpmevlog` output to analyze to confirm.

hughsie, 2020-03-02:
> please report!

We probably ought to make that a wiki page...

superm1, 2020-03-02:
> > please report!
> 
> We probably ought to make that a wiki page...

Yeah, and maybe change
`TPM PCR0 differs from reconstruction, please report!` to `TPM PCR0 differs from reconstruction, please see $URL`.

Then in `$URL` can explain, and tell them what to capture and include on the bug.
I think this is worth getting into 1.3.9 too before you tag it.  I'm going to push for 1.3.9 to be whats in Ubuntu 20.04, so it will be around for a while.  If the bugs get too noisy around this, we can SRU a patch to just blacklist the TPM event log plugin.

s950449, 2020-03-03:
` $ sudo /usr/lib/fwupd/fwupdtpmevlog`
Output:
```
PCR:                     BIOS (0)
Type:                    0x8
Description:             EV_S_CRTM_VERSION
ChecksumSha1:            c08b69e95fe24a4203477d1b632dec7cbf02c18c685d8632f6ef9012b0ba3f39
BlobStr:                 N.2.2.E.T.6.1.W. ...

PCR:                     BIOS (0)
Type:                    0x80000008
Description:             EV_EFI_PLATFORM_FIRMWARE_BLOB
ChecksumSha1:            cbe4e3c25b63554ec19a981b5c2f5283f3b4539b72a35b8468759fda0a4999d9

PCR:                     BIOS (0)
Type:                    0x80000008
Description:             EV_EFI_PLATFORM_FIRMWARE_BLOB
ChecksumSha1:            e0ab4fa9ea5c5069b464fbcc3e83e2ec748273f8c34a87d771b7e40b6ef2da01

PCR:                     BIOS (0)
Type:                    0x80000008
Description:             EV_EFI_PLATFORM_FIRMWARE_BLOB
ChecksumSha1:            bfb8a75dbd5969eb8eb8e61174af9d83bf166f4e68879b23aee487a3bdbe955e

PCR:                     BIOS (0)
Type:                    0x80000008
Description:             EV_EFI_PLATFORM_FIRMWARE_BLOB
ChecksumSha1:            9a0fcce5d648aa4cd3abcdd26bc1590d8eb776dc6d7045c52fc0007ff7cf415c
BlobStr:                 ..........C.....

PCR:                     BIOS (0)
Type:                    0x1
Description:             EV_POST_CODE
ChecksumSha1:            14fe3bd3308ede3481cae58d0efd76cafa6922395f0692e1c49cb58d0a3f33e8
BlobStr:                 ACPI DATA

PCR:                     BIOS (0)
Type:                    0x1
Description:             EV_POST_CODE
ChecksumSha1:            1efad6adef14bd2607d26be9cdf8dbc7d26a6f2db9452d685ec64e0dc8c60d35
BlobStr:                 ACPI DATA

PCR:                     BIOS (0)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     BIOS Configuration (1)
Type:                    0x80000001
Description:             EV_EFI_VARIABLE_DRIVER_CONFIG
ChecksumSha1:            b6d26e2d437c457c9d7b0d7aa57d0a27be376284b10f3469319ad2241c286617
BlobStr:                 ..M*.A.E.o-.4..e................L.e.n.o.v.o.C.o.n.f.i.g......................................................................................................................................................................................................................................................

PCR:                     BIOS Configuration (1)
Type:                    0x80000001
Description:             EV_EFI_VARIABLE_DRIVER_CONFIG
ChecksumSha1:            88957fb52384ff1ded4bca3825fadb6434dfd5d1acae84efdf161e467d37d3a6
BlobStr:                 ....O..L....A..I................L.e.n.o.v.o.S.e.c.u.r.i.t.y.C.o.n.f.i.g............................................................................................................................................

PCR:                     BIOS Configuration (1)
Type:                    0x80000009
Description:             EV_EFI_HANDOFF_TABLES
ChecksumSha1:            d4196de4e73e3520df59ef3a65caf01f7f7b5ad2b0c0175a70f5ca0da6d597ea
BlobStr:                 ........D.....,J..... ....fz....

PCR:                     BIOS Configuration (1)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

[... 161 lines cut ...]
BlobStr:                 ....:=.E.....geo................d.b.x.

PCR:                     Platform manufacturer specific measurements (7)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     Static operating system (8)
Type:                    0xd
ChecksumSha1:            c0f77eb36da072014ab972e059b3701f6d614c19b8fdbb25c11d2f4f94e07199
BlobStr:                 i.n.i.t.r.d.=.\.i.n.t.e.l.-.u.c.o.d.e...i.m.g. .i.n.i.t.r.d.=.\.i.n.i.t.r.a.m.f.s.-.l.i.n.u.x.-.m.a.i.n.l.i.n.e...i.m.g. .r.o.o.t.=.P.A.R.T.U.U.I.D.=.3.8.5.7.c.e.0.c.-.0.6.4.d.-.c.7.4.9.-.b.4.9.2.-.5.f.7.3.4.8.f.1.7.1.0.1. .r.w..

PCRs:
  0:                     dd398dc28996b714bd159116dc9623b2b2b8843f
  1:                     ed5307dc56d0b7c90bb780152e69f9c223e15a3d
  2:                     3a4285c323c5af121f270900e88ffeefb2a34233
  3:                     3a4285c323c5af121f270900e88ffeefb2a34233
  4:                     ad64bc864df73fa2368114d4f5096e4eb22e7e0b
  5:                     3648796d97987bdb6cedd4173195dde9ef1f0cb4
  6:                     3a4285c323c5af121f270900e88ffeefb2a34233
  7:                     42c2439d553ea14f74bff3f1a776361daef85d0e
  8:                     6bde2cf8cd07fa5144f5f363f6c58e990b5a6349

```
Thanks!

superm1, 2020-04-01:
The exact root cause of this needs to be analyzed by Lenovo to fix it in firmware.  Moving it to the Lenovo repo.