fwupd/fwupd #1832: 1.3.8: /usr/libexec/fwupd/fwupd freezes

kloczek, 2020-03-03:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
** fwupdmgr --version
client version: 1.3.8
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.3.8
**
```
```shell
$ fwupdmgr get-devices
**# fwupdmgr --version
client version: 1.3.8
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.3.8
[root@barrel ~]# fwupdmgr get-devices
SYS-7039A-I
│
├─INTEL SSDPED1D480GA:
│     Device ID:           4137fae1df2d3c599da6871412fda6d26e9269f8
│     Summary:             NVM Express Solid State Drive
│     Current version:     E2010435
│     Vendor:              Intel Corporation (NVME:0x8086)
│     Serial Number:       PHMB747400JN480DGN
│     GUIDs:               82914f54-002c-5a60-bff9-6edfdb9363e8 ← NVME\VEN_8086&DEV_2700&REV_00
│                          bd38a208-a535-5db5-b5f6-b20195a93991 ← NVME\VEN_8086&DEV_2700
│                          e65e4817-a33c-598a-b945-5352c8c822c4 ← INTEL SSDPED1D480GA
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
├─INTEL SSDPEKNW020T8:
│     Device ID:           73e5c64ab94eb335fb590af99ecd15b5ff191bfa
│     Summary:             NVM Express Solid State Drive
│     Current version:     002C
│     Vendor:              Intel Corporation (NVME:0x8086)
│     Serial Number:       BTNH90420DAE2P0C
│     GUIDs:               c6cd9ab0-8f20-512e-9e1f-1af55b8454b9 ← NVME\VEN_8086&DEV_F1A8&REV_03
│                          82741c78-f5dc-5c23-a152-00de5799edc8 ← NVME\VEN_8086&DEV_F1A8
│                          7c0505be-b609-566e-bf94-561e441a9171 ← INTEL SSDPEKNW020T8
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
├─INTEL SSDPEKNW020T8:
│     Device ID:           c01308a71adb0e814e4f3697578d6d1ef5e58978
│     Summary:             NVM Express Solid State Drive
│     Current version:     002C
│     Vendor:              Intel Corporation (NVME:0x8086)
│     Serial Number:       PHNH9283003U2P0C
│     GUIDs:               c6cd9ab0-8f20-512e-9e1f-1af55b8454b9 ← NVME\VEN_8086&DEV_F1A8&REV_03
│                          82741c78-f5dc-5c23-a152-00de5799edc8 ← NVME\VEN_8086&DEV_F1A8
│                          7c0505be-b609-566e-bf94-561e441a9171 ← INTEL SSDPEKNW020T8
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│
├─INTEL SSDPEKNW020T8:
│     Device ID:           1413e47ade259411f2e591e5554e3ceeca9c5bcc
│     Summary:             NVM Express Solid State Drive
[... 662 lines cut ...]
23:21:34:0462 XbSilo               ignoring for OR statement: indexed string '82741c78-f5dc-5c23-a152-00de5799edc8' was unfound
23:21:34:0462 XbSilo               ignoring for OR statement: indexed string 'c6cd9ab0-8f20-512e-9e1f-1af55b8454b9' was unfound
23:21:34:0462 XbSilo               ignoring for OR statement: indexed string '82741c78-f5dc-5c23-a152-00de5799edc8' was unfound
23:21:34:0462 XbSilo               ignoring for OR statement: indexed string 'c6cd9ab0-8f20-512e-9e1f-1af55b8454b9' was unfound
23:21:34:0462 XbSilo               ignoring for OR statement: indexed string '82741c78-f5dc-5c23-a152-00de5799edc8' was unfound
23:21:34:0463 XbSilo               ignoring for OR statement: indexed string 'c6cd9ab0-8f20-512e-9e1f-1af55b8454b9' was unfound
23:21:34:0463 XbSilo               ignoring for OR statement: indexed string '82741c78-f5dc-5c23-a152-00de5799edc8' was unfound
23:21:34:0463 XbSilo               ignoring for OR statement: indexed string 'c6cd9ab0-8f20-512e-9e1f-1af55b8454b9' was unfound
23:21:34:0463 XbSilo               ignoring for OR statement: indexed string '82741c78-f5dc-5c23-a152-00de5799edc8' was unfound
23:21:34:0463 XbSilo               ignoring for OR statement: indexed string '95a1ee2f-402f-570b-9c51-fde95db6849a' was unfound
23:21:34:0463 XbSilo               ignoring for OR statement: indexed string '5b23aa59-694d-5622-a0b6-d294b9821711' was unfound
23:21:34:0463 XbSilo               ignoring for OR statement: indexed string '95a1ee2f-402f-570b-9c51-fde95db6849a' was unfound
23:21:34:0463 XbSilo               ignoring for OR statement: indexed string '5b23aa59-694d-5622-a0b6-d294b9821711' was unfound
23:21:34:0464 XbSilo               ignoring for OR statement: indexed string '95a1ee2f-402f-570b-9c51-fde95db6849a' was unfound
23:21:34:0464 XbSilo               ignoring for OR statement: indexed string '5b23aa59-694d-5622-a0b6-d294b9821711' was unfound
23:21:34:0464 XbSilo               ignoring for OR statement: indexed string '95a1ee2f-402f-570b-9c51-fde95db6849a' was unfound
23:21:34:0464 XbSilo               ignoring for OR statement: indexed string '5b23aa59-694d-5622-a0b6-d294b9821711' was unfound
23:21:34:0464 FuCommon             creating path /var/cache/fwupd/motd.d
23:21:34:0471 GLib-GIO             Failed to initialize portal (GMemoryMonitorPortal) for gio-memory-monitor: Not using portals
23:21:34:0472 GLib-GIO             _g_io_module_get_default: Found default implementation dbus (GMemoryMonitorDBus) for ‘gio-memory-monitor’
23:21:34:0477 FuMain               Started with locale en_GB.UTF-8
23:21:34:0477 FuMain               Daemon ready for requests
23:21:34:0479 FuMain               FuMain: acquired name: org.freedesktop.fwupd
^C
```

hughsie, 2020-03-04:
> acquired name: org.freedesktop.fwupd

Would `acquired name: org.freedesktop.fwupd, waiting for D-Bus requests` have clarified what it's doing there?

kloczek, 2020-03-04:
I just checked and as long as I'm executing `/usr/libexec/fwupd/fwupd --verbose` from non--root it is ok.. However started from root strace in last lines shows me:
```
[pid 3073621] write(18, "\1\0\0\0\0\0\0\0", 8) = 8
[pid 3073621] write(18, "\1\0\0\0\0\0\0\0", 8) = 8
[pid 3073621] poll([{fd=18, events=POLLIN}], 1, 0) = 1 ([{fd=18, revents=POLLIN}])
[pid 3073621] read(18, "\3\0\0\0\0\0\0\0", 16) = 8
[pid 3073621] write(19, "\1\0\0\0\0\0\0\0", 8) = 8
[pid 3073621] poll([{fd=17, events=POLLIN}], 1, 0 <unfinished ...>
[pid 3073617] <... poll resumed>)       = 1 ([{fd=19, revents=POLLIN}])
[pid 3073621] <... poll resumed>)       = 0 (Timeout)
[pid 3073617] read(19,  <unfinished ...>
[pid 3073621] write(18, "\1\0\0\0\0\0\0\0", 8 <unfinished ...>
[pid 3073617] <... read resumed>"\1\0\0\0\0\0\0\0", 16) = 8
[pid 3073621] <... write resumed>)      = 8
[pid 3073617] write(19, "\1\0\0\0\0\0\0\0", 8 <unfinished ...>
[pid 3073621] poll([{fd=17, events=POLLIN}, {fd=18, events=POLLIN}], 2, -1 <unfinished ...>
[pid 3073617] <... write resumed>)      = 8
[pid 3073621] <... poll resumed>)       = 1 ([{fd=18, revents=POLLIN}])
[pid 3073617] futex(0x55e3cf6ea6e0, FUTEX_WAKE_PRIVATE, 2147483647 <unfinished ...>
[pid 3073621] read(18,  <unfinished ...>
[pid 3073617] <... futex resumed>)      = 0
[pid 3073621] <... read resumed>"\1\0\0\0\0\0\0\0", 16) = 8
[pid 3073621] poll([{fd=17, events=POLLIN}, {fd=18, events=POLLIN}], 2, -1 <unfinished ...>
[pid 3073617] close(19)                 = 0
[pid 3073617] futex(0x55e3cf501030, FUTEX_WAKE_PRIVATE, 1) = 1
[pid 3073617] write(2, "\33[32m10:47:40:0004 ", 19 <unfinished ...>
[pid 3073624] <... futex resumed>)      = 0
10:47:40:0004 [pid 3073617] <... write resumed>)      = 19
[pid 3073624] futex(0x55e3cf501030, FUTEX_WAIT_PRIVATE, 1, {tv_sec=0, tv_nsec=499999121} <unfinished ...>
[pid 3073617] write(2, "FuMain               ", 21FuMain               ) = 21
[pid 3073617] write(2, "\33[34mStarted with locale en_GB.U"..., 41Started with locale en_GB.UTF-8
) = 41
[pid 3073617] write(2, "\33[32m10:47:40:0005 ", 1910:47:40:0005 ) = 19
[pid 3073617] write(2, "FuMain               ", 21FuMain               ) = 21
[pid 3073617] write(2, "\33[34mDaemon ready for requests\n\33"..., 35Daemon ready for requests
) = 35
[pid 3073617] poll([{fd=3, events=POLLIN}, {fd=13, events=POLLIN}, {fd=15, events=POLLIN}, {fd=16, events=POLLIN}], 4, 0) = 1 ([{fd=3, revents=POLLIN}])
[pid 3073617] read(3, "\4\0\0\0\0\0\0\0", 16) = 8
[pid 3073617] futex(0x7f00adaf78b8, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[pid 3073617] write(18, "\1\0\0\0\0\0\0\0", 8) = 8
[pid 3073621] <... poll resumed>)       = 1 ([{fd=18, revents=POLLIN}])
[pid 3073621] read(18,  <unfinished ...>
[pid 3073617] futex(0x55e3cf598660, FUTEX_WAIT_PRIVATE, 2, NULL <unfinished ...>
[pid 3073621] <... read resumed>"\1\0\0\0\0\0\0\0", 16) = 8
[pid 3073621] futex(0x55e3cf598660, FUTEX_WAKE_PRIVATE, 1 <unfinished ...>
[pid 3073617] <... futex resumed>)      = 0
[pid 3073621] <... futex resumed>)      = 1
[pid 3073617] write(18, "\1\0\0\0\0\0\0\0", 8 <unfinished ...>
[pid 3073621] futex(0x55e3cf598660, FUTEX_WAIT_PRIVATE, 2, NULL <unfinished ...>
[pid 3073617] <... write resumed>)      = 8
[pid 3073617] futex(0x55e3cf598660, FUTEX_WAKE_PRIVATE, 1) = 1
[pid 3073621] <... futex resumed>)      = 0
[pid 3073621] futex(0x55e3cf598660, FUTEX_WAKE_PRIVATE, 1) = 0
[pid 3073621] futex(0x55e3cf59a068, FUTEX_WAIT_PRIVATE, 2, NULL <unfinished ...>
[pid 3073617] futex(0x55e3cf59a068, FUTEX_WAKE_PRIVATE, 1 <unfinished ...>
[pid 3073621] <... futex resumed>)      = -1 EAGAIN (Resource temporarily unavailable)
[pid 3073617] <... futex resumed>)      = 0
[pid 3073621] futex(0x55e3cf59a068, FUTEX_WAKE_PRIVATE, 1 <unfinished ...>
[pid 3073617] eventfd2(0, EFD_CLOEXEC|EFD_NONBLOCK <unfinished ...>
[pid 3073621] <... futex resumed>)      = 0
[pid 3073617] <... eventfd2 resumed>)   = 19
[pid 3073621] sendmsg(17, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="l\1\0\1\260\0\0\0\24\0\0\0y\0\0\0\1\1o\0\25\0\0\0/org/fre"..., iov_len=320}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, MSG_NOSIGNAL <unfinished ...>
[pid 3073617] write(19, "\1\0\0\0\0\0\0\0", 8 <unfinished ...>
[pid 3073621] <... sendmsg resumed>)    = 320
[pid 3073617] <... write resumed>)      = 8
[pid 3073621] poll([{fd=17, events=POLLIN}, {fd=18, events=POLLIN}], 2, 0) = 1 ([{fd=18, revents=POLLIN}])
[pid 3073617] poll([{fd=19, events=POLLIN}], 1, 25000 <unfinished ...>
[pid 3073621] read(18,  <unfinished ...>
[pid 3073617] <... poll resumed>)       = 1 ([{fd=19, revents=POLLIN}])
[pid 3073621] <... read resumed>"\1\0\0\0\0\0\0\0", 16) = 8
[... 555 lines cut ...]
[pid 3073735] write(18, "\1\0\0\0\0\0\0\0", 8) = 8
[pid 3073735] poll([{fd=17, events=POLLIN}, {fd=18, events=POLLIN}], 2, -1) = 1 ([{fd=18, revents=POLLIN}])
[pid 3073735] read(18, "\1\0\0\0\0\0\0\0", 16) = 8
[pid 3073735] poll([{fd=17, events=POLLIN}, {fd=18, events=POLLIN}], 2, -1 <unfinished ...>
[pid 3073737] <... futex resumed>)      = -1 ETIMEDOUT (Connection timed out)
[pid 3073737] futex(0x55a4839b1860, FUTEX_WAIT_PRIVATE, 0, {tv_sec=14, tv_nsec=999999423} <unfinished ...>
[pid 3073734] <... poll resumed>)       = 0 (Timeout)
[pid 3073734] poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 2000) = 0 (Timeout)
[pid 3073734] poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 2000) = 0 (Timeout)
[pid 3073734] poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 2000) = 0 (Timeout)
[... cut ...]
[pid 3073737] exit(0)                   = ?
[pid 3073737] +++ exited with 0 +++
[pid 3073734] <... poll resumed>)       = 0 (Timeout)
[pid 3073734] poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 2000) = 0 (Timeout)
[pid 3073734] poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 2000) = 0 (Timeout)
[pid 3073734] poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 2000^C <unfinished ...>
[pid 3073731] <... poll resumed>)       = ? ERESTART_RESTARTBLOCK (Interrupted by signal)
```

superm1, 2020-03-04:
I'm confused; so is the daemon freezing or not?  When frozen can you define what that means?  Clients can't connect?  or "appears" stalled?

When executed from a console it will wait be idle waiting for a client to connect, but it also has `inotify` running on various directories that are monitored.

kloczek, 2020-03-04:
A .. ok so it is normal behaviour :)

Closing.

closed 2020-03-04 by superm1