fwupd/fwupd #1840: Group list of devices when using get-updates by state

dadosch, 2020-03-05:
Currently for every device either

> $name has no available firmware updates

or

> $name  has the latest available firmware version

or the equivalent when there is a new update is shown when doing get-updates.

This is very crowded and you have to look exactly do distinguish between the states.

I propose instead:

> **Devices with the latest available firmare version:**
> - name
> - name
>
> **Devices with no available firmware updates:**
> - name
> 
> **Device with updates available:**
> - name

where the latter is only show if there devices with updates.

superm1, 2020-03-05:
I think it seems like a reasonable proposal.  Would you be interested at taking a stab at implementing it?  It should mean patching the 4 times this comes up (in `src/fu-util.c` and `src/fu-tool.c`)

hughsie, 2020-03-05:
How would this work with tree nesting of composite devices?

superm1, 2020-03-05:
I think you would show the tree at the end, and those two groups at the start.

dadosch, 2020-03-05:
> I think it seems like a reasonable proposal. Would you be interested at taking a stab at implementing it? It should mean patching the 4 times this comes up (in `src/fu-util.c` and `src/fu-tool.c`)

My knowledge of C is too limited for that🙈

superm1, 2020-03-05:
@stoiandan maybe you can take a stab?

commit 0966b5cc (2020-07-02): fu-util/fu-tool: Group devices in `get-updates`/`update` calls

Fixes: #1840
commit 003820d5 (2020-07-03): fu-util/fu-tool: Group devices in `get-updates`/`update` calls

Fixes: #1840
commit eb7be16b (2020-07-03): fu-util/fu-tool: Group devices in `get-updates`/`update` calls

Fixes: #1840
closed 2020-07-03 by superm1