fwupd/fwupd #1841: Invalid ESRT GUID error shortly after boot

ciardullo-apps, 2020-03-05:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.2.10
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Samsung SSD 860 EVO 500GB
  DeviceId:             e421b2fc248391f6fe3e55ddbb3c9043be068bd0
  Guid:                 bc985be8-9f3e-5dfc-bee5-1e852e3204a3
  Guid:                 3831476a-add3-5bc8-88ff-31739e99ad07
  Guid:                 bd25ba7e-8dd0-5aed-90dd-db6a84712c8f
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              RVT03B6Q
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2020-03-05
```

```shell
$ efibootmgr -v
BootCurrent: 0003
Timeout: 1 seconds
BootOrder: 0003,0004,0001,0002
Boot0001* Hard Drive	BBS(HD,,0x0)/VenHw(5ce8128b-2cec-40f0-8372-80640e3dc858,0200)..GO..NO..........S.a.m.s.u.n.g. .S.S.D. .8.6.0. .E.V.O. .5.0.0.G.B...................\.,.@.r.d.=.X..........A...........................>..Gd-.;.A..MQ..L.5.S.8.9.E.N.M.0.4.9.8.8.2.5. .M. . . . ........BO
Boot0002* USB KEY	BBS(HD,,0x0)/VenHw(5ce8128b-2cec-40f0-8372-80640e3dc858,0b00)..GO..NO........y.P.N.Y. .U.S.B. .2...0. .F.D. .1.1.0.0...................\.,.@.r.d.=.X..........A.......................6..Gd-.;.A..MQ..L.A.A.0.0.0.0.0.0.0.0.0.0.2.2.7.5........BO
Boot0003* ubuntu	HD(1,GPT,7a422443-0d6d-4247-948b-c897de988cd7,0x800,0x100000)/File(\EFI\Ubuntu\shimx64.efi)..BO
Boot0004* UEFI: PNY USB 2.0 FD 1100, Partition 1	PciRoot(0x0)/Pci(0x14,0x0)/USB(9,0)/HD(1,MBR,0xc3072e18,0x70,0x1df3690)..BO
```

```shell
$ efivar -l | grep fw
(no output)
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
No .cap files reported

Excerpt:
/boot
├── config-5.3.0-28-generic
├── config-5.3.0-40-generic
├── efi
│   └── EFI
│       ├── BOOT
│       │   ├── BOOTX64.EFI
│       │   └── fbx64.efi
│       └── ubuntu
│           ├── BOOTX64.CSV
│           ├── fw
│           ├── fwupx64.efi
│           ├── grub.cfg
│           ├── grubx64.efi
│           ├── mmx64.efi
│           └── shimx64.efi

```

Please answer the following questions:

- Operating system and version: Ubuntu 18.04.4
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get
- Have you tried rebooting? Yes
- Are you using an NVMe disk? No
- Is secure boot enabled (only for the UEFI plugin)? no

Using 18.04.4 with kernel version 5.3.0-40, I recently started seeing a System Program error shortly after boot. I found this in /var/log/syslog:

```
Mar  5 10:20:27 mypc systemd[1]: Starting Firmware update daemon...
Mar  5 10:20:27 mypc boltd[1648]: power: guard '2' for 'fwupd' active
Mar  5 10:20:27 mypc fwupd[2579]: 15:20:27:0603 FuPluginUefi         failed to add /sys/firmware/efi/esrt/entries/entry0: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid
```
I found the invalid GUID in the following file

```
$ sudo cat /sys/firmware/efi/esrt/entries/entry0/fw_class
00000000-0000-0000-0000-000000000000
```
This [link](https://kernel.googlesource.com/pub/scm/linux/kernel/git/rzhang/linux/+/release/Documentation/ABI/testing/sysfs-firmware-efi-esrt) shows each file in this directory and what they mean, but I cannot determine where the GUID comes from or what would constitute a valid value. According to info at that link, the GUID should be found in the directory name, e.g. `entry00000000-0000-0000-0000-000000000000`

I haven't tried simply deleting the entry0 directory because the file sizes are all 4096 bytes, but the file contents are much smaller than that, figuring I won't be able to easily reconstruct the directory's contents:

```
$ ls -l /sys/firmware/efi/esrt/entries/entry0
total 0
-r-------- 1 root root 4096 Mar  5 10:20 capsule_flags
-r-------- 1 root root 4096 Mar  5 10:20 fw_class
-r-------- 1 root root 4096 Mar  5 10:20 fw_type
-r-------- 1 root root 4096 Mar  5 10:20 fw_version
-r-------- 1 root root 4096 Mar  5 10:20 last_attempt_status
-r-------- 1 root root 4096 Mar  5 10:20 last_attempt_version
-r-------- 1 root root 4096 Mar  5 10:20 lowest_supported_fw_version
$ sudo tail -n +1  /sys/firmware/efi/esrt/entries/entry0/* 
==> /sys/firmware/efi/esrt/entries/entry0/capsule_flags <==
0x50000

==> /sys/firmware/efi/esrt/entries/entry0/fw_class <==
00000000-0000-0000-0000-000000000000

==> /sys/firmware/efi/esrt/entries/entry0/fw_type <==
1

==> /sys/firmware/efi/esrt/entries/entry0/fw_version <==
1

==> /sys/firmware/efi/esrt/entries/entry0/last_attempt_status <==
0

==> /sys/firmware/efi/esrt/entries/entry0/last_attempt_version <==
1

==> /sys/firmware/efi/esrt/entries/entry0/lowest_supported_fw_version <==
1
```
Other than the disconcerting System Program Error popup shortly after boot and the messages in /var/log/syslog, there are no other manifestations of the issue or stability problems.

Appreciate any leads.

hughsie, 2020-03-05:
Interesting! What hardware is this please?

superm1, 2020-03-05:
@ciardullo-apps This is most likely a firmware bug on your system.
What vendor system do you have?

ciardullo-apps, 2020-03-05:
Homegrown, here are the specs

[Intel i5 8400](https://ark.intel.com/content/www/us/en/ark/products/126687/intel-core-i5-8400-processor-9m-cache-up-to-4-00-ghz.html)
[MSI B-360M Pro VDH](https://www.msi.com/Motherboard/B360M-PRO-VDH/Specification)
[Corsair Vengeance LPX 16GB (2x8GB) DDR4 2666MHz](https://www.corsair.com/us/en/Categories/Products/Memory/vengeance-lpx-black/p/CMK16GX4M2A2666C16#tab-tech-specs)
[Samsung EVO 860 500GB](https://www.samsung.com/us/computing/memory-storage/solid-state-drives/ssd-860-evo-2-5--sata-iii-500gb-mz-76e500b-am/#specs)

superm1, 2020-03-05:
This sounds like it should be a bug in the MSI BIOS (or the BIOS vendor they use).  If they're not supporting UEFI capsule updates, they shouldn't advertise an EFI system resource table.

I think we should be able to catch this though and not make it so noisy.

ciardullo-apps, 2020-03-05:
I flashed the MSI BIOS to the [latest version 7B24vA8](https://www.msi.com/Motherboard/support/B360M-PRO-VDH#down-bios), issue still persists.

superm1, 2020-03-05:
> This sounds like it should be a bug in the MSI BIOS (or the BIOS vendor they use). If they're not supporting UEFI capsule updates, they shouldn't advertise an EFI system resource table.
> 
> I think we should be able to catch this though and not make it so noisy.

Well actually it's all being caught properly.  The code is emitting a warning from the invalid data (like it should) and nothing explodes.

> Other than the disconcerting System Program Error popup shortly after boot and the messages in /var/log/syslog, there are no other manifestations of the issue or stability problems.

What is the system program error popup concerning?  Is it actually fwupd?  Have you sent in the error report for it?

ciardullo-apps, 2020-03-05:
Watching `/var/log/syslog` shows that the popup occurs at the moment the `ESRT GUID was invalid` message is logged. The system program error popup does not specify the cause, the popup simply reads `System program problem detected, Do you want to report the problem now?` 

When I click _Report Problem_, nothing happens and syslog reads "Refusing to render service to dead parents". I have never sent in an error report before, would you be able to direct me how to do so?  Thank you @superm1 and @hughsie for the quick and helpful replies

superm1, 2020-03-05:
Hmm, I wonder if the Ubuntu error detector is flagging it because it was saved as a warning then.

Clicking that button is all you're supposed to do.  Assuming the cause is that fwupd logged a warning, as a workaround for now so it will stop popping up, you can try to add `uefi` to `/etc/fwupd/daemon.conf` in the plugin blacklist and fwupd won't try to probe your broken firmware again when you reboot.

stalkerg, 2021-06-18:
I have the same issue with MSI motherboard for AM4 base on X470

hughsie, 2021-06-18:
@stalkerg can you report this to MSI please.

stalkerg, 2021-06-18:
I made comment on reddit but I don't know how to make this report legally. I don't think MSI has support on that level, will be much better somehow find developer's contacts.

itaranto, 2024-02-18:
I'm experiencing the same issue with a Gigabyte [GA-AB350-Gaming 3](https://www.gigabyte.com/us/Motherboard/GA-AB350-Gaming-3-rev-1x#kf) motherboard.

It started all of a sudden, I **didn't update** the BIOS. `fwupd` shows the following:

```
FuPluginUefiCapsule  skipping device that failed coldplug: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid
Failed to load daemon: failed to load engine: Failed to prepare SQL to get history: no such column: version_format
```

What I can recall I did recently is to re-create my ESP (FAT) partition again (reasons are unimportant) while keeping the the same filesystem UUID for convenience reasons. In this case, since the partition was re-created, the partition UUID is now different.

I don't know if that could have anything to do with the issue or not.

I'm also using Arch Linux, so kernel updates are frequent.

itaranto, 2024-02-18:
Oh it seems there are two unrelated issues here: there's the GUID issue and also there's seems to be an SQL schema problem which prevents the `fwupd` service to start.

I'll create a separate issue for this.

hughsie, 2024-02-18:
> ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid

This is something you should report to the OEM.