fwupd/fwupd #1856: fwupd.shutdown fails due to wrong path to fwupdtool

christophgysin, 2020-03-11:
**Describe the bug**
fwupd.shutdown fails because it refers to /usr/lib/fwupd/fwupdtool instead of /usr/bin/fwupdtool

**Steps to Reproduce**
Install fwupd, shutdown. Or build fwupd, and look at build/data/fwupd.shutdown.

**Expected behavior**
fwupd.shutdown should point to where fwupdtool gets installed.

**fwupd version information**
```shell
$ fwupdmgr --version
client version:1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:37
daemon version:1.3.9
```
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman -S fwupd

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
[not relevant]
```

**Additional questions**
- Operating system and version: ArchLinux (latest)
- Have you tried rebooting? Yes.
- Is this a regression? Yes.

superm1, 2020-03-11:
It was reported right after 1.3.9 came out.  It was a mistake, fixed in 1_3_X branch.

https://github.com/fwupd/fwupd/commit/4e381f2a61c6ea4eb7438c14f76b8301762a800c

christophgysin, 2020-03-11:
Ah, my mistake. Searched through the issues, but couldn't find this one. Closing.

closed 2020-03-11 by christophgysin

superm1, 2020-03-11:
Well my mistake in missing that path when we switched things around to `bindir` too😜