fwupd/fwupd #1857: Error message when restarting fwupd.service

smutel, 2020-03-12:
**Describe the bug**
- First I have found that when the service fwupd has been restarted, the message below appears: `fwupd[9693]: 06:57:48:0597 FuPluginUefi         Failed to get PCR0s: Le processus fils s’est terminé avec le code 1`
- So I installed the debian package `tpm2-tools`
- Now when I restart the fwupd service, the message below appears: `fwupd[10090]: 07:01:45:0221 FuPluginUefi         Failed to get PCR0s: Le processus fils s’est terminé avec le code 1`
In english: the child process exit with code 1

**Steps to Reproduce**
Don't know if this behavior could be reproduce

**Expected behavior**
No error message when restarting fwupd service

**fwupd version information**

```shell
client version: 1.2.10
compile-time dependency versions
        gusb:   0.2.11
        efivar: 34
daemon version: 1.2.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

**fwupd device information**

```shell
T480 Thunderbolt Controller
  DeviceId:             35e90d4ea4e894601a5927bec48bd722dfcb2320
  Guid:                 7d131734-64ed-5865-a46e-a67cfb9b5a46
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2020-03-12

Intel AMT [unprovisioned]
  DeviceId:             e2623122c99d58220498aacbfcfdb1baebbae3c5
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.8.70.3626
  VersionBootloader:    11.8.70.3626
  VersionFormat:        intel-me
  Icon:                 computer
  Created:              2020-03-12

20L5CTO1WW System Firmware
  DeviceId:             48477ba14bd9bd0c97323a07caa54b7766f917c2
  Guid:                 dc7f0308-1ef8-4774-9ba5-89a58c4d731c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.31
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2020-03-12
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             5decfff65e3391259bca4003a090e59bf50e3bea
  Guid:                 dd31d983-cf3b-4c84-acc9-70caa94e827b
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.70.3626
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-03-12

UEFI Device Firmware
  DeviceId:             c19580db344e418cf2b23297bcf46d75c06e432a
  Guid:                 b87a926d-189c-49a1-b1ff-921099b1de89
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.20
  VersionLowest:        0.1.20
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-03-12

T480 Thunderbolt Controller
  DeviceId:             35e90d4ea4e894601a5927bec48bd722dfcb2320
  Guid:                 7d131734-64ed-5865-a46e-a67cfb9b5a46
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2020-03-12

Intel AMT [unprovisioned]
  DeviceId:             e2623122c99d58220498aacbfcfdb1baebbae3c5
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.8.70.3626
  VersionBootloader:    11.8.70.3626
  VersionFormat:        intel-me
  Icon:                 computer
  Created:              2020-03-12

20L5CTO1WW System Firmware
  DeviceId:             48477ba14bd9bd0c97323a07caa54b7766f917c2
  Guid:                 dc7f0308-1ef8-4774-9ba5-89a58c4d731c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.31
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2020-03-12
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             5decfff65e3391259bca4003a090e59bf50e3bea
  Guid:                 dd31d983-cf3b-4c84-acc9-70caa94e827b
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.70.3626
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-03-12

UEFI Device Firmware
  DeviceId:             c19580db344e418cf2b23297bcf46d75c06e432a
  Guid:                 b87a926d-189c-49a1-b1ff-921099b1de89
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.20
  VersionLowest:        0.1.20
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-03-12

T480 Thunderbolt Controller
  DeviceId:             35e90d4ea4e894601a5927bec48bd722dfcb2320
  Guid:                 7d131734-64ed-5865-a46e-a67cfb9b5a46
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2020-03-12

Intel AMT [unprovisioned]
  DeviceId:             e2623122c99d58220498aacbfcfdb1baebbae3c5
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.8.70.3626
  VersionBootloader:    11.8.70.3626
  VersionFormat:        intel-me
  Icon:                 computer
  Created:              2020-03-12

20L5CTO1WW System Firmware
  DeviceId:             48477ba14bd9bd0c97323a07caa54b7766f917c2
  Guid:                 dc7f0308-1ef8-4774-9ba5-89a58c4d731c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.31
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2020-03-12
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             5decfff65e3391259bca4003a090e59bf50e3bea
  Guid:                 dd31d983-cf3b-4c84-acc9-70caa94e827b
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.70.3626
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-03-12

UEFI Device Firmware
  DeviceId:             c19580db344e418cf2b23297bcf46d75c06e432a
  Guid:                 b87a926d-189c-49a1-b1ff-921099b1de89
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.20
  VersionLowest:        0.1.20
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-03-12
________________________________________________

Devices that have been updated successfully:

 • 20L5CTO1WW System Firmware (0.1.30 → 0.1.31)
 • UEFI Device Firmware (184.60.3561 → 184.65.3590)
 • UEFI Device Firmware (184.65.3590 → 184.70.3626)
 • UEFI Device Firmware (0.1.16 → 0.1.19)
 • UEFI Device Firmware (0.1.19 → 0.1.20)
```

**Additional questions**
- Operating system and version: Ubuntu 18.04
- Have you tried rebooting? No
- Is this a regression? Don't know

superm1, 2020-03-12:
@smutel Can you try to manually use the tpm2 tools for getting the PCRs and see if they fail?  If they fail running by hand there isn't much that fwupd can do about it.

smutel, 2020-03-12:
Could you please tell me how to do this test manually ?

superm1, 2020-03-12:
Run one of these 3 commands (depending which is on your system):
https://github.com/fwupd/fwupd/blob/1_2_X/plugins/uefi/fu-uefi-pcrs.c#L149

smutel, 2020-03-13:
$ sudo tpm2_listpcrs
```
Failed to initialize tcti context: 0x1
```

smutel, 2020-03-13:
Seems to be the same issue than https://github.com/tpm2-software/tpm2-tools/issues/444.
I am still trying to understand what I need to do as workaround ...

superm1, 2020-03-13:
Try removing the `tpm2-abrmd` package and manually stop the service if it's running still and try it again.

smutel, 2020-03-13:
I don't have any process running and any package installed with this name

superm1, 2020-03-13:
I'm not sure then; but I would mention there isn't anything fwupd can do here in that instance.   You can remove the tpm2-tools package from your machine and fwupd will function just fine without it.

smutel, 2020-03-13:
OK. What about the message when I restart the fwupd service ?

superm1, 2020-03-13:
You can ignore it; it will be harmless

smutel, 2020-03-13:
OK. Fine.

closed 2020-03-13 by smutel