fwupd/fwupd #1860: Strange problem on Debian buster and a HP laptop

jsbien, 2020-03-13:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=953679

Any suggestions how to diagnose the problem? The Debian package maintainer gave up.

hughsie, 2020-03-13:
If you've getting "Error calling StartServiceByName" then what does `sudo /usr/libexec/fwupd --verbose` say?

jsbien, 2020-03-13:
On Fri, Mar 13 2020 at  1:27 -07, Richard Hughes wrote:
> If you've getting "Error calling StartServiceByName" then what does sudo /usr/libexec/fwupd/fwupd --verbose say?

No such file or directory...

/usr/lib/fwupd/fwupd --verbose

gives the enclosed output.

After the command the terminal behaves strangely - looks like the
command has not finished (I get no prompt), but I cannot kill it by
Ctrl-C.

BTW, I've tries to reinstall the package and got

--8<---------------cut here---------------start------------->8---
Setting up fwupd (1.2.5-2) ...
fwupd-offline-update.service is a disabled or a static unit, not starting it.
fwupd.service is a disabled or a static unit, not starting it.
--8<---------------cut here---------------end--------------->8---

-- 
             ,   
Janusz S. Bien
emeryt (emeritus)
https://sites.google.com/view/jsbien

superm1, 2020-03-13:
> Any suggestions how to diagnose the problem? The Debian package maintainer gave up.

I would just like to assert that my not knowing what to do and giving up are two distinctly different conclusions.

> gives the enclosed output.

At least from the debug output nothing looks off here.

>After the command the terminal behaves strangely - looks like the
command has not finished (I get no prompt), but I cannot kill it by
Ctrl-C.

Is your CPU getting pegged after this?  Can you check top?

jsbien, 2020-03-13:
On Fri, Mar 13 2020 at  6:12 -07, Mario Limonciello wrote:
>  Any suggestions how to diagnose the problem? The Debian package maintainer gave up.
>
> I would just like to assert that my not knowing what to do and giving up are two distinctly different conclusions.

I apologize for the wording. If you write quickly in a non-native
language sometimes you are not precise enough.

>
>  gives the enclosed output.
>
> At least from the debug output nothing looks off here.
>
>  After the command the terminal behaves strangely - looks like the
>  command has not finished (I get no prompt), but I cannot kill it by
>  Ctrl-C.
>
> Is your CPU getting pegged after this? Can you check top?

Neither top nor system monitor show anything unusual (I have a long task
running to find duplicate files on the disks).

The console with fwupd is still opened and there is no new output. The
process is still running

ps -a |grep fwupd
 8636 pts/0    00:00:00 fwupd

When I switch VT 1 I see several lines of some output and the screen
goes blank. I understand this is the debug information mentioned in the
first line of output

09:19:41:0650 FuDebug              Verbose debugging enabled (on console
1)

but how to make it readable?

Best regards

Janusz

-- 
             ,   
Janusz S. Bien
emeryt (emeritus)
https://sites.google.com/view/jsbien

superm1, 2020-03-13:
> Neither top nor system monitor show anything unusual (I have a long task
running to find duplicate files on the disks).

So the process will just sit there idle until something interacts with it.  It doesn't run into the background, but rather stays on the console.

You can run `fwupdmgr get-devices` in another terminal when you've manually launched it to see that working and more debug output will show up in the daemon window.

jsbien, 2020-03-13:
On Fri, Mar 13 2020 at  6:33 -07, Mario Limonciello wrote:
> Neither top nor system monitor show anything unusual (I have a long task
>  running to find duplicate files on the disks).
>
> So the process will just sit there idle until something interacts with it. [...]

No, the same timeout error.

> and more debug output will show up in the daemon window.

No additional output.

-- 
             ,   
Janusz S. Bien
emeryt (emeritus)
https://sites.google.com/view/jsbien

superm1, 2020-03-13:
> 09:19:41:0781 FuPlugin             performing coldplug() on synapticsmst

I notice that the last thing is this line above.
By chance does your machine have an AMD GPU?

If so, this sounds like https://github.com/hughsie/fwupd/issues/1121 with your old version of fwupd.  You have three options:

1. Upgrade to 1.3.8 from testing.
2. Upgrade to 1.3.9 from unstable.
3. Blacklist the synapticsmst plugin from loading.

jsbien, 2020-03-13:
On Fri, Mar 13 2020 at 10:19 -07, Mario Limonciello wrote:
>  09:19:41:0781 FuPlugin performing coldplug() on synapticsmst
>
> I notice that the last thing is this line above.
> By chance does your machine have an AMD GPU?
>
> If so, this sounds like #1121 with your old version of fwupd. You have three options:
>
> 1 Upgrade to 1.3.8 from testing.

Looks like it helped, I get now some output from fwupdmgr get-devices.
More testing tomorrow.

Thank you very much!

Best regards

Janusz

-- 
             ,   
Janusz S. Bien
emeryt (emeritus)
https://sites.google.com/view/jsbien

superm1, 2020-03-13:
I'd say it's likely that AMD issue then from your above symptoms.

closed 2020-03-13 by superm1