fwupd/fwupd #1861: /usr/lib/systemd/system-shutdown/fwupd.shutdown failed with exit status -127

freefreeno, 2020-03-14:
**Describe the bug**
/usr/lib/systemd/system-shutdown/fwupd.shutdown failed with exit status -127
On every reboot I see this message in Arch Linux

**Steps to Reproduce**
Reboot

**Expected behavior**
No errors during reboot

**fwupd version information**
$ fwupdmgr --version
client version: 1.3.9
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.3.9

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman in Arch Linux

**fwupd device information**
```
Precision 7540
│
├─Thunderbolt Controller:
│     Device ID:           6155c2b78f43685ec91f1b228ffe0214770eee20
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     41.01
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               98693963-c8d9-5f94-a80a-ce945716849f
│                          b982caab-eb37-5b45-9bf5-f92b70a93aac
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─PM981a NVMe Samsung 256GB:
│     Device ID:           4eaff2823dc972f445fd5eca658978950f45609d
│     Summary:             NVM Express Solid State Drive
│     Current version:     15302029
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               92129901-9091-586c-8986-cbd5e9d5523f
│                          4dd8e6a7-2e7f-884f-9879-73328600e2d3
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Samsung SSD 970 EVO Plus 250GB:
│     Device ID:           03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:             NVM Express Solid State Drive
│     Current version:     2B2QEXM7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          0a385028-4a26-5bc4-8282-40889897c5e8
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           cac9de2369fa3c52f34448afb98d95c9d18cc6d4
│     Current version:     1.7.0
│     Minimum Version:     1.7.0
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     GUID:                74992bad-d49d-4f82-bb77-bbe865bea34e
[... 121 lines cut ...]
└─UHD Graphics 630 (Mobile):
      Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
      Vendor:              Intel Corporation (PCI:0x8086)
      GUIDs:               12c60b98-52cd-500a-b99e-087aa6b47f0b ← PCI\VEN_8086&DEV_3E9B&REV_00
                           db677673-1102-5481-a11e-6f397e123ced ← PCI\VEN_8086&DEV_3E9B
      Device Flags:        • Internal device
                           • Cryptographic hash verification is available

**Additional questions**
This has been like this on two different Arch installs on same PC.

- Operating system and version:
Operating System: Arch Linux 
KDE Plasma Version: 5.18.3
KDE Frameworks Version: 5.67.0
Qt Version: 5.14.1
Kernel Version: 5.5.8-arch1-1
OS Type: 64-bit
Processors: 12 × Intel® Core™ i7-9750H CPU @ 2.60GHz
Memory: 15.4 GiB of RAM

- Have you tried rebooting?
This is when the error occurs
- Is this a regression?
This started for me seems like after the last update in Arch.

FFY00, 2020-03-14:
We also received a bug report downstream that seems to actually identiofy the issue.

https://bugs.archlinux.org/task/65814

freefreeno, 2020-03-14:
@FFY00 Ok thanks I have changed that line so mine should be fixed.

superm1, 2020-03-16:
This was fixed in master here https://github.com/fwupd/fwupd/commit/d9597df863ca21e0fb2ff821f0c612e0eb0558b2 and backported to 1_3_X as well.  I think about when we tag 1.4.0 we should probably also tag a 1.3.10.

closed 2020-03-16 by superm1