fwupd/fwupd #1862: Stopping of the fwupd-activate.service times out

estan, 2020-03-14:
**Describe the bug**

When the fwupd snap is installed, a hook installs an `fwupd-activate.service` systemd service.

When I reboot my computer, there's a 90 second delay due to the stopping of that service timing out.

**Steps to Reproduce**

On an up to date Kubuntu 18.04.4 system:

```
snap install --classic fwupd
<reboot>
```

There's a 90 second boot delay due to the stopping of `fwupd-activate.service` timing out.

I determined this as the cause of the shutdown delay by following the [systemd instructions for debugging shutdown](https://freedesktop.org/wiki/Software/systemd/Debugging/#index2h1).

Looking at the resulting [shutdown-log.txt](https://github.com/fwupd/fwupd/files/4332576/shutdown-log.txt) I could see:

```
[  302.531276] systemd[1]: fwupd-activate.service: Stopping timed out. Terminating.
[  302.531513] systemd[1]: fwupd-activate.service: Changed stop -> stop-sigterm
```

The full shutdown log linked above may have some more clues as to why it's timing out.

**Expected behavior**

Stopping of the `fwupd-activate.service` should be swift and not time out.

**fwupd version information**

Please provide the version of the daemon and client.
```shell
estan@edison:~$ fwupdmgr --version
client version: 1.3.9
compile-time dependency versions
        gusb:   0.2.11
        efivar: 34
daemon version: 1.3.9
estan@edison:~$
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `snap install fwupd`

**fwupd device information**

```shell
estan@edison:~$ fwupdmgr get-devices --show-all-devices
20KH007BMX
│
├─Thunderbolt Controller:
│     Device ID:           78ca7e95ac3ebdf65c703b654a2e2fca129f7d56
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               2da42a33-cd30-5ef5-a8fb-2c800a4b760f
│                          faf1991e-526e-59df-ab27-a3c6efb23ed5
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─SAMSUNG MZVLB512HAJQ-000L7:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     5L2QEXA7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          79d6cfae-a5a2-5936-9248-5aebd23480f7
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:     0.1.46
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:     7.63.12.18432
│ │   Vendor:              Infineon (TPM:IFX)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                        5eebb112-75ad-5536-b173-a11eb3399402
│ │                        ddf995da-1b32-5a8a-bc1b-8d5af4b38b51
│ │                        6d81ab63-db2e-50ac-934f-6be9accf5e02
│ │                        301555de-680d-5ddc-b995-7553fc9138f1
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       Vendor:            Infineon (TPM:IFX)
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:      • Internal device
│     
├─UEFI Device Firmware:
│     Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
│     Current version:     184.70.3626
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
      Current version:     0.1.20
      Minimum Version:     0.1.20
      Vendor:              DMI:LENOVO
      GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
    
estan@edison:~$
```

**Additional questions**

- Operating system and version: Kubuntu 18.04.4
- Have you tried rebooting? Yes :)
- Is this a regression? I don't know, I only started using fwupd from the snap recently. I know it happens on a colleages Kubuntu 18.04.4 laptop as well.

estan, 2020-03-14:
And just to clarify, `sudo systemctl stop fwupd-activate.service` when the system is running is fast. It seems to be only during shutdown that it times out.

superm1, 2020-03-14:
Sounds like a duplicate of #1654 

Perhaps stop for fwupdate-activate needs to be called earlier in shutdown. You can certainly experiment with modifying that unit and calling `systemctl daemon-reload`.

estan, 2020-03-14:
@superm1 Ah thanks, I somehow missed that issue. Indeed seems like a duplicate. I'll close this one. I may have time to tinker some with the unit file this weekend.

closed 2020-03-14 by estan