fwupd/fwupd #1863: Failed to obtain auth after successful upgrade Thinkpad t480s

kflak, 2020-03-14:
After running `fwupdmgr get-updates` I get a request to upload report with 4 alternatives. 0 runs without issue, but 1-3 gives me a `Failed to obtain auth` error. 

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```

Installed with pacman from Arch community

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20L8002UMH
│
├─Thunderbolt Controller:
│     Device ID:           da143c16aff44e7895fa7c15c1c2889e271a1f69
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               3b125a20-5ed2-5c0f-b22d-ab0e0b687bbc
│                          03f688aa-3369-58aa-94cf-83a11d9c3fe2
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Samsung SSD 970 EVO Plus 1TB:
│     Device ID:           c82980cd3fe66f7d0a8bb83b353cbc271177efd4
│     Summary:             NVM Express Solid State Drive
│     Current version:     1B2QEXM7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          f05ecac9-5cc2-5f77-b23b-a08ad8879ada
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           e7cc234e82d08b6a14e246beb6b5f8142091e570
│     Current version:     0.1.39
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                ebfe8df8-dee7-4692-a721-cbcf5095c5cf
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:     7.63.13.6400
│ │   Vendor:              Infineon (TPM:IFX)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                        5eebb112-75ad-5536-b173-a11eb3399402
│ │                        ddf995da-1b32-5a8a-bc1b-8d5af4b38b51
│ │                        6d81ab63-db2e-50ac-934f-6be9accf5e02
│ │                        301555de-680d-5ddc-b995-7553fc9138f1
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       Vendor:            Infineon (TPM:IFX)
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:      • Internal device
│     
├─UEFI Device Firmware:
│     Device ID:           cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
│     Current version:     184.70.3626
│     Minimum Version:     184.70.3626
│     Vendor:              DMI:LENOVO
│     GUID:                676af093-2a5c-4238-9c29-db8063a33532
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           323bda90831520867f8c02544efb9711dfca9fb6
│     Current version:     0.1.20
│     Minimum Version:     0.1.20
│     Vendor:              DMI:LENOVO
│     GUID:                4d254d6e-cd67-477b-97d5-bc3048af45c4
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UHD Graphics 620:
      Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
      Current version:     07
      Vendor:              Intel Corporation (PCI:0x8086)
      GUIDs:               fed2efa4-7045-55a1-b4fc-b29283d59fe5
                           8b72a10c-1279-5f8e-a28a-34fb11a58240
      Device Flags:        • Internal device
                           • Cryptographic hash verification is available
    

```

**Additional questions**
- Operating system and version: Arch Linux 5.4.24-rt15-2-rt
- Have you tried rebooting? Yes
- Is this a regression? Yes

superm1, 2020-03-14:
Are you running it over SSH or something similar workout a policy kit authentication frontend?

If so, please run command with "sudo"

kflak, 2020-03-25:
Hi there, sorry about dropping off the radar, been a hectic and weird time lately... Thanks for the sudo tip, I am able to do upload the report that way!

closed 2020-03-25 by kflak