fwupd/fwupd #1864: Add support for detecting security flawed firmware and just reporting instructions or information to user.

oiaohm, 2020-03-15:
**Is your feature request related to a problem? Please describe.**
The problem is how do I rapidly detect I have machines with know bad firmware that can and cannot be updated without needing to run special software?   Like particular Intel ME versions are know bad but you need a full motherboard firmware update of some form to fix this.

Also the cases of like a NVME drive that needs to be cleared after firmware update so you need to back up data before firmware update.   This is a case you don't want to update firmware straight now but give end user message sorry there is a problem here are the actions you need to-do to fix. 

**Describe the solution you'd like**
fwupd is all ready processing hardware checks to work out if it should or should not apply update.   

It would be nice if fwupd support add metadata that does not have a update the firmware but checks hardware and simple reports if hardware is bad or placed into a special state before updated with instruction text for end user to deal with problem.  

Yes these instructions could include the instruction to force the issue once the device is in a state to have the update done.

Bad intel ME version for example fwupd could tell end user they have it and check for a motherboard update even in the case fwupd does not support the motherboard directly.

**Describe alternatives you've considered**
Really a lot of the other ideas I have looked at like inventory tools and so on require running extra software that general users are not going run.

**Additional context**
This is a double sided thing.   This project wants vendors to provide the project with updates.   End users knowing that vendor should be providing them with update and they have not means end users will be pressure point to vendor and helping end user to know when vendor is not protecting them.

So basically implementing above allows listing vendors that cannot provide updates for because they are not providing updates and their products are insecure to put more pressure on those vendors to provide updates. 

It also allows for the cases that vendor knows that X update to fix a issue has some serous horrible requirements to be done to protect user from harm from the update.

Do also remember some motherboard vendors these days don't want you updating firmware by OS but have the firmware only bootloader do it.   So in this case cab file contains a file that need to be transferred to a USB fat32 to be placed in the flash USB port then user does action.

There could be other devices in time that update their firmware themselves where the user need to be informed that the device need to be powered off and rebooted or equal to apply updated firmware.

I know it would be nice for everything to be automatic but the reality there are a lot of cases where detect report issue/instructions to fix is the correct answer.  Since  fwupd is already doing hardware detection to work out if or if not it should apply updates I don't see it as much of extend to take this hardware detection and instead of update give users information/instructions instead of straight update.

hughsie, 2020-03-16:
> could tell end user they have it

How would this be done? As a message on the command line? In the GUI?

> Bad intel ME version

We're planning to add this data to the LVFS, but I'd not thought about exporting it in the metadata. One to ponder I guess.

hughsie, 2020-08-18:
> Bad intel ME version

We're doing this as part of HSI now.

closed 2020-08-18 by hughsie