fwupd/fwupd #1866: Missing dependency

siro20, 2020-03-15:
**Describe the bug**
Missing dependency "libgusb-devel". Ninja fails to build fwupd.

**Steps to Reproduce**
Clone the repo, run meson, install all requested packages and the run ninja on a fresh Fedora 31.

**Expected behavior**
ninja is able to build fwupd.

superm1, 2020-03-15:
When you say install all requested packages, did you follow instructions at compilation wiki page?

https://github.com/fwupd/fwupd/wiki/Compilation
```./contrib/ci/generate_dependencies.py```

siro20, 2020-03-16:
I didn't run  `./contrib/ci/generate_dependencies.py` as I wasn't aware of this nice script. I would expect meson to complain if this package is missing, but it doesn't.

superm1, 2020-03-16:
So currently there is a dependency for `libgusb`: https://github.com/fwupd/fwupd/blob/master/meson.build#L190

Can you share the full error message if you still have it?  Otherwise I don't think there is anything else we can do.

closed 2020-03-16 by superm1

siro20, 2020-03-17:
Meson reports:

> Dependency gusb found: YES (cached)

But it fails with the following error:

> cc1: warning: /usr/include/gusb-1: No such file or directory [-Wmissing-include-dirs]
In file included from ../libfwupdplugin/fu-plugin-vfuncs.h:9,
                 from ../plugins/colorhug/fu-plugin-colorhug.c:9:
../libfwupdplugin/fu-plugin.h:10:10: fatal error: gusb.h: No such file or directory
   10 | #include <gusb.h>
      |          ^~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

superm1, 2020-03-17:
Which package contains the .pc file? It should have been in the development package

siro20, 2020-03-21:
The .pc is in the devel package. No idea what's wrong here.

superm1, 2020-03-23:
Did you perhaps have gusb-devel installed previously and remove it between builds of fwupd in a local source tree clone?