fwupd/fwupd #1870: allow "fwupdmgr update DEVICE-ID"

evgeni, 2020-03-17:
**Is your feature request related to a problem? Please describe.**
When there are multiple updates available (for different devices), I'd prefer to be able to apply them one by one, instead of all at once. Thus making it easier to know where to look  if something is broken after an update.

**Describe the solution you'd like**
`fwupdmgr update DEVICE-ID` should work next to `fwupdmgr update`

**Describe alternatives you've considered**
`--filter DEVICE-ID`? Not sure that's a good UX tho.

**Additional context**
https://twitter.com/zhenech/status/1238726782209097729

@hughsie suggested it's only a doc issue, but looking at the code, I am not 100% sure

it seems to be implemented:
https://github.com/fwupd/fwupd/blob/252b0c704ed72908c74ee5528d2bb3ad126f4d19/src/fu-util.c#L1992-L1995

but `values` is never passed?
https://github.com/fwupd/fwupd/blob/252b0c704ed72908c74ee5528d2bb3ad126f4d19/src/fu-util.c#L2543-L2548

hughsie, 2020-03-17:
> but values is never passed?

It's passed from the cmd_array helper. Does the PR help?

evgeni, 2020-03-17:
> It's passed from the cmd_array helper. Does the PR help?

Ah, interesting, then I've miss-read your code.

Yeah, the PR seems to do what I had in mind.

closed 2020-03-17 by hughsie