fwupd/fwupd #1876: logitech-hidpp refactoring

FFY00, 2020-03-17:
So, I am looking at the code and there are a few thing I think could be improved. I would like to refactor it in order to make it easier to support different devices.

Some things I would like to be able to achieve in the future:
  - Easily add more receivers
  - Support wired HID++ 2.0 devices

I have some questions:

1) The receiver bootloader protocol identification routine ({texas,nordic}/{,un}signed) seems to be split between `fu_logitech_hidpp_runtime_setup_internal` (usigned/unsigned) and `fu_logitech_hidpp_runtime_probe` (texas/nordic). Is there any reason for this?

2) To prevent all this regex craziness, I think we should mark have 4 different protocols, not just `com.logitech.unifying{,signed}`. Would it be possible to make this change? As this requires changes to the both metadata and the code we would need a min and max deamon version field in the metadata? This way we could keep updates for older daemons in the server. Is there anything like this in place? If not, is it doable to add>

3) As said in #1384, we will make the kernel issue a udev update event when an unavailable device becomes online again. Does fwupd already listen to those events?

superm1, 2020-03-17:
2. I think I'd lean upon instead implementing support for a payload to be used for multiple protocols, particularly if the daemon on older versions is able to recognize the first one and use that.  We got lucky with version format working that way.
Then we can make a switch on LVFS side to add a second protocol to all payloads as needed.

3. Yeah, the plugin just needs to indicate which subsystem of events it needs to pick up.  You can see that a bunch of other plugins do this already too.

FFY00, 2020-04-17:
/notstale

closed 2020-05-24 by stale[bot]