fwupd/fwupd #1877: Logitech peripherals not showing up in fwupd

superm1, 2020-03-18:
**Describe the bug**
I noticed with a new K830 keyboard hooked up through a unifying dongle that the keyboard is not actually enumerated by fwupd.

It is however enumerated by solaar.

**Steps to Reproduce**
`# fwupdtool get-devices --plugin-whitelist=logitech_hidpp`

**fwupd version information**
c0f1f1c7e3adbef4346807528b7b9d515118b059

I dug around a little bit, and see it's broken at least as far back as d9f52c75738939cc9a6edcf268cc139904523c98 which from commit history seems like the time it should have been working.

I think what is happening is that the quirks are tagging the GType for the peripheral as a `FuLogitechHidPpRuntime` since it happens to have the same VEN/DEV as the unifying receiver it's connected through.   This seems to follow with the device being added "twice" too.

```
04:46:08:0499 FuEngine             using plugins: logitech_hidpp
04:46:08:0500 FuEngine             failed to find specified plugin dell: no plugin dell found
04:46:08:0502 FuEngine             failed to find specified plugin dell_dock: no plugin dell_dock found
04:46:08:0503 FuEngine             failed to find specified plugin dell_dock: no plugin dell_dock found
04:46:08:0519 FuEngine             6 devices with subsystem hidraw
04:46:08:0520 FuEngine             failed to find specified plugin synaptics_rmi: no plugin synaptics_rmi found
04:46:08:0521 FuDevice             using 279ed287-3607-549e-bacc-f873bb9838c4 for HIDRAW\VEN_046D&DEV_C52B
04:46:08:0522 XbSilo               0.02ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='d25c014a-6dea-58cf-ac0f-762daf8be6ab']` limit=1 -> 0 results
04:46:08:0522 XbSilo               0.02ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='82b088b2-dc64-5a93-abb3-d9f2cfa14fc0']` limit=1 -> 0 results
04:46:08:0522 XbSilo               0.02ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='63cf2859-c965-5a77-9353-b1047125d7d9']` limit=1 -> 0 results
04:46:08:0522 FuPlugin             d25c014a-6dea-58cf-ac0f-762daf8be6ab,82b088b2-dc64-5a93-abb3-d9f2cfa14fc0,63cf2859-c965-5a77-9353-b1047125d7d9 has no updates, so ignoring device
04:46:08:0523 FuDevice             already set GType to FuLogitechHidPpRuntime, ignoring FuLogitechHidPpPeripheral
04:46:08:0524 FuDevice             already set GType to FuLogitechHidPpRuntime, ignoring FuLogitechHidPpRuntime
04:46:08:0524 FuDevice             already set GType to FuLogitechHidPpRuntime, ignoring FuLogitechHidPpPeripheral
04:46:08:0524 FuDevice             using 4e419dded9789c89c0050b3537950d03d13eb7a9 for DEVPATH=/devices/pci0000:00/0000:00:1d.0/0000:03:00.0/0000:04:04.0/0000:3c:00.0/0000:3d:02.0/0000:3e:00.0/usb5/5-2/5-2.3/5-2.3.1/5-2.3.1:1.2
04:46:08:0731 FuPlugin             emit added from logitech_hidpp: 4e419dded9789c89c0050b3537950d03d13eb7a9
04:46:08:0732 FuDeviceList         ::added 4e419dded9789c89c0050b3537950d03d13eb7a9
04:46:08:0732 FuMain               ADDED:
FuLogitechHidPpRuntime:
Unifying Receiver
  DeviceId:             4e419dded9789c89c0050b3537950d03d13eb7a9
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 fcf55bf5-767b-51ce-9c17-f6f538c4ee9f <- HIDRAW\VEN_046D&DEV_C52B&REV_00
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4 <- HIDRAW\VEN_046D&DEV_C52B
  Summary:              A miniaturised USB wireless receiver
  Plugin:               logitech_hidpp
  Protocol:             com.logitech.unifying
  Flags:                updatable|registered
  VendorId:             USB:0x046D
  Version:              RQR12.10_B0032
  VersionBootloader:    BOT01.02_B0015
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      30
  Created:              2020-03-18
  PhysicalId:            DEVPATH=/devices/pci0000:00/0000:00:1d.0/0000:03:00.0/0000:04:04.0/0000:3c:00.0/0000:3d:02.0/0000:3e:00.0/usb5/5-2/5-2.3/5-2.3.1/5-2.3.1:1.2
  SignedFirmware:        false

04:46:08:0732 XbSilo               0.23ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='9d131a0c-a606-580f-8eda-80587250b8d6']/../..|components/component/provides/firmware[@type='flashed'][text()='fcf55bf5-767b-51ce-9c17-f6f538c4ee9f']/../..|components/component/provides/firmware[@type='flashed'][text()='279ed287-3607-549e-bacc-f873bb9838c4']/../..` limit=1 -> 0 results
04:46:08:0732 FuHistory            get device
04:46:08:0735 FuDevice             already set GType to FuLogitechHidPpRuntime, ignoring FuLogitechHidPpRuntime
04:46:08:0736 FuDevice             already set GType to FuLogitechHidPpRuntime, ignoring FuLogitechHidPpPeripheral
04:46:08:0737 FuDevice             using 4e419dded9789c89c0050b3537950d03d13eb7a9 for DEVPATH=/devices/pci0000:00/0000:00:1d.0/0000:03:00.0/0000:04:04.0/0000:3c:00.0/0000:3d:02.0/0000:3e:00.0/usb5/5-2/5-2.3/5-2.3.1/5-2.3.1:1.2
04:46:08:0945 FuPlugin             emit added from logitech_hidpp: 4e419dded9789c89c0050b3537950d03d13eb7a9
04:46:08:0946 FuDeviceList         device 4e419dded9789c89c0050b3537950d03d13eb7a9 already exists, ignoring
04:46:08:0946 XbSilo               0.23ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='9d131a0c-a606-580f-8eda-80587250b8d6']/../..|components/component/provides/firmware[@type='flashed'][text()='fcf55bf5-767b-51ce-9c17-f6f538c4ee9f']/../..|components/component/provides/firmware[@type='flashed'][text()='279ed287-3607-549e-bacc-f873bb9838c4']/../..` limit=1 -> 0 results
04:46:08:0946 FuHistory            get device
04:46:08:0951 XbSilo               0.05ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='9d131a0c-a606-580f-8eda-80587250b8d6']/../..|components/component/provides/firmware[@type='flashed'][text()='fcf55bf5-767b-51ce-9c17-f6f538c4ee9f']/../..|components/component/provides/firmware[@type='flashed'][text()='279ed287-3607-549e-bacc-f873bb9838c4']/../..` limit=1 -> 0 results
04:46:08:0951 FuEngine             Emitting PropertyChanged('Status'='idle')
Loading…                 [***************************************]
04:46:08:0951 XbSilo               ignoring for OR statement: indexed string '9d131a0c-a606-580f-8eda-80587250b8d6' was unfound
04:46:08:0951 XbSilo               ignoring for OR statement: indexed string 'fcf55bf5-767b-51ce-9c17-f6f538c4ee9f' was unfound
04:46:08:0951 FuCommon             creating path /home/supermario/src/fwupd/dist/cache/fwupd/motd.d
04:46:08:0964 FuMain               Unifying Receiver
  DeviceId:             4e419dded9789c89c0050b3537950d03d13eb7a9
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 fcf55bf5-767b-51ce-9c17-f6f538c4ee9f <- HIDRAW\VEN_046D&DEV_C52B&REV_00
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4 <- HIDRAW\VEN_046D&DEV_C52B
  Summary:              A miniaturised USB wireless receiver
  Plugin:               logitech_hidpp
  Protocol:             com.logitech.unifying
  Flags:                updatable|registered
  VendorId:             USB:0x046D
  Version:              RQR12.10_B0032
  VersionBootloader:    BOT01.02_B0015
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      30
  Created:              2020-03-18
```

hughsie, 2020-03-19:
Urgh. Perhaps an ugly fix; could we move the `HIDRAW\VEN_046D` type `FuLogitechHidPpPeripheral` **up** the file so that it gets probed first?

hughsie, 2020-03-19:
I won't have access to hardware until tomorrow afternoon, but I can test myself then.

superm1, 2020-03-19:
Wouldn't that cause problems for receivers?  Or I guess at least the probing would fail as a receiver in the peripheral code and continue possibly.

hughsie, 2020-03-19:
My thinking there is that the probe will fail, so it might be _noisy_, but work.

superm1, 2020-03-20:
I gave that a try; Just re-ordering in the quirk file isn't enough it seems.

FFY00, 2020-03-22:
A quick test with a K400 Plus shows that it is not detected.

During the summer, I got it to show up somehow, I don't remember exactly.

hughsie, 2020-03-26:
I see:

      FuLogitechHidPpPeripheral:
    Logitech K780
      DeviceId:             bdb162e4e35f17a9fc595ae66377b7b469b1d5ce
      ParentDeviceId:       d3fae86d95e5d56626129d00e332c4b8dac95442
      Guid:                 0c4bc668-1df8-536b-b6d1-87c825d42842 <- HIDRAW\VEN_046D&DEV_405B&REV_00
      Guid:                 4c10fc7b-08c5-5ea6-a6b1-cabf1b210f6b <- HIDRAW\VEN_046D&DEV_405B
      Guid:                 3932ba15-2bbe-5bbb-817e-6c74e7088509 <- UFY\VID_046D&PID_405B
      Plugin:               logitech_hidpp
      Protocol:             com.logitech.unifying
      Flags:                registered|only-supported
      VendorId:             USB:0x046D
      VersionFormat:        plain
      Icon:                 preferences-desktop-keyboard
      Created:              2020-03-26
        PhysicalId:          HID_PHYS=usb-0000:00:14.0-4.4.2.2/input2:2
        HidppVersion:        0
        HidppId:             0xfe
        BatteryLevel:        0
        IsUpdatable:         false
        IsActive:            false

So there, but `IsUpdatable:         false` is certainly weird. Debugging.

hughsie, 2020-03-26:
Ohh, if I actually wake it up I get the slightly more expected:

    └─Unifying Receiver:
      │   Device ID:           d3fae86d95e5d56626129d00e332c4b8dac95442
      │   Summary:             A miniaturised USB wireless receiver
      │   Current version:     RQR24.03_B0027
      │   Bootloader Version:  BOT03.01_B0008
      │   Vendor:              USB:0x046D
      │   Install Duration:    30 seconds
      │   GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
      │                        fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
      │                        279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      │   Device Flags:        • Updatable
      │                        • Supported on remote server
      │ 
      └─Logitech K780:
            Device ID:         bdb162e4e35f17a9fc595ae66377b7b469b1d5ce
            Summary:           Unifying Keyboard
            Current version:   MPK01.03_B0024
            Bootloader Version:BOT25.00_B0005
            Vendor:            USB:0x046D
            GUIDs:             0c4bc668-1df8-536b-b6d1-87c825d42842 ← HIDRAW\VEN_046D&DEV_405B&REV_00
                               4c10fc7b-08c5-5ea6-a6b1-cabf1b210f6b ← HIDRAW\VEN_046D&DEV_405B
                               3932ba15-2bbe-5bbb-817e-6c74e7088509 ← UFY\VID_046D&PID_405B
            Device Flags:      • Updatable
                               • Supported on remote server

superm1, 2020-03-26:
Huh.. I'm not seeing that for either of my two keyboards or mouse with master at all.

superm1, 2020-03-26:
Oh hmm looking through your above it perhaps because of `FWUPD_DEVICE_FLAG_ONLY_SUPPORTED` that I'm not seeing any of my stuff show up? 

@FFY00 is solar doing something similar?  Or is it opening everything?  All my devices show up there so at least the stuff on my side it's safe enough items.

FFY00, 2020-03-26:
I think this all makes sense, if the device is not active (offline) we try to ping it and it doesn't respond. This is the kernel issue we have been talking about, the device node will get exported by the `hid-logitech-dj` kernel driver even if the device it not active (online).

> `IsActive:            false`

superm1, 2020-03-26:
I guess @hughsie  and I are experiencing different things though.
Mine don't show up at all though in fwupd, and they're definitely online.

FFY00, 2020-03-26:
Something wrong is going on here, my K400+ is also not showing up.

It seems like the behavior @hughsie is experiencing is the intended.

commit ce9f7de2 (2020-03-30): Add a new daemon configuration option to control EnumerateAllDevices

For plugins that are configured to only enumerate devices supported
by metadata this will allow showing them in all contexts

Fixes: #1877
commit 370010d7 (2020-03-30): Add a new daemon configuration option to control EnumerateAllDevices

For plugins that are configured to only enumerate devices supported
by metadata this will allow showing them in all contexts

Fixes: #1877
commit 8e38513d (2020-03-31): Add a new daemon configuration option to control EnumerateAllDevices

For plugins that are configured to only enumerate devices supported
by metadata this will allow showing them in all contexts

Fixes: #1877
PR #1917 (2020-03-31): logitech_hidpp: show devices for all peripherals not just supported ones
At least to discuss.  This fixes all my Logitech peripherals not showing up.  I finally now understand the experience for stuff not currently connected too and why the DJ changes are important.

@FFY00 I think that the plugin will need to respond to change events to update version string (both connect and disconnect most likely).

Fixes: #1877

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Docume [... cut]
closed 2020-03-31 by superm1