fwupd/fwupd #1881: Dell laptop dock: Update hangs at "Restarting device" state post update

peeyush-sharma, 2020-03-18:
**Describe the bug**
Update operation "hangs" after updating dell Dock firmware, maybe the timeout values are too large?

**Steps to Reproduce**
sudo fwupdmgr update
• Thunderbolt Controller has the latest available firmware version
• Dell WD15 has no available firmware updates
• Dell WD15 Passive Cable has no available firmware updates
• Dell WD15 Port Controller 1 has no available firmware updates
• SAMSUNG SSD PM871b M.2 2280 256GB has no available firmware updates
• System Firmware has the latest available firmware version
Upgrade available for VMM3332 inside Dell WD15/TB16/TB18 wired Dock from 3.10.02 to 3.12.002
VMM3332 inside Dell WD15/TB16/TB18 wired Dock and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading 3.12.002 for VMM3332 inside Dell WD15/TB16/TB18 wired Dock...
Fetching firmware https://fwupd.org/downloads/e806066e4d61970b6e764bb7e9b28288493b0b62-mst_3_12_002.cab
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating VMM3332 inside Dell WD15/TB16/TB18 wired Dock…          ]
Restarting device…       [***************************************] Less than one minute remaining…

**Expected behavior**
The command should have exited successfully/failed with a message post update.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
Latitude 7480
│
├─Thunderbolt Controller:
│     Device ID:           d400caf5d2fbf470e06475e9d5a1d8e3990861db
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     33.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               afbdf062-a250-54d2-b49e-a3c1ebd9493c ← TBT-00d407a0
│                          2a62071d-a3e3-5361-82d1-bf5b1b7c767f ← TBT-00d407a0-0000:05:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Dell WD15:
│ │   Device ID:           ed9027c86c336ba5d3312f83c06f78b2ea6ef915
│ │   Summary:             A USB type-C docking station
│ │   Current version:     0.0.0
│ │   Vendor:              Dell Inc. (PCI:0x1028)
│ │   GUID:                e7ca1f36-bf73-4574-afe6-a4ccacabf479
│ │   Device Flags:        • Updatable
│ │                        • Requires AC power
│ │                        • Needs a reboot after installation
│ │ 
│ ├─Dell WD15 Passive Cable:
│ │     Device ID:         29bca9f02e33e23330ae801ae7d6c22eb2b758e0
[... 99 lines cut ...]
    │   Summary:           Alternate mode for platform TPM device
    │   Vendor:            Dell Inc. (PCI:0x1028)
    │   Update Error:      mode switch disabled due to TPM ownership
    │   GUID:              f06e62b7-7039-5d5d-9c65-f7fccacbafbf ← 07a0-1.2
    │   Device Flags:      • Internal device
    │                      • Requires AC power
    │                      • Device is locked
    │ 
    └─TPM 1.2 System Update:
          New version:     89195009
          Remote ID:       lvfs
          Summary:         Firmware for the Dell TPM 1.2
          License:         Proprietary
          Size:            554.9 kB
          Vendor:          Dell Inc.
          Description:     Initial release
        
Upload message: cd9163cf5e9b24610d4a6d20ab13f31218efc549 replaces old report
Successfully uploaded 1 report
```

**Additional questions**
- Operating system and version: Arch Linux, x86_64 Linux 5.5.9-zen1-2-zen
- Have you tried rebooting? NO
- Is this a regression? Unknown

superm1, 2020-03-18:
After it hung, what happened eventually?  Did it timeout?  Can you please check your `fwupd.service` journal log to see if anything was indicated?

peeyush-sharma, 2020-03-18:
It did not timeout.

Here's the service journal log:

```shell
ps@ps-laptop ~> sudo journalctl -u fwupd.service -b
[sudo] password for ps: 
-- Logs begin at Mon 2019-06-10 15:40:52 MDT, end at Wed 2020-03-18 12:13:53 MDT. --
Mar 16 09:39:13 ps-laptop systemd[1]: Starting Firmware update daemon...
Mar 16 09:39:15 ps-laptop fwupd[1566]: 15:39:15:0294 FuEngine             device 64ec4cd1e1c9565e79b00f4e7221b1c689b33e96 [SAMSUNG SSD PM871b M.2 2280 256GB] does not define a vendor-id!
Mar 16 09:39:15 ps-laptop systemd[1]: Started Firmware update daemon.
Mar 16 09:39:15 ps-laptop fwupd[1566]: 15:39:15:0948 FuPluginThunderbolt  got change event for unknown device, adding instead
ps@ps-laptop ~>
```

superm1, 2020-03-18:
So the update was successful? Or it failed?

peeyush-sharma, 2020-03-18:
I am assuming that the update itself seems to have succeeded, since during the update process the dock-connected devices like the monitors: glitched -> went offline -> came back okay.

Additionally the command: _fwupdmgr get-devices --show-all-devices_
reports that the VMM3332 version is 3.12.02 (which is the expected updated version)

I haven't tried restarting the dock/laptop just yet though so that part of the equation is still unknown.

superm1, 2020-03-18:
Ok so specifically you were just expecting a message that firmware upgraded successfully or failed rather than just exit codes?

peeyush-sharma, 2020-03-18:
Okay, so I was expecting:

1. For the command to actually exit (it still hasn't exited yet).
2. Provide an effective status post-exit, it may have been a statuscode or a message string or anything non-null really, just not a silent non-informative exit.

superm1, 2020-03-18:
> For the command to actually exit (it still hasn't exited yet).

Huh?  well that's certainly not expected.  So you checked in another terminal for the status and were able to?  The deamon is single threaded, so I am confused how that could have happened.

Would be possible to attach gdb to your process that's hanging and see where it is spinning?

> Provide an effective status post-exit, it may have been a statuscode or a message string or anything non-null really, just not a silent non-informative exit.

Actually I forgot; we did add that in already.  It's supposed to be there in 1.3.9, you should see this when the process exits (with a success return):
https://github.com/fwupd/fwupd/blob/master/src/fu-util.c#L581

So we need to figure out why your frontend is hanging.

peeyush-sharma, 2020-03-18:
Does this help?

```shell
ps@ps-laptop ~> ps -a
    PID TTY          TIME CMD
   1145 tty2     00:00:00 gnome-session-b
 116518 pts/2    00:00:00 sudo
 116519 pts/2    00:00:00 fwupdmgr
 118189 pts/0    00:00:00 sudo
 118199 pts/0    00:00:01 openconnect
 121275 pts/3    00:00:00 ps
ps@ps-laptop ~> sudo gdb -a 116519
[sudo] password for ps: 
gdb: option '-a' is ambiguous; possibilities: '-annotate' '-args'
Use `gdb --help' for a complete list of options.
ps@ps-laptop ~ [1]> sudo gdb -p 116519
GNU gdb (GDB) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Attaching to process 116519
[New LWP 116520]
[New LWP 116521]
[New LWP 116534]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
0x00007fa1a9627abf in poll () from /usr/lib/libc.so.6
(gdb) where full
#0  0x00007fa1a9627abf in poll () at /usr/lib/libc.so.6
#1  0x00007fa1a995e7a0 in  () at /usr/lib/libglib-2.0.so.0
#2  0x00007fa1a995f843 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#3  0x00007fa1a9c92d73 in fwupd_client_install () at /usr/lib/libfwupd.so.2
#4  0x000056405be31c20 in  ()
#5  0x000056405be3645f in  ()
#6  0x000056405be3aa56 in  ()
#7  0x000056405be30da1 in  ()
#8  0x00007fa1a955a023 in __libc_start_main () at /usr/lib/libc.so.6
#9  0x000056405be3108e in  ()
(gdb)
```

superm1, 2020-03-18:
I guess running bt won't actually get you line numbers in `fwupd_client_install` will it?

superm1, 2020-03-18:
That's basically just meaning it's hanging out here: https://github.com/fwupd/fwupd/blob/7d82a0980860ad1bd785a40a416ae8d2f879d1ad/libfwupd/fwupd-client.c#L1085

peeyush-sharma, 2020-03-18:
I'll be glad to to help you folks debug it, just LMK.

superm1, 2020-03-18:
That's great thanks for the offer.  @hughsie thoughts on this?

I guess the main loop is supposed to have quit when the daemon sent a proxy message back, but that never happened for some reason for you.
https://github.com/fwupd/fwupd/blob/7d82a0980860ad1bd785a40a416ae8d2f879d1ad/libfwupd/fwupd-client.c#L975

Is this your first firmware update on this system?

peeyush-sharma, 2020-03-18:
No this isn't my first fwupd on this system (I have run it several times over the last year or so)

peeyush-sharma, 2020-03-23:
Another problematic byproduct of this hang was the message I kept getting on trying to reboot.
```shell
ps@ps-laptop ~> reboot
Operation inhibited by "fwupd" (PID 1566 "fwupd", user root), reason is "Firmware Update in Progress".
Please retry operation after closing inhibitors and logging out other users.
Alternatively, ignore inhibitors and users with 'systemctl reboot -i'.
ps@ps-laptop ~ [1]>
```

closed 2020-04-29 by stale[bot]