fwupd/fwupd #1892: Add a plugin for Steelseries device

FFY00, 2020-03-22:
I have reverse engineered the update protocol for the Steelseries Rival 310 (inc. the howl and pubg variants) and Sensei 310 mice. I suspect similar models might be using the same protocol, I would need a Wireshark dump to confirm.

Does it make sense to write a plugin for them?

Perhaps it would be best to see if Steelseries is interested in helping out. If we want to put the firmware in LVFS we would still need to ask them for permission to distribute the binaries.

FFY00, 2020-03-22:
Totally forgot, a plugin already exists, but it is only able to get the firmware version.

hughsie, 2020-03-23:
I'd be totally okay with adding the writing code to the existing plugin. I did ask SteelSeries a few years ago if they wanted to upload firmware to the LVFS, but they declined. Maybe 14 million firmware downloads later they've changed their mind? :)

FFY00, 2020-03-23:
Do you think it makes sense to write the code before asking Steelseries? I have other things to do and I would like to avoid this being just a wait of time.

If they are not interested in uploading the firmware themselves, could we do it instead?

hughsie, 2020-03-23:
> Do you think it makes sense to write the code before asking Steelseries

I think if you were to email them asking if they wanted to do the upload, and for you to write the code it might work.

> If they are not interested in uploading the firmware themselves, could we do it instead?

No, we are only allowed to redistribute nonfree firmware files with explicit permission.

FFY00, 2020-03-23:
> I think if you were to email them asking if they wanted to do the upload, and for you to write the code it might work.

Wouldn't it make more sense for you to email them as you run LVFS? You can mention that I am willing to write the code. Also, my @archlinux.org or @gmail.com addresses might not bring as much attention as yours.

> No, we are only allowed to redistribute nonfree firmware files with explicit permission.

Yes, obviously. I am trying to understand if there's any interest from our part to do it, at which point we would ask them for permission.

hughsie, 2020-03-23:
> Also, my @archlinux.org or @gmail.com addresses might not bring as much attention as yours

Fair point. I've emailed my contact just now, I'll let you know what he says if there's anything public he can share.

superm1, 2020-08-13:
Did this go anywhere?  If not, I think we should close the issue.  If the company has no interest in fwupd support we shouldn't waste our eyeball cycles on them.

FFY00, 2020-08-15:
I am planning to write a plugin. @openinput-fw will provide open source firmware for the Steelseries Rival 310, I would like for fwupd to be able to flash the firmware. We will probably also write a custom bootloader, but I want the users to at least be able to use fwupd to manually flash the bootloader replacement payload. They should still be able to keep the original bootloader if they want.

superm1, 2020-10-22:
So to me this makes sense to adopt with the new switch branch that landed.  You would make a new branch of firmware (say "oss") and then this proposed plugin would allow switching to it.

FFY00, 2020-10-22:
I was thinking of basing the firmware on the vid/pid. The original one would use the official firmware, but still allow manually updating by supplying a `.cap` file. The custom firmware would have its own USB ID (via https://github.com/pidcodes/pidcodes.github.com or similar, we would also have open hardware and would share the ID) which could get a custom vendor in LVFS.

But we can also go the option in plugin route if we have trouble securing the USB ID.

superm1, 2020-10-22:
By doing it with the switch-branch stuff it doesn't matter if it uses the same VID/PID or not (IIRC).  You can keep it the same, but have a marker in it that indicates it's the OSS firmware.  And then the plugin can look for that marker and know to treat it differently.

FFY00, 2020-10-22:
That works, sure. I will look into it when I start :sweat_smile:

hughsie, 2021-01-19:
@FFY00 is this still something you want to do?

closed 2021-01-19 by hughsie