fwupd/fwupd #1893: AIAIAI H05 headband no longer detected.

Salt-Factory, 2020-03-22:
**Describe the bug**
When I plug in my AIAIAI H05 headband, fwupd is unable to detect it.
I tracked down the commit where this bug started: c04f5a3e151cac589d48f7abee6faed4e4478d85
In the commit before, c134451aebfe8ff9e010a8007c8986a3284ad9fe, the device is still detected.
If I copy the device-ID from the older commit, and try to run `fwupdmgr update ID` in newer commits, it also reports that it failed to find the device.

I wish that I could find the bug myself, but for that I am sadly not experienced enough with this project.

**Steps to Reproduce**
Build c04f5a3e151cac589d48f7abee6faed4e4478d85, plug in headband and set it to pair mode.

**Expected behavior**
The headband should show up in the device list, as it did before.

## After commit c134451aebfe8ff9e010a8007c8986a3284ad9fe, when it worked:

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.8-124-gc134451a
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.8-124-gc134451a
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Installed from source.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices

Z97X-UD5H
│
├─88SE9172 SATA 6Gb/s Controller:
│     Device ID:           0fba4a8bedba53076fc9b147655e59455bc25a87
│     Current version:     12
│     Vendor:              Marvell Technology Group Ltd. (PCI:0x1B4B)
│     GUIDs:               4a090d9b-362e-580e-b0bc-47003768f3e6
│                          76a209d6-e1dc-598a-b9e9-a8dd911da340
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─BLUETOOTH USB +EDR ADAPTER v2.1 UHE:
│     Device ID:           5d00620651c7c80da4d93a876b5d3a6138b0da7a
│     Current version:     3.88
│     Vendor:              Broadcom Corp (USB:0x050D)
│     GUIDs:               ef2af201-4ca1-5b3d-b36a-c00e8aec666b
│                          ba1e3114-39b3-57a0-b44a-3b1b36cfcf95
│     Device Flags:        • Updatable
│   
├─GM204 [GeForce GTX 970]:
│     Device ID:           71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Current version:     a1
│     Vendor:              NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:               c32e752c-53d9-53dd-a7ab-f7b9a72e9b7d
│                          e6857de4-0602-5690-90d1-df576c354a9b
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─H05:
│     Device ID:           4745cb127beea10281899d84cd01952ed74156a6
│     Summary:             Bluetooth Headphones
│     Current version:     1.5
│     Vendor:              AIAIAI (USB:0x0A12)
│     GUIDs:               6818cd18-4d60-5210-92ad-6a9bc39c8e00
[... 190 lines cut ...]
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─WDC WD6400AAKS-65A7B2:
      Device ID:           7b78ec0c6dd1b9234d1f8e64bee30b594d054399
      Summary:             ATA Drive
      Current version:     01.03B01
      Vendor:              Western Digital (ATA:0x101C)
      GUIDs:               ab5d87c6-74a4-53dd-8569-aff20d8b3e51
                           eb29fb40-64a5-5e8d-9d1e-a95092f2b464
                           887f344c-c0d5-5973-9c91-37fe389429cb
                           a4a79a6c-8336-52de-b056-1db0d22d6671
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
    
```

**Additional questions**
- Operating system and version: Arch Linux,  Kernel 5.5.10-arch1-1
- Have you tried rebooting? Yes
- Is this a regression? Yes

hughsie, 2020-03-23:
Hmm, I get: `failed to add USB device 0a12:1337: failed to GetStatus: only read 7bytes` -- debugging now.

Salt-Factory, 2020-03-23:
The headband is now detectable.
However, it seems that no updates are found.

```
# fwupdmgr update
• BLUETOOTH USB +EDR ADAPTER v2.1 UHE has no available firmware updates
• H05 has no available firmware updates
• Samsung SSD 850 EVO 500GB has no available firmware updates
• Samsung SSD 860 EVO 250GB has no available firmware updates
• WDC WD20EZRZ-00Z5HB0 has no available firmware updates
• WDC WD6400AAKS-65A7B2 has no available firmware updates
No updatable devices
```

```
# fwupdmgr reinstall 4745cb127beea10281899d84cd01952ed74156a6
No releases found
```

I will try downloading the cab and install manually now.

Salt-Factory, 2020-03-23:
Executing `fwupdmgr install aiaiai1.6.cab` installed the update, and my headset now works again.
It correctly updated to the 1.6 version, as I now have features that were not in 1.5.
Thanks!

However, it didn't update the current version.
`fwupdmgr get-devices` now shows the following:
```
├─H05:                                                         
│     Device ID:           4745cb127beea10281899d84cd01952ed74156a6
│     Summary:             Bluetooth Headphones
│     Current version:     1.5                
│     Vendor:              AIAIAI (USB:0x0A12)                
│     Update State:        success                                                                                                      
│     Update Error:        device version not updated on success, 1.6 != 1.5
│     Last modified:       2020-03-23 10:21
│     GUIDs:               6818cd18-4d60-5210-92ad-6a9bc39c8e00                                                                                                                                                                                                                 
│                          7e8318e1-27ae-55e4-a7a7-a35eff60e9bf
│     Device Flags:        • Updatable 
```

Luckily this bug has no impact whatsoever on the headband.

hughsie, 2020-03-23:
> However, it didn't update the current version.

If you reboot the headphones and re-attach, does it update?

Salt-Factory, 2020-03-23:
Yes, it does. 
Thanks again for doing this amazing work!
Never in a million years would I have expected to be able to update my headband using Linux.

closed 2020-03-23 by hughsie