fwupd/fwupd #1906: Error: No Gsettings scehmas are installed on this system [get-updates]

Modelmat, 2020-03-24:
**Describe the bug**
`fwupdmgr` crashes with `No Gsettings schemas are installed on this system`

**Steps to Reproduce**
Unable to run get-updates or anything further that requires it.
```shell
$ fwupdmgr get-updates -v         
(fwupdmgr:2106944): GLib-DEBUG: 10:40:15.691: setenv()/putenv() are not thread-safe and should not be used after threads are created
Firmware metadata has not been updated for 30 days and may not be up to date.

Update now? (Requires internet connection) [y|N]: y
(fwupdmgr:2106944): FuCommon-DEBUG: 10:40:16.482: creating path /home/modelmat/.cache/fwupd
(fwupdmgr:2106944): Fwupd-DEBUG: 10:40:16.482: looking for os-release at /etc/os-release
(fwupdmgr:2106944): FuMain-DEBUG: 10:40:16.484: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz to /home/modelmat/.cache/fwupd/lvfs-metadata.xml.gz
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
(fwupdmgr:2106944): GLib-GIO-DEBUG: 10:40:16.496: _g_io_module_get_default: Found default implementation gnutls (GTlsBackendGnutls) for ‘gio-tls-backend’

(fwupdmgr:2106944): GLib-GIO-ERROR **: 10:40:16.592: No GSettings schemas are installed on the system
[1]    2106944 trace trap (core dumped)  fwupdmgr get-updates -v
```

**Expected behavior**
Updates are downloaded and I can update firmware.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version: 1.3.9
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.3.9
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20NECTO1WW
│
├─SAMSUNG MZALQ512HALU-000L1:
│     Device ID:           71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:             NVM Express Solid State Drive
│     Current version:     7L1QFXV7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               e8ae98bc-bc37-56b1-9680-8d64884a2792
│                          8b86707a-1412-5389-8f66-725224d28879
│                          09dc27dc-8bd6-5b2f-b26f-1e6f8133e6ef
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           3ba7ea853f9613adbcbb9326dbe5f0decd677a67
│     Current version:     0.1.11
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                9bfd8b34-080b-40a1-47b3-ca1efa289e6d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM:
│     Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│     Current version:     73.8.17568.5511
│     Vendor:              ST Microelectronics (TPM:STM)
│     GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6
│                          84df3581-f896-54d2-bd1a-372602f04c32
│                          bfaed10a-bbc1-525b-a329-35da2f63e918
│                          70b7b833-7e1a-550a-a291-b94a12d0f319
│                          06f005e9-cb62-5d1a-82d9-13c534c53c48
│     Device Flags:        • Internal device
│   
├─UEFI Device Firmware:
│     Device ID:           1c8d3b6c7cb08805789e84e05fc8f3b097237f1e
│     Current version:     0.0.512
│     Vendor:              DMI:LENOVO
│     GUID:                3b7f92a6-36d0-49ae-a2d1-272f9a067274
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           e376734757f9aa50ce6f14bf7a32153141802c45
      Current version:     1.0.4
      Vendor:              DMI:LENOVO
      GUID:                19f39c71-f7b5-4df3-87d6-a01046ad6c0a
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

**Additional questions**
- Operating system and version: Arch Linux (`uname -a`: `Linux thoughtpad 5.1.16-arch1-1-ARCH #1 SMP PREEMPT Wed Jul 3 20:23:07 UTC 2019 x86_64 GNU/Linux`). Running on KDE - never used GNOME (from which GSettings is from?)
  - running an older version of the kernel because of issues with graphics drivers; I would prefer not to have to use the latest kernel.
- Have you tried rebooting? Yes
- Is this a regression? No idea.

superm1, 2020-03-24:
Try to run `glib-compile-schemas /usr/share/glib-2.0/schemas/`

Modelmat, 2020-03-24:
Nope, no change.
```
❯ sudo glib-compile-schemas /usr/share/glib-2.0/schemas/ 
[sudo] password for modelmat: 

~                                                                                                                                                                             1.65s 13:17:58
❯ fwupdmgr get-updates -v         
(fwupdmgr:76841): GLib-DEBUG: 13:18:00.554: setenv()/putenv() are not thread-safe and should not be used after threads are created
Firmware metadata has not been updated for 30 days and may not be up to date.

Update now? (Requires internet connection) [y|N]: y
(fwupdmgr:76841): FuCommon-DEBUG: 13:18:01.391: creating path /home/modelmat/.cache/fwupd
(fwupdmgr:76841): Fwupd-DEBUG: 13:18:01.393: looking for os-release at /etc/os-release
(fwupdmgr:76841): FuMain-DEBUG: 13:18:01.393: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz to /home/modelmat/.cache/fwupd/lvfs-metadata.xml.gz
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
(fwupdmgr:76841): GLib-GIO-DEBUG: 13:18:01.401: _g_io_module_get_default: Found default implementation gnutls (GTlsBackendGnutls) for ‘gio-tls-backend’

(fwupdmgr:76841): GLib-GIO-ERROR **: 13:18:01.445: No GSettings schemas are installed on the system
[1]    76841 trace trap (core dumped)  fwupdmgr get-updates -v

```

superm1, 2020-03-24:
Do you have `XDG_DATA_DIRS` set to something funny?

Modelmat, 2020-03-24:
Thanks, that fixes it :)
Turns out that I was doing `XDG_DATA_DIRS=$XDG_DATA_DIRS:/some/other/path` but I had nothing previously set.

closed 2020-03-24 by superm1