fwupd/fwupd #1914: Force update to latest firmware

FFY00, 2020-03-26:
**Is your feature request related to a problem? Please describe.**
When debugging firmware updates we need to force a firmware install. The only way I know how to do this is by forcing the install of a .cab file. This, however, is not optimal.

**Describe the solution you'd like**
I would like to have a flag to force an update to the latest version in the repos. At least in `fwupdtool`.

**Describe alternatives you've considered**
Update from a .cab file.

hughsie, 2020-03-26:
Why do you need to force it? Is there a requirements issue or something?

FFY00, 2020-03-26:
For eg. testing new changes for the logitech-hidpp plugin. There are 4 firmware variants for receivers, I need to get .cab files for all and update. If I were able to just pass a flag to force reinstalling the latest firmware from the repos it would save some time.

Or am I doing something wrong?

hughsie, 2020-03-26:
`fwupdmgr reinstall` ?

FFY00, 2020-03-26:
That should work. It would be great to also have that in `fwupdtool`.

hughsie, 2020-03-26:
Ahh! I'm covered in children at the moment but if you could do a PR I'd be glad to review it. It should be simple to port from fwupdmgr like the other commands. TIA.

FFY00, 2020-03-26:
Sure, I'll do it when I can find some time.

FFY00, 2020-03-30:
So, fwupd is not finding any available firmware for Logitech receivers, which are my primary targets with this.

You should be able to reproduce:
```
$ fwupdmgr reinstall
No releases found: Bootloader is not compatible
```

I also tried, as a hack, not stopping in the first firmware found ans it's still the same.

superm1, 2020-03-30:
Hmm, I had no problem with master just now (at fe4b3ead)

My current firmware:
```
└─Unifying Receiver:
      Device ID:           8ad41d663d3f4ea61a81cbb32a54411a101eb850
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0014
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server
```

Reinstalling:
```
$ fwupdmgr reinstall 
Choose a device:
0.	Cancel
1.	8357922f7f208e1f3b7537141ce17db04bc219d9 (Thunderbolt controller in Dell dock)
2.	b5a854df85ab30fff48e4365ec7ad2249c90806e (Package level of Dell dock)
3.	0acceea54e71c4d8002593822afe0f4705616613 (RTS5413 in Dell dock)
4.	96969b82d977fdcbf607df084ef0dcf10119409b (RTS5487 in Dell dock)
5.	228ac6846b64d532dd1ea1b7e651e8f5f55fbc34 (VMM5331 in Dell dock)
6.	570867f4ddda9915445da59bd85cd0e0e507270e (WD19TB)
7.	b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2 (System Firmware)
8.	8ad41d663d3f4ea61a81cbb32a54411a101eb850 (Unifying Receiver)
8
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: y
Downloading RQR12.10_B0032 for Unifying Receiver...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Installing on Unifying Receiver…
Idle…                    [***************************************]
Installing on Unifying Receiver…*********************************]
Writing…                 [***************************************]
Successfully installed firmware
```

superm1, 2020-03-30:
@FFY00 if you're not running master, does your receiver support signed updates?  My signed update receivers didn't work until c6ae0d998b7e5577b3de4763c25d9d90056bebf1

FFY00, 2020-03-30:
I tested running the daemon for master.

```
├─Unifying Receiver:
│     Device ID:           7f79a34e07df5677125dce5c4a147e2a7c2c45ef
│     Summary:             A miniaturised USB wireless receiver
│     Current version:     RQR12.09_B0030
│     Bootloader Version:  BOT01.04_B0015
│     Vendor:              USB:0x046D
│     Install Duration:    30 seconds
│     GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
│                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
│                          279ed287-3607-549e-bacc-f873bb9838c4
│     Device Flags:        • Updatable
│                          • Supported on remote server
```

```
$ fwupdmgr reinstall 7f79a34e07df5677125dce5c4a147e2a7c2c45ef
No releases found: Device Unifying Receiver doesn't support com.logitech.unifying
```

superm1, 2020-03-30:
Looking at https://fwupd.org/lvfs/devices/com.logitech.Unifying.RQR12.firmware I don't see any RQR12.09.* at all.  So that would explain why you can't reinstall anything.

FFY00, 2020-03-30:
Yes, that makes sense. However, this presents another issue:

```
$ fwupdmgr update
• Unifying Receiver has the latest available firmware version
```

This means if the firmware is not in the remote, users won't be able to update it.

superm1, 2020-03-30:
> Yes, that makes sense. However, this presents another issue:
> 
> ```
> $ fwupdmgr update
> • Unifying Receiver has the latest available firmware version
> ```
> 
> This means if the firmware is not in the remote, users won't be able to update it.

Hmm, so why won't it let you upgrade to RQ12.10.* like me?  I guess some bootloader requirement isn't matching?  You seem to have a newer bootloader than me on your particular dongle.  I'll have to see what I have on my others

superm1, 2020-03-30:
Here is the firmware requirement:

`    <firmware compare="regex" version="BOT01.0[0-3]_*">bootloader</firmware>`

So your bootloader is newer than that.  @hughsie how hard is that requirement?  Should it be expanded?

FFY00, 2020-03-30:
Hum, IIRC bootloader 1.04 is the signed version. Are you sure there are no firmwares for that version?

superm1, 2020-03-30:
Signed?  Then is your dongle not setting the right protocol?

hughsie, 2020-03-31:
> bootloader 1.04 is the signed version

This is correct. What's the protocol listed as if you do `--verbose`?

FFY00, 2020-03-31:
So, it detects the right protocol.

```
(fwupdmgr:3048502): FuMain-DEBUG: 13:35:29.274: Unifying Receiver
  DeviceId:             740d50bec148ff851e8050fd16138bea259ef939
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Summary:              A miniaturised USB wireless receiver
  Plugin:               logitech_hidpp
  Protocol:             com.logitech.unifyingsigned
  Flags:                updatable|supported|registered
  VendorId:             USB:0x046D
  Version:              RQR12.09_B0030
  VersionBootloader:    BOT01.04_B0015
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      30
  Created:              2020-03-31
```

hughsie, 2020-03-31:
You can't install unsigned firmware on an signed receiver.

FFY00, 2020-03-31:
Right, fwupd is making a bad job of finding the correct firmware. The unsupported firmware should be filtered out and never be returned by `fu_engine_get_releases_for_device` or maybe `fu_engine_get_upgrades`.

superm1, 2020-03-31:
> You can't install unsigned firmware on an signed receiver.

So it seems that a RQR12 release that supports signed firmware is missing on LVFS then too.

superm1, 2020-03-31:
I double checked the signed receivers that I have and I do have one with the RQR12 and one with RQR24.  The one with RQR24 reinstalls fine (although I opened a PR for a problem I noticed).

The two I have:
```
├─Unifying Receiver:
│     Device ID:           4e419dded9789c89c0050b3537950d03d13eb7a9
│     Summary:             A miniaturised USB wireless receiver
│     Current version:     RQR24.11_B0036
│     Bootloader Version:  BOT03.02_B0008
│     Vendor:              USB:0x046D
│     Install Duration:    30 seconds
│     GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
│                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
│                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
│     Device Flags:        • Updatable
│                          • Supported on remote server

├─Unifying Receiver:
│     Device ID:           4e419dded9789c89c0050b3537950d03d13eb7a9
│     Summary:             A miniaturised USB wireless receiver
│     Current version:     RQR12.09_B0030
│     Bootloader Version:  BOT01.04_B0016
│     Vendor:              USB:0x046D
│     Install Duration:    30 seconds
│     GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
│                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
│                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
│     Device Flags:        • Updatable
```

FFY00, 2020-03-31:
Can anyone explain how we select the supported releases? I was trying to find the code but I got stuck at `fu_engine_add_releases_for_device_component`.

superm1, 2020-03-31:
I think the problem is that the device is getting the supported flag applied when it shouldn't.
We probably need a pass of examining some of the requirements when that flag is normally added.

hughsie, 2020-03-31:
> I think the problem is that the device is getting the supported flag applied when it shouldn't.

That sounds plausible. I think we just match GUIDs for that.

>  We probably need a pass of examining some of the requirements when that flag is normally added.

Certainly. It's not "free", but I don't think it'll be particularly expensive. Will ponder.

hughsie, 2020-03-31:
> Will ponder.

https://github.com/fwupd/fwupd/pull/1939

superm1, 2020-04-01:
Fixed in #1939.

commit b2f377a0 (2020-03-31): fu-tool: add reinstall command

Closes #1914
closed 2020-04-01 by superm1