fwupd/fwupd #1915: Failing to run, unable to open ca-certifcates.crt (snap, regression)

tjyrinki, 2020-03-26:
**Describe the bug**
$ fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz

(fwupdmgr:873): GLib-Net-WARNING **: 18:18:32.768: couldn't load TLS file database: Failed to open file “/etc/ssl/certs/ca-certificates.crt”: No such file or directory

Failed to download https://cdn.fwupd.org/downloads/firmware.xml.gz: SSL handshake failed

**Steps to Reproduce**
1. Install/update fwupd snap
2. Try to run fwupdmgr

**Expected behavior**
Run normally, and in this case not using distribution specific assumptions in cross-distro packaging.

**fwupd version information**
fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.3.9

Installed with snap install fwupd --classic

**Additional questions**
- Operating system and version: openSUSE Tumbleweed 20200322
- Have you tried rebooting? Yes.
- Is this a regression? Yes. It used to work at least a couple of weeks ago, and has worked for a long time without problems.

hughsie, 2020-03-26:
I have no idea about what could have changed with snap. @superm1?

tjyrinki, 2020-03-26:
Possibly something starting to use SSL in a different way that assumes /etc/ca-certificates while SUSE keeps certificates at eg /usr/share/pki/trust ([stackexchange](https://unix.stackexchange.com/questions/80131/how-do-i-install-a-system-wide-ssl-certificate-on-opensuse))

tjyrinki, 2020-03-26:
Hmm, I first thought I can't install older versions ("Access by specifying a revision is not allowed for this Snap."), but I now found a few older ones from other channels. Turns out also version from January fails, while I'm sure it worked then.

Thus, this is an openSUSE Tumbleweed bug, and digging further something has removed a symlink, so this workaround helps:

`sudo ln -s /var/lib/ca-certificates/ca-bundle.pem /etc/ssl/certs/ca-certificates.crt`

Sorry for the trouble, I'll try to investigate the symlink problem on openSUSE side.

closed 2020-03-26 by tjyrinki