fwupd/fwupd #1918: 'not root-owned' and 'not found' error messages

LinuxOnTheDesktop, 2020-03-27:
**Describe the bug**
Suddenly, I'm seeing a new error - or two:

$ fwupdmgr
/ not root-owned 1000:1000
```
~ $ sudo fwupdmgr
[sudo] password for nicholas: 
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.2' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.7' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.1.1' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.8' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.4' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.6' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.1' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.1.3' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.5' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.9' not found (required by fwupdmgr)

```

**Steps to Reproduce**
Unknown, I'm afraid. But today I upgraded libre-office and with it, perhaps, some libraries.

**Expected behavior**
Clearly, not the above.

**fwupd version information**

```shell
$ fwupdmgr --version
/ not root-owned 1000:1000 [!]
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

snap

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
/ not root-owned 1000:1000 [!]
```

**Additional questions**
- Operating system and version: Mint 19.3 x64 Cinnamon
- Have you tried rebooting? No. I will try it now.
- Is this a regression? Yes, in that I have not had this problem before.

LinuxOnTheDesktop, 2020-03-27:
The plot thickens. I found that doing

~ $ sudo chown root /
~ $ sudo chown root:root /

allowed the snap to be reinstaled. I did so something recently that might have caused permission problems. Yet, [this web page](https://stackoverflow.com/questions/49506159/how-to-fix-go-not-root-owned) reports someone else having a similar problem . . having installed a snap with `--classic`. Suspicious!

hughsie, 2020-03-27:
I think you need to report this against snap in the Ubuntu bugtracker, sorry.

closed 2020-03-27 by hughsie

superm1, 2020-03-27:
> /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2

This sounds like a ✋ install that you should remove too if you want to use a classic snap.