fwupd/fwupd #1929: Firmware update fails to install on ThinkPad X270

hema8374, 2020-03-30:
**Describe the bug**
The system reboots without installing the new firmware. The issue happens on multiple Lenovo ThinkPad X270 models. The firmware can be updated when booting via a CD image, however.

**Steps to Reproduce**
On Arch Linux, run` fwupdmgr update`
Note that the firmware is downloaded and the update is scheduled as expected.
Reboot
After the reboot, note that the firmware was not installed.

The update and reboot steps can be repeated as many times as one wishes and the firmware will continue to download but not get applied on the reboot.

Also, the journal contains an error:

`fwupd[917]: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)`

**Expected behavior**
Expected firmware to be installed during the reboot process, as happens on other ThinkPad models I have.

Is it possible to install the fw cap file using install FILE [DEVICE-ID]

**fwupd version information**
```shell
fwupdmgr --version
client version: 1.3.9
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
20HN
│
├─HD Graphics 620:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     02
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               redacted ← PCI\VEN_8086&DEV_5916&REV_02
│                          redacted ← PCI\VEN_8086&DEV_5916
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─INTEL SSDPEKKF256G7L:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     121P
│     Vendor:              Intel Corporation (NVME:0x8086)
│     Serial Number:       redacted
│     GUIDs:               redacted ← NVME\VEN_8086&DEV_F1A5&REV_03
│                          redacted ← NVME\VEN_8086&DEV_F1A5
│                          redacted ← INTEL SSDPEKKF256G7L
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.6.29.3287
│     Bootloader Version:  11.6.29.3287
│     Vendor:              Intel Corporation
│     GUID:                redacted
[... 158 lines cut ...]
Mar 30 07:44:21 X270 polkitd[949]: Loading rules from directory /etc/polkit-1/rules.d
Mar 30 07:44:21 X270 polkitd[949]: Started polkitd version 0.116
Mar 30 07:44:21 X270 systemd[1]: Starting Authorization Manager...
Mar 30 07:44:21 X270 dbus-daemon[484]: [system] Activating via systemd: service name='org.freedesktop.PolicyKit1' unit='polkit.service' requested by ':1.188' (uid=0 pid=917 comm="/usr/lib/fwupd/fwupd ")
Mar 30 07:44:20 X270 fwupd[917]: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)
Mar 30 07:44:20 X270 kernel: audit: type=1130 audit(1585568660.938:86): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=upower comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? termi>
Mar 30 07:44:19 X270 systemd[1]: Condition check resulted in Commit a transient machine-id on disk being skipped.
Mar 30 07:44:19 X270 systemd[1]: Condition check resulted in First Boot Wizard being skipped.
Mar 30 07:44:19 X270 systemd[1]: Condition check resulted in Set Up Additional Binary Formats being skipped.
Mar 30 07:44:19 X270 systemd[1]: Condition check resulted in FUSE Control File System being skipped.
Mar 30 07:44:19 X270 dbus-daemon[484]: [system] Activating via systemd: service name='org.freedesktop.UPower' unit='upower.service' requested by ':1.188' (uid=0 pid=917 comm="/usr/lib/fwupd/fwupd ")
Mar 30 07:44:19 X270 systemd[1]: Starting Firmware update daemon...
Mar 30 07:44:19 X270 dbus-daemon[484]: [system] Activating via systemd: service name='org.freedesktop.fwupd' unit='fwupd.service' requested by ':1.187' (uid=0 pid=914 comm="fwupdmgr --version ")
```
**Additional questions**
- Operating system and version:
Arch Linux
- Have you tried rebooting?
Yes
- Is this a regression?
Not Sure
- Are you using an NVMe disk?
Yes
- Is secure boot enabled?
No

hema8374, 2020-03-31:
The issue might be related to the inability to set a bootnext value.

I intended to do these steps as part of troubleshooitng:

```
efibootmgr -v --create --disk /dev/sda --part 1 --loader /EFI/arch/fwupdx64.efi --label "ThinkPad X270 System Update 0.1.38"
efibootmgr --bootnext 000X
```
The output of the first command is:

```
...
Boot0001* ThinkPad X270 System Update 0.1.38    HD(1,GPT,redacted,0x800,0xeff7df)/File(\EFI\arch\fwupdx64.efi)
Could not read variable 'MirrorCurrent': No such file or directory
error trace:
 efivarfs.c:258 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/MirrorCurrent-7b9be2e0-e28a-4197-ad3e-32f062f9462c): No such file or directory
 lib.c:139 efi_get_variable(): ops->get_variable failed: No such file or directory
Could not read variable 'MirrorRequest': No such file or directory
error trace:
 efivarfs.c:258 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/MirrorRequest-7b9be2e0-e28a-4197-ad3e-32f062f9462c): No such file or directory
 lib.c:139 efi_get_variable(): ops->get_variable failed: No such file or directory
```
The new boot option is not created. 

As a further step, I did the following:
```
mv /boot/EFIBOOT/BOOTX64.EFI /boot/EFIBOOT/BOOTX64.EFI.original
mv /boot/arch/fwupdx64.efi /boot/EFIBOOT/BOOTX64.EFI
```
I rebooted and now the fw update was applied. I had to restore the original BOOTX64.EFI in order to boot my system normally. After doing so, history shows the fw update was applied as expected.

```
fwupdmgr get-history
20HN
│
└─System Firmware:
  │   Device ID:           8c997bbd20dfdf9c471aaae577d182d9ae77bca7
  │   Previous version:    0.1.37
  │   Update State:        success
  │   Last modified:       2020-03-30 11:21
  │   GUID:                redacted
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Cryptographic hash verification is available
  │                        • Device is usable for the duration of the update
  │
  └─ThinkPad X270 System Update:
        New version:       0.1.38
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X270 System Firmware
        License:           Proprietary
        Size:              9.4 MB
        Vendor:            Lenovo Ltd.
        Description:       Lenovo ThinkPad X270 System Firmware version 1.38.Update includes a security fix.Supported BIOS password authentication before entering into MEBx.Updated the Diagnostics module to version 04.11.000.

        Security issues fixed:

         • CVE-2019-0151
         • CVE-2019-0152
         • CVE-2019-0123
         • CVE-2019-0124
         • CVE-2019-0117
         • CVE-2019-0184
         • CVE-2019-0185
         • CVE-2019-14607
```

Is it possible to fix this so it works normally? 
In the mean time, is there a better workaround? Specifically, it seems that if I could set a bootnext value, it might solve the issue. What should I try next?

superm1, 2020-03-31:
Turn off boot order lock in BIOS setup

hema8374, 2020-03-31:
> Turn off boot order lock in BIOS setup

I will try that on the next available fw update. Is the solution really that simple?

superm1, 2020-03-31:
If you had it turned on in firmware setup yes.

hema8374, 2020-03-31:
> If you had it turned on in firmware setup yes.

It was on. Thank you for the tip.

superm1, 2020-03-31:
duplicate of https://github.com/fwupd/fwupd/issues/1730

closed 2020-03-31 by superm1